volkszaehler/vzlogger #256: Build requirements outdated

andig, 2016-05-13:
Ref: http://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version is not sufficient on recent raspbian:

	sudo apt-get install  libcurl3

	-- Could NOT find CURL (missing:  CURL_LIBRARY CURL_INCLUDE_DIR)
	-- FindGnuTls check
	-- ==> ''
	-- GNUTLS_HOME env is not set, setting it to /usr/local
	-- Looking for gnutls in /usr/local

	sudo apt-get install  libcurl4-gnutls-dev

	-- Found CURL: /usr/lib/arm-linux-gnueabihf/libcurl.so (found version "7.38.0")
	-- FindGnuTls check
	-- ==> ''
	-- GNUTLS_HOME env is not set, setting it to /usr/local
	-- Looking for gnutls in /usr/local
	==> GNUTLS_LIBRARIES='/usr/lib/arm-linux-gnueabihf/libgnutls.a;/usr/lib/arm-linux-gnueabihf/libz.a;-lp11-kit;/usr/lib/arm-linux-gnueabihf/libgmp.a;/usr/lib/arm-linux-gnueabihf/libhogweed.a;/usr/lib/arm-linux-gnueabihf/libgmp.a;/usr/lib/arm-linux-gnueabihf/libnettle.a;/usr/lib/arm-linux-gnueabihf/libtasn1.a;-lp11-kit;/usr/lib/arm-linux-gnueabihf/libz.a;-lsasl2;-lgcrypt'
	-- Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the system variable OPENSSL_ROOT_DIR (missing:  OPENSSL_LIBRARIES OPENSSL_INCLUDE_DIR)

	sudo apt-get install libssl-dev

	CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
	Please set them or make sure they are set and tested correctly in the CMake files:
	LIBGCRYPT
	    linked by target "vzlogger" in directory /home/pi/vzlogger/src

	sudo apt-get install libgcrypt20-dev

justinotherguy, 2016-05-13:
Moin,
sorry - I don't get it.
You use Jessy based raspbian, after running apt-get update; apt-get upgrade, right?
The output after the "sudo apt-get install" commands don't appear to be the ouput of the apt-get, right?
Can you please post the ouput of this on your RPi:
  sudo dpkg -l | grep -E "libcurl4-gnutls-dev|libcurl3|libssl-dev"

I get this on my RPi (v1 B) running Jessie:
ii  libcurl3:armhf                        7.38.0-4+deb8u3                           armhf        easy-to-use client-side URL transfer library (OpenSSL flavour)
ii  libcurl3-gnutls:armhf                 7.38.0-4+deb8u3                           armhf        easy-to-use client-side URL transfer library (GnuTLS flavour)
ii  libssl-dev:armhf                      1.0.1k-3+deb8u5                           armhf        Secure Sockets Layer toolkit - development files

At what stage do you get the output you have included?

Regards, J.

andig, 2016-05-14:
> sorry - I don't get it.

My bad. This was more a reminder, not a properly done error report. I've compiled vzlogger using `install.sh` and after each missing dependency tried to install it and restart compile. The log is the install steps plus follow-on compile output one by one .

> Can you please post the ouput of this on your RPi:

Pi2 running jessie

	ii  libcurl3:armhf                 7.38.0-4+deb8u3            armhf        easy-to-use client-side URL transfer library (OpenSSL flavour)
	ii  libcurl3-gnutls:armhf          7.38.0-4+deb8u3            armhf        easy-to-use client-side URL transfer library (GnuTLS flavour)
	ii  libcurl4-gnutls-dev:armhf      7.38.0-4+deb8u3            armhf        development files and documentation for libcurl (GnuTLS flavour)
	ii  libssl-dev:armhf               1.0.1k-3+deb8u5            armhf        Secure Sockets Layer toolkit - development files

For example this shows that imho libcurl3 is not sufficient (for whatever reason):

    sudo apt-get install  libcurl3
    (then the implicit restart of the compile here)
    -- Could NOT find CURL (missing:  CURL_LIBRARY CURL_INCLUDE_DIR)

andig, 2016-05-29:
Ping @justinotherguy

andig, 2016-05-29:
From the ml: libmbus now seems to require 

    apt-get install libltdl-dev

justinotherguy, 2016-05-29:
apparently the wiki is up to date:
(sudo) apt-get install build-essential git-core cmake pkg-config subversion libcurl4-openssl-dev libgnutls28-dev libsasl2-dev uuid-dev libtool libssl-dev libgcrypt20-dev libmicrohttpd-dev libltdl-dev libjson-c-dev
to run install.sh "dh-autoreconf" and "sudo" are also missing

seems to work:
 ./vzlogger/src/vzlogger -V
0.5.3
 based on git version: heads/master-0-gdba6644d4e
 last commit date: Sun, 29 May 2016 17:03:53 +0200

I have tested this on a clean jessie-x86-vm - test on a RPi is still pending - can you try again with these packages?

I have not changed anything regarding mbus - seems to be included:

ldd ./vzlogger/src/vzlogger | grep -i mbus
	libmbus.so.0 => /usr/local/lib/libmbus.so.0 (0x00007fe100246000)

andig, 2016-05-30:
Seems wiki has been updated :)

closed 2016-05-30 by andig