volkszaehler/vzlogger #258: make doesnt work - googlemock svn url outdated

tobwolf, 2016-05-14:
Hi,

make doesnt work because the googlemock svn url ist outdated.

vzlogger/gmock/CMakeLists.txt
http://googlemock.googlecode.com/svn/tags/release-1.7.0/

```
[ 34%] Performing download step (SVN checkout) for 'googlemock'
cd /home/tobii/vzlogger/gmock/src && /usr/bin/svn co http://googlemock.googlecode.com/svn/tags/release-1.7.0/ --non-interactive googlemock
svn: E160013: Unable to connect to a repository at URL 'http://googlemock.googlecode.com/svn/tags/release-1.7.0'
svn: E160013: »/svn/tags/release-1.7.0« Pfad nicht gefunden
gmock/CMakeFiles/googlemock.dir/build.make:85: recipe for target 'gmock/src/googlemock-stamp/googlemock-download' failed
```

they moved it to github: https://github.com/google/googlemock/

andig, 2016-05-14:
As a workaround you could remove svn for now

mbehr1, 2016-05-14:
I’m fixing it already. PR will come soon.

> 
> As a workaround you could remove svn for now

Gruß

Matthias

mbehr1, 2016-05-14:
See PR #259.

closed 2016-05-14 by andig