volkszaehler/vzlogger #260: "identifier" in "protocol":"file" is not used

Genosse274, 2016-05-16:
From the log file:
`[May 16 13:37:17][mtr2] Number of readers: 32
[May 16 13:37:17][mtr2] Config.daemon: 0
[May 16 13:37:17][mtr2] Config.local: 1
[May 16 13:37:17][]     MeterFile::read: 32, 32
[May 16 13:37:17][]     MeterFile::read: 'Temp:20.0'
[May 16 13:37:17][]     MeterFile::read: 0.000000, Temp:20.0, -1.000000
[May 16 13:37:17][]     MeterFile::read: 'Humidity:33.0'
[May 16 13:37:17][]     MeterFile::read: 0.000000, Humidity:33.0, -1.000000
[May 16 13:37:17][mtr2] Got 2 new readings from meter:
[May 16 13:37:17][mtr2] Reading: id=Temp:20.0/**StringItentifier: value=0.00** ts=1463398637723
[May 16 13:37:17][mtr2] Reading: id=Humidity:33.0/StringItentifier: value=0.00 ts=1463398637723
[May 16 13:37:17][chn3] Buffer dump (size=0): {}
[May 16 13:37:17][mtr2] Next reading in 10 seconds`
The other meters(d0) are working fine. Here the config:
`
      "enabled": true,
      "allowskip": false,
      "interval": 10,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": "270edb40-0d72-11e6-8659-e5297280cf1a",
          "identifier": "Temp",
          "api": "volkszaehler",
          "middleware": "http://localhost/middleware.php",
          "secretKey": "",
          "type": "device",
          "scaler": 1,
          "aggmode": "none",
          "duplicates": 0
        }
      ],
      "protocol": "file",
      "path": "/home/pi/Adafruit_Python_DHT/examples/DHT.log",
      "format": "$i:$v" //,
     //"rewind": true
`

Genosse274, 2016-05-27:
Update: I tried the workaround with protocol exec and got the same problem.
The Problem was that "format":"$i**:**$v" does not work because of the **# :** i think.
It works with empty space for protocol exec. Protocol file needs to be checked.
This "format":"$i**:**$v" is from the example in the explanation text of the vzlogger config editior.

mbehr1, 2016-05-28:
I wonder why there is no log printout for:
print(log_debug, "Parsed format string \"%s\" => \"%s\"", name().c_str(), config_format, scanf_format);

mbehr1, 2016-05-28:
Hmm. I guess sscanf is not smart enough. The "$i" get's translated into "%%2$ms". 
sscanf for %n$s does a 
```
 s      Matches a sequence of non-white-space characters; the next pointer must be a pointer to character array that is long  enough
              to  hold  the  input sequence and the terminating null byte ('\0'), which is added automatically.  The input string stops at
              white space or at the maximum field width, whichever occurs first.
```
so the identifier will be parsed to "Temp:20.0"
```
MeterFile::read: 0.000000, Temp:20.0, -1.000000
```
sscanf relies on whitespace for value separation.
--> do we need a smarter way for parsing strings? (some regex but we don't have c++ regex support due to some old gcc that needs to be supported...)

andig, 2016-05-28:
Any chance for newer gcc on jessie? Use the pcre library?

mbehr1, 2016-05-28:
We have a 
HAVE_CPP_REGEX

already and use regex with configoptions. So I could add support for MeterExec/File as well.

Shall I?

> 
> Any chance for newer gcc on jessie? Use the pcre library?
> 
> > 
> > Hmm. I guess sscanf is not smart enough. The "$i" get's translated into "%%2$ms". 
> > sscanf for %n$s does a
> > 
> > s Matches a sequence of non-white-space characters; the next pointer must be a pointer to character array that is long enough
> > to hold the input sequence and the terminating null byte ('\0'), which is added automatically. The input string stops at
> > white space or at the maximum field width, whichever occurs first.
> > so the identifier will be parsed to "Temp:20.0"
> > 
> > MeterFile::read: 0.000000, Temp:20.0, -1.000000
> > sscanf relies on whitespace for value separation.
> > --> do we need a smarter way for parsing strings? (some regex but we don't have c++ regex support due to some old gcc that needs to be supported...)

Gruß

Matthias

mbehr1, 2016-05-28:
regex needs at least gcc 4.9 to work properly.

> 
> Any chance for newer gcc on jessie? Use the pcre library?
> 
> > 
> > Hmm. I guess sscanf is not smart enough. The "$i" get's translated into "%%2$ms". 
> > sscanf for %n$s does a
> > 
> > s Matches a sequence of non-white-space characters; the next pointer must be a pointer to character array that is long enough
> > to hold the input sequence and the terminating null byte ('\0'), which is added automatically. The input string stops at
> > white space or at the maximum field width, whichever occurs first.
> > so the identifier will be parsed to "Temp:20.0"
> > 
> > MeterFile::read: 0.000000, Temp:20.0, -1.000000
> > sscanf relies on whitespace for value separation.
> > --> do we need a smarter way for parsing strings? (some regex but we don't have c++ regex support due to some old gcc that needs to be supported...)

Gruß

Matthias

mbehr1, 2016-05-28:
raspbian jessie seems to have gcc-4.9

> 
> regex needs at least gcc 4.9 to work properly.
> 
>> 
>> Any chance for newer gcc on jessie? Use the pcre library?
>> 
>> > 
>> > Hmm. I guess sscanf is not smart enough. The "$i" get's translated into "%%2$ms". 
>> > sscanf for %n$s does a
>> > 
>> > s Matches a sequence of non-white-space characters; the next pointer must be a pointer to character array that is long enough
>> > to hold the input sequence and the terminating null byte ('\0'), which is added automatically. The input string stops at
>> > white space or at the maximum field width, whichever occurs first.
>> > so the identifier will be parsed to "Temp:20.0"
>> > 
>> > MeterFile::read: 0.000000, Temp:20.0, -1.000000
>> > sscanf relies on whitespace for value separation.
>> > --> do we need a smarter way for parsing strings? (some regex but we don't have c++ regex support due to some old gcc that needs to be supported...)
>> > 
>> > 
>> 
> 
> Gruß
> 
> Matthias

Gruß

Matthias

andig, 2016-06-25:
VZ meeting: use regex of gcc 4.9, make MeterFile available only if regex support is available.

r00t-, 2023-03-03:
looks like this is still not fixed? :scream: 
https://github.com/volkszaehler/vzlogger/commits/master/src/protocols/MeterFile.cpp