volkszaehler/vzlogger #261: Improve retry mechnism

andig, 2016-05-25:
When middleware is unable to process data sent by vzlogger, especially when vzlogger sends multiple rows (e.g. >200), vzlogger keeps retrying and the amount of rows sent increases.

We need to check if there's a better way to handle resending large amounts of data, especially since https://github.com/volkszaehler/volkszaehler.org/pull/435 is implemented.

mbehr1, 2016-05-28:
would be easy to implement. But if max items are send at once the remaining ones would be send directly after (in max items chunks). Would this be ok or do we need some kind of delay between the transmissions?

andig, 2016-05-29:
Immediate send is ok, just not at once. Also recognizable by 'Maximum message size exceede'

mbehr1, 2016-05-29:
Shall we just fix a size and a avoid another config option?

> 
> Immediate send is ok, just not at once. Also recognizable by 'Maximum message size exceede'

Gruß

Matthias

andig, 2016-05-29:
I can live with fixed size. One record should be something like ~16 bytes, 16x64 = 1024 which is 1k memory. So how about 64? That should be enough to clean the backlog faster than new records arrive?

mbehr1, 2016-05-29:
see PR in #263.

andig, 2016-05-30:
Should we set a default max body size of 2K then for the middleware?

andig, 2016-06-25:
VZ meeting: merged.

closed 2016-06-25 by andig