volkszaehler/vzlogger #268: Build errors with make -j3

asdil12, 2016-06-30:
```
cc1plus: Fehler: /home/dominik/devel/vzlogger/gmock/src/googlemock/googlemock/include: Datei oder Verzeichnis nicht gefunden [-Werror=missing-include-dirs]
cc1plus: Fehler: /home/dominik/devel/vzlogger/gmock/src/googlemock/googletest/include: Datei oder Verzeichnis nicht gefunden [-Werror=missing-include-dirs]
/home/dominik/devel/vzlogger/tests/ut_api_volkszaehler.cpp:1:25: schwerwiegender Fehler: gtest/gtest.h: Datei oder Verzeichnis nicht gefunden
 #include "gtest/gtest.h"
                         ^
```

mbehr1, 2016-06-30:
Does this only happen with -j>=3? Seems to be missing dependency. I'll check how to add properly for cmake

mbehr1, 2016-06-30:
I can't report it here. Can you try adding
```
add_dependencies(vzlogger_unit_tests gmock_builder)
```
to tests/CMakeLists.txt
?

closed 2016-12-23 by andig