volkszaehler/vzlogger #274: [enhancement]: empty buffer faster

schnello, 2016-08-24:
Hallo.

Vorgeschichte:
Ich verwende hier einen Pi inkl vzlogger und den Volkszähler auf einem Ubuntu Server.
Den Ubuntu Server würde ich gerne über Nacht abschalten. 

Aktuelles Verhalten:
vzlogger puffert alle Werte solange der Server nicht erreichbar ist. Die Werte werden dann sobald der Server wieder online ist... nach und nach in die middleware geschrieben. 6 Stunden dauern aktuell etwa 4 - 6 Stunden.

Anfrage:
Wäre es möglich das der Puffer schneller in die Datenbank geschrieben wird?

andig, 2016-08-24:
> Die Werte werden dann sobald der Server wieder online ist... nach und nach in die middleware geschrieben

Schau mal in https://github.com/volkszaehler/vzlogger/commit/49694a2cf87690031e73cf319ab013546b971655 nach `MAX_CHUNKSIZE`. Damit verhindern wir DOS Attacken- evtl. kann der Defaultwert auch größer sein?

ping @mbehr1 was würde uns daran hindern zwar nur 64 Tupel, diese aber schneller hintereinander zu schreiben solange noch welche im Buffer sind?

schnello, 2016-08-24:
@andig 
Danke für den Hinweis. Schau ich mir an ob es mit einem höheren Wert schneller geht.

schnello, 2016-08-25:
Hab die `MAX_CHUNK_SIZE` angehoben und alle Warte waren innerhalb einer Stunde geschrieben. 
Also von meiner Seite her kann man das issue closen.

closed 2016-12-23 by andig