volkszaehler/vzlogger #275: How to change thestandard port from curl?

TylonHH, 2016-10-07:
Hi,
I changed in `/etc/apache2/port.conf` and `/etc/apache2/site-available/000-default.conf` and `.../sites-enabled/...' to the Port **81**, and rebooted. No success.
In the vzlogger I still see the wrong port 80.
````
[Oct 08 00:00:35][chn0] CURL: Connection 6 seems to be dead!
[Oct 08 00:00:35][chn0] CURL: Closing connection 6
[Oct 08 00:00:35][chn0] CURL: Hostname was found in DNS cache
[Oct 08 00:00:35][chn0] CURL:   Trying ::1...
[Oct 08 00:00:35][chn0] CURL: Connected to localhost (::1) port 80 (#7)
[Oct 08 00:00:35][chn0] CURL: Sent 41 bytes..
[Oct 08 00:00:35][chn0] CURL: Sent '[ [ 1475877635760, 219.74000000000001 ] ]' bytes
[Oct 08 00:00:35][chn0] CURL: upload completely sent off: 41 out of 41 bytes
[Oct 08 00:00:35][chn0] CURL: Server lighttpd/1.4.35 is not blacklisted
[Oct 08 00:00:35][chn0] CURL: Received 61 bytes
````

schnello, 2016-10-08:
Hallo.

Einfach bei der Adresse angeben

```
  "channels": [
        {
          "uuid": "45d3f960-7050-11e5-ac60-e51506d36ea9",
          "identifier": "1.7.0",
          "aggmode": "NONE",
          "api": "volkszaehler",
          "middleware": "http://192.168.1.210:81/middleware.php"
```

andig, 2016-10-08:
Please use the mailing list for howto questions.

closed 2016-10-08 by andig