volkszaehler/vzlogger #276: Missing protocol ocr

J-A-U, 2016-10-30:
Ich mache gerade meine ersten (echte) Schritte mit OCR am Wasserzähler:
```
[Oct 30 13:41:34][main] vzlogger v0.6.0 based on heads/master-0-g1484f9aad1 from Mon, 22 Aug 2016 22:32:51 +0200 started.
[Oct 30 13:41:34][push] Ignoring push entry due to empty array or duplicate section
[Oct 30 13:41:34][mtr0] Creating new meter with protocol ocr.
[Oct 30 13:41:34][mtr0] Missing protocol or invalid type (Protocol not found.)
[Oct 30 13:41:34][]     parse configuration failed due to:
[Oct 30 13:41:34]       Failed to parse configuration due to: Protocol not found.
```
[vzlogger.ocrtest.conf.txt](https://github.com/volkszaehler/vzlogger/files/560461/vzlogger.ocrtest.conf.txt)
Erstellt mit http://volkszaehler.github.io/vzlogger/

Fehler in der Config halte ich aber für fraglich weil:
```
vzlogger -c ~/vzlogger/etc/vzlogger.conf.meterOCR
[Oct 30 13:49:55][main] vzlogger v0.6.0 based on heads/master-0-g1484f9aad1 from Mon, 22 Aug 2016 22:32:51 +0200 started.
[Oct 30 13:49:55][mtr0] Missing protocol or invalid type (Protocol not found.)
[Oct 30 13:49:55][]     parse configuration failed due to:
[Oct 30 13:49:55]       Failed to parse configuration due to: Protocol not found
```

andig, 2016-10-30:
Mhhm. Hast Du das OCR mit eincompiliert?

mbehr1, 2016-10-30:
wie hast du vzlogger gebaut?
(Stichwort: configure ENABLE_OCR=true)

> Ich mache gerade meine ersten (echte) Schritte mit OCR am Wasserzähler:
>
> [Oct 30 13:41:34][main] vzlogger v0.6.0 based on heads/master-0-g1484f9aad1 from Mon, 22 Aug 2016 22:32:51 +0200 started. [...]

Gruß

Matthias

J-A-U, 2016-10-30:
Das ist ein Scherz, oder?
`cmake ENABLE_OCR=true .`

Der Hinweis das libleptonica-dev fehlt ist dann mitten in der Ausgabe versteckt und cmake wird als Erfolgreich abgeschlossen.
Nicht schön.

J-A-U, 2016-10-30:
Zusätzlich wäre es hilfreich wenn bei `vzlogger -V` auch erwähnt mit welchen Optionen er ausgerüstet ist.

andig, 2016-11-02:
Wärst Du so nett und machst ein Issue dafür auf?

Vielen Dank, Andreas

Das ist ein Scherz, oder?
cmake ENABLE_OCR=true .

closed 2016-11-02 by J-A-U