volkszaehler/vzlogger #278: Server stopped wenn daemon false

milefork, 2016-11-09:
Moin,

wollte mal den vzlogger im Vordergrund laufen lassen, aber er beendet sich auf unerklärliche weise.
Wenn er aber mit -d oder in der config "daemon": true steht startet er ohne Probleme und läuft Wochen druch.

Ich vermute es liegt an den Zeilen im Log? `Logging thread not started`

Hier mal der log vom Versuch ihn ohne daemon zu starten:

http://pastebin.com/XdA1DSXu

Gruß

J-A-U, 2017-01-19:
Ich kenne zwar den genauen Grund nicht, aber vzlogger wurde dieses Verhalten nachträglich hinzugefügt.
Bei deamon=false macht er genau einen Durchgang und wird dann beendet. Er gibt alle Ergebnisse über die Konsole aus und schickt _keine_ Daten an die Middleware.

andig, 2017-01-19:
Das hab ich vor langer Zeit verbockt und keiner korrigiert. By design ist es nicht!

Viele Grüße,
Andreas

J-A-U, 2017-01-19:
Zeile 151 in src/threads.cpp ist der Auslöser für beenden des vzloggers:
`		} while (options.daemon() || options.local() || options.logging() );`

Da lag milefork wohl richtig mit seiner Annahme das abgestelltes logging die Ursache ist.
Nachdem ich Zeile 401 in src/vzlogger.cpp 
`	options.logging(options.daemon());`
auskommentiert hatte lief er durch und schickte die Daten auch an die MW.

Was mich stutzig macht ist der Kommentar davor:
`	// only logging to file if daemonizing`
File? Klingt als wär das logfile gemeint!?
Options.logging ist aber der Schalter fürs Senden der Daten an die MW. Das logfile wird gefüttert egal ob "daemon": true oder false ist.

andig, 2017-01-19:
> File? Klingt als wär das logfile gemeint!?
> Options.logging ist aber der Schalter fürs Senden der Daten an die MW. Das logfile wird gefüttert egal ob "daemon": true oder false ist.

Das klingt genau so als hätte ich es damals aus diesem Grunde kaputt gemacht. Die Benenung ist Grütze. 

Ich sehe eigentlich überhaupt keinen Grund das `options.logging` i.S. von MW überhaupt zu haben. Falls doch sollte es `options.sendtobackend` oder so ähnlich heissen. Da würde sich ein search/replace über die ganze Codebase anbieten damits nicht nochmal schief geht :)

J-A-U, 2017-01-19:
Als ich über ne Alternative gegrübelt hab kam mir in Sinn das logging doch nicht soooo weit entfernt ist. Immerhin heißt das Ding vz_logger_. ;) 

Was ich im Moment noch nicht überblicken kann: Hätte das entferne der Option Auswirkungen auf den Betrieb als local ohne backend (VZ oder MySmartGrid)?

andig, 2017-01-19:
> Was ich im Moment noch nicht überblicken kann: Hätte das entferne der Option Auswirkungen auf den Betrieb als local ohne backend (VZ oder MySmartGrid)?

Denke nein- schließlich muss die Backend Connectivity ja für jedes Meter/Kanal ohnehin konfiguriert werden. Wenn da nix drin steht sollte auch nicht ge(backend)loggt werden.

J-A-U, 2017-01-19:
Schau mal bitte drüber: https://github.com/J-A-U/vzlogger/pull/7

Die Endloschleife stört mich ein wenig. Vzlogger bekommt man im Vordergrund nur mit <Strg>+<C> beendet.
Wobei die Schleife, von der Logik her, auch schon vorher so war.

J-A-U, 2017-01-20:
Ich hab bei mir drüben mal die Endlosschleife aufgelöst: https://github.com/J-A-U/vzlogger/pull/8
Das entspricht zwar nicht den Erwartungen von milefork, eröffnet Usern aber die Möglichkeit vzlogger z.B. auch über crontab aufzurufen. Wer den Dauerläufer haben will soll ihn als daemon laufen lassen (dafür isser da).
Mal davon ab das so ne Endlosschleife einfach nur Murks ist.

Glaube auch das es Irritationen reduziert wenn die Meldung dazu in jedem Fall (log_error) kommt. Anhand des mitgelieferten Parameters erkennt man ja schnell das der vzlogger wegen daemon=false geschlossen wurde.

andig, 2017-01-20:
> Ich hab bei mir drüben mal die Endlosschleife aufgelöst: J-A-U#8

Eigentlich müsste das laufen bis SIG_HUP?

Viele Grüße,
Andreas

J-A-U, 2017-01-24:
SIGHUP zur planmäßigen Beendigung des vzlogger wenn er nicht als Hintergrundprpzess läuft?

Müsste er? Muss er? Soll er?

closed 2017-02-14 by J-A-U