volkszaehler/vzlogger #279: vzlogger not working with local sml meter while vzlogger.conf contains "host"?

justinotherguy, 2016-11-12:
Moin,

I do this:
- I have an sml meter connected to a serial port of a RPi
- I generate a vzlogger.conf with editor: the https://volkszaehler.github.io/vzlogger/
- the conf file (see below) does not work, unless I comment out the "host" line
  it comes out like this:
      "host": "",
  result (vzlogger.conf):
[Nov 12 22:17:51][main] vzlogger v0.6.0 based on heads/master-0-g1484f9aad1 from Mon, 22 Aug 2016 22:32:51 +0200 started.
[Nov 12 22:17:51][mtr0] Creating new meter with protocol sml.
[Nov 12 22:17:51][sml]  pullseq len:0 found
[Nov 12 22:17:51][mtr0] Meter configured, enabled.
[Nov 12 22:17:51]       New meter initialized (protocol=sml)
[Nov 12 22:17:51]       Configure channel.
[Nov 12 22:17:51][chn0] New channel initialized (uuid=...<...> api=volkszaehler id=1-0:15.7.0*255)
[Nov 12 22:17:51]       Have 1 meters.
[Nov 12 22:17:51][main] log level is 15
[Nov 12 22:17:52][main] daemon=1, local=0
[Nov 12 22:17:52]       Daemonize process...
[Nov 12 22:17:52]       Opened logfile /var/log/vzlogger.log
[Nov 12 22:17:52][push] No pushDataServer defined.
[Nov 12 22:17:52][]     ===> Start meters
[Nov 12 22:17:52][mtr0] Meter connection established
[Nov 12 22:17:52][mtr0] Meter thread started
[Nov 12 22:17:52][mtr0] Meter is opened. Starting channels.
[Nov 12 22:17:52][chn0] Logging thread started
[Nov 12 22:17:52][]     Startup done.
[Nov 12 22:17:52][chn0] Start logging thread for volkszaehler-api. Running as daemon: yes
[Nov 12 22:17:52][chn0] Using default volkszaehler api.
[Nov 12 22:17:52][mtr0] Number of readers: 32
[Nov 12 22:17:52][mtr0] Config.daemon: 1
[Nov 12 22:17:52][mtr0] Config.local: 0

-> no further log entries until I stop vzlogger.

commenting out "host" leads to a regular operation of vzlogger, the log starts like this:
[Nov 12 22:19:37][main] vzlogger v0.6.0 based on heads/master-0-g1484f9aad1 from Mon, 22 Aug 2016 22:32:51 +0200 started.
[Nov 12 22:19:37][mtr0] Creating new meter with protocol sml.
[Nov 12 22:19:37][sml]  pullseq len:0 found
[Nov 12 22:19:37][mtr0] Meter configured, enabled.
[Nov 12 22:19:37]       New meter initialized (protocol=sml)
[Nov 12 22:19:37]       Configure channel.
[Nov 12 22:19:37][chn0] New channel initialized (uuid=...<...> api=volkszaehler id=1-0:15.7.0*255)
[Nov 12 22:19:37]       Have 1 meters.
[Nov 12 22:19:37][main] log level is 15
[Nov 12 22:19:37][main] daemon=1, local=0
[Nov 12 22:19:37]       Daemonize process...
[Nov 12 22:19:37]       Opened logfile /var/log/vzlogger.log
[Nov 12 22:19:37][push] No pushDataServer defined.
[Nov 12 22:19:37][]     ===> Start meters
[Nov 12 22:19:37][mtr0] Meter connection established
[Nov 12 22:19:37][mtr0] Meter thread started
[Nov 12 22:19:37][mtr0] Meter is opened. Starting channels.
[Nov 12 22:19:37][chn0] Logging thread started
[Nov 12 22:19:37][]     Startup done.
[Nov 12 22:19:37][mtr0] Number of readers: 32
[Nov 12 22:19:37][mtr0] Config.daemon: 1
[Nov 12 22:19:37][mtr0] Config.local: 0
[Nov 12 22:19:37][chn0] Start logging thread for volkszaehler-api. Running as daemon: yes
[Nov 12 22:19:37][chn0] Using default volkszaehler api.
[Nov 12 22:19:39][mtr0] Got 6 new readings from meter:
[Nov 12 22:19:39][mtr0] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=10340848.80 ts=1478985579033
[Nov 12 22:19:39][mtr0] Reading: id=1-0:2.8.0*255/ObisIdentifier:1-0:2.8.0*255 value=459.20 ts=1478985579033
[Nov 12 22:19:39][mtr0] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=10340848.80 ts=1478985579033
[Nov 12 22:19:39][mtr0] Reading: id=1-0:2.8.1*255/ObisIdentifier:1-0:2.8.1*255 value=459.20 ts=1478985579033
[Nov 12 22:19:39][mtr0] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=0.00 ts=1478985579033
[Nov 12 22:19:39][mtr0] Reading: id=1-0:15.7.0*255/ObisIdentifier:1-0:15.7.0*255 value=467.60 ts=1478985579033
[Nov 12 22:19:39][chn0] Adding reading to queue (value=467.60 ts=1478985579033)
[Nov 12 22:19:39][chn0] ==> number of tuples: 1
[Nov 12 22:19:39][chn0] compare: 0 1478985579033
[Nov 12 22:19:39][chn0] copied 1/1 values for middleware transmission
[Nov 12 22:19:39][chn0] JSON request body: [ [ 1478985579033, 467.60000000000002 ] ]
[Nov 12 22:19:39][chn0] CURL: Hostname was NOT found in DNS cache
[Nov 12 22:19:39][chn0] CURL:   Trying 5.45.110.30...
[Nov 12 22:19:39][chn0] CURL: Connected to demo.volkszaehler.org (5.45.110.30) port 80 (#0)
[Nov 12 22:19:39][chn0] CURL: Sent 41 bytes..
[Nov 12 22:19:39][chn0] CURL: Sent '[ [ 1478985579033, 467.60000000000002 ] ]' bytes
[Nov 12 22:19:39][chn0] CURL: upload completely sent off: 41 out of 41 bytes
[Nov 12 22:19:39][chn0] CURL: HTTP 1.0, assume close after body
[Nov 12 22:19:39][chn0] CURL: Received 26 bytes
[Nov 12 22:19:39][chn0] CURL: Received '{"version":"0.3","rows":1}' bytes
[Nov 12 22:19:39][chn0] CURL: Closing connection 0
[Nov 12 22:19:39][chn0] CURL Request succeeded with code: 200
[Nov 12 22:19:39][chn0] emptied all (1) values
...and so on.

The log says it - I use vzlogger in this version:
vzlogger -V
0.6.0
 based on git version: heads/master-0-g1484f9aad1
 last commit date: Mon, 22 Aug 2016 22:32:51 [+0200](url)

Here's my (broken) vzlogger.conf:
{
  "retry": 0,
  "daemon": true,
  "verbosity": 15,
  "log": "/var/log/vzlogger.log",
  "meters": [
    {
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "channels": [
        {
          "uuid": <uuid>,
          "identifier": "1-0:15.7.0*255",
          "api": "volkszaehler",
          "middleware": "http://demo.volkszaehler.org/middleware.php",
          "aggmode": "none",
          "duplicates": 0
        }
      ],
      "protocol": "sml",
      "host": "",
      "device": "/dev/ttyUSB0",
      "pullseq": "",
      "baudrate": 9600,
      "parity": "8n1"
    }
  ]
}

Best regards, J.

andig, 2016-12-15:
Empty host parameter should be raised as error imho.

J-A-U, 2016-12-16:
Zeile 74 in src/protocol/MeterD0.cpp
			if ( !_device.length() ) throw vz::VZException("device without length");
ist doch eine solche Überprüfung?

Klappt das auch als
		if ( !_host.length() ) throw vz::VZException("host without length");
?

andig, 2016-12-16:
Klar, wieso nicht?

mbehr1, 2016-12-16:
aber wenn eines von beiden angegeben ist (mit Länge) sollten wir die andere Angabe ignorieren und nicht Exception schmeißen. Nur wenn beide fehlen -> Exc.
Oder?

> 
> Klar, wieso nicht?

Gruß

Matthias

J-A-U, 2016-12-16:
Es wäre zumindest eine schnelle Lösung das vzlogger auch ne Meldung ausgibt und nicht einfach nur Leistung verbrät ohne produktiv zu sein..

Wenn du eine schöne Lösung in absehbarer Zeit hast wäre mir die auch lieber als mein Gefrickel von dem ich eh nur die Hälfte versteh.

J-A-U, 2017-01-16:
> aber wenn eines von beiden angegeben ist (mit Länge) sollten wir die andere Angabe ignorieren und nicht Exception schmeißen. Nur wenn beide fehlen -> Exc.
Oder?

Soweit ich das überblicke liegt unser Problem ja darin das der nachfolgende Code die Variable _host vor _device bearbeitet. Wenn _host fehlt macht das nix, ist _host aber mit Wert "" belegt gehts in die Hose.
Also müssen wir die Länge prüfen bevor wir die Variable _host erstellen/schreiben?
Zusätzliche Funktion in Options.cpp?

andig, 2017-01-19:
Imho ja.

mbehr1, 2018-08-05:
Fix sollte mit #358 vorliegen.

closed 2018-08-05 by andig