volkszaehler/vzlogger #294: Fehler beim Kompilieren von vzlogger unter Ubuntu Server 16.10

s1nn3r, 2017-01-27:
Bis auf den vzlogger kann ich problemlos alles kompilieren, beim vzlogger kommt beim Linken der Objekte folgender Fehler:
`[ 33%] Linking CXX executable vzlogger
cd /tmp/vzlogger/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/vzlogger.dir/link.txt --verbose=1
/usr/bin/c++    -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++11 -fpermissive -Wno-error=redundant-decls -Wno-ignored-qualifiers   CMakeFiles/vzlogger.dir/vzlogger.cpp.o CMakeFiles/vzlogger.dir/ltqnorm.cpp.o CMakeFiles/vzlogger.dir/Meter.cpp.o CMakeFiles/vzlogger.dir/__/gitSha1.cpp.o CMakeFiles/vzlogger.dir/CurlSessionProvider.cpp.o CMakeFiles/vzlogger.dir/PushData.cpp.o  -o vzlogger -rdynamic protocols/libproto.a libvz.a api/libvz-api.a /usr/local/lib/libjson-c.a -lrt -Wl,-Bstatic -lsml -Wl,-Bdynamic -lrt /usr/local/lib/libmbus.so -lm -Wl,-Bstatic -lmicrohttpd -Wl,-Bdynamic -lrt -lgcrypt -lpthread -lm -luuid -ldl -lcurl -Wl,-Bstatic -lgnutls -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lgmp -lhogweed -lgmp -lnettle -ltasn1 -lidn -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lz -Wl,-Bdynamic -lsasl2 -lgcrypt -lssl -lcrypto -Wl,-Bstatic -lsml -Wl,-Bdynamic /usr/local/lib/libmbus.so -lm -Wl,-Bstatic -lmicrohttpd -Wl,-Bdynamic -lgcrypt -lpthread -lm -luuid -ldl -lcurl -Wl,-Bstatic -lgnutls -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lgmp -lhogweed -lnettle -ltasn1 -lidn -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lgmp -lhogweed -lnettle -ltasn1 -lidn -lz -Wl,-Bdynamic -lsasl2 -lgcrypt -lssl -lcrypto -Wl,-rpath,/usr/local/lib
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libmicrohttpd.a(libmicrohttpd_la-connection.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libmicrohttpd.a(libmicrohttpd_la-reason_phrase.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libmicrohttpd.a(libmicrohttpd_la-daemon.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libmicrohttpd.a(libmicrohttpd_la-response.o): relocation R_X86_64_32 against `.text' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libmicrohttpd.a(libmicrohttpd_la-connection_https.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Nicht darstellbarer Abschnitt bei Ausgabe`

Jemand eine Idee, wie ich das trotzdem kompiliert bekomme? Danke!

Viele Grüße
Marc

andig, 2017-01-27:
Du compilierst auf Windows? Schonmal geschaut obs bei libmicrohttpd einen Fehler dazu gibt? Ansonsten wäre die exakte Plattform sicher hilfreich.

s1nn3r, 2017-01-27:
Sorry, hatte ich nur in den Betreff geschrieben: Ubuntu Server 16.10

s1nn3r, 2017-01-27:
Habe nun libmicrohttpd aus den Quellen selber kompiliert, danach lief auch vzlogger durch. 
Kann also geschlossen werden. Danke!

closed 2017-01-28 by J-A-U