volkszaehler/vzlogger #295: Installation von vzlogger schlägt fehl

UdoSchake, 2017-02-12:
Bei make kommt folgender Fehler:
`[ 64%] Building CXX object src/api/CMakeFiles/vz-api.dir/Volkszaehler.cpp.o
cd /vzlogger/src/api && /usr/bin/c++   -DHAVE_CONFIG_HPP -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-he                       aders -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++11 -fpermissi                       ve -Wno-error=redundant-decls -I/usr/local/include -I/vzlogger -I/vzlogger/include    -g3 -o CMakeFiles/vz-api.dir/ Volkszaehler.cpp.o -c /vzlogger/src/api/Volkszaehler.cpp
/vzlogger/src/api/Volkszaehler.cpp: In member function ‘virtual void vz::api::Volkszaehler::send()’:
**/vzlogger/src/api/Volkszaehler.cpp:149:25: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]**
cc1plus: all warnings being treated as errors
make[2]: *** [src/api/CMakeFiles/vz-api.dir/Volkszaehler.cpp.o] Error 1
make[2]: Leaving directory `/vzlogger'
make[1]: *** [src/api/CMakeFiles/vz-api.dir/all] Error 2
make[1]: Leaving directory `/vzlogger'
make: *** [all] Error 2
`

J-A-U, 2017-02-13:
Hast du irgendwas anders gemacht als sonst?
Optionen, andere Distri, neue gcc-Version?

Hab gestern nur mein Repo neu gebaut, da hatte ich keine Probleme in der Art. Lass jetzt aber mal das öffentliche durchrödeln...

UdoSchake, 2017-02-13:
Alsooo: Das System ist ein älteres Bananian:
Linux bananapi 3.4.103+ #1 SMP PREEMPT Mon Dec 15 17:37:57 CET 2014 armv7l
gcc ist die richtige Version.
root@bananapi ~ # gcc --version
gcc (Debian 4.7.2-5) 4.7.2
Alle weiteren dependencies für Wheezy wurden installiert.
Heute noch mal neu ausprobiert (war mir nicht sicher ob ich gcc erneuert hatte.) Aber der Fehler trat wieder auf:

`[ 64%] Building CXX object src/api/CMakeFiles/vz-api.dir/Volkszaehler.cpp.o
cd /root/vzlogger/src/api && /usr/bin/c++   -DHAVE_CONFIG_HPP -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++11 -fpermissive -Wno-error=redundant-decls -Wno-ignored-qualifiers -I/usr/local/include -I/root/vzlogger -I/root/vzlogger/include    -g3 -o CMakeFiles/vz-api.dir/Volkszaehler.cpp.o -c /root/vzlogger/src/api/Volkszaehler.cpp
/root/vzlogger/src/api/Volkszaehler.cpp: In member function ‘virtual void vz::api::Volkszaehler::send()’:
/root/vzlogger/src/api/Volkszaehler.cpp:149:25: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
cc1plus: all warnings being treated as errors
make[2]: *** [src/api/CMakeFiles/vz-api.dir/Volkszaehler.cpp.o] Error 1
make[2]: Leaving directory `/root/vzlogger'
make[1]: *** [src/api/CMakeFiles/vz-api.dir/all] Error 2
make[1]: Leaving directory `/root/vzlogger'
make: *** [all] Error 2
`

Gruß
Udo

J-A-U, 2017-02-13:
Mein Bananian ist leider etwas neuer: 3.4.108, basiernd auf Debian 8.4.
Gcc ist 4.9.2 wie auch aufm Rpi.

Die betreffende Stelle lief gerade durch, ohne Probleme.

Was mich allerdings stutzig macht: Volkzaehler.cpp kommt bei mir bei ~20%, nicht erst bei 64%.

UdoSchake, 2017-02-13:
wie installierst du denn? Zu Fuß oder mit dem script?

Gruß
Udo

J-A-U, 2017-02-13:
Ich hab das damals glaub ohne Script gemacht.

J-A-U, 2017-02-13:
Grad auch nochmal mit neuem libsml und json-c probiert - geht. Ohne Script.

J-A-U, 2017-02-13:
install.sh geht gar nicht weil mein Bananian kein sudo kennt.

mbehr1, 2017-02-13:
Hallo Udo,

schau ich mir gleich an. 

> Alsooo: Das System ist ein älteres Bananian:
> Linux bananapi 3.4.103+ #1 <https://github.com/volkszaehler/vzlogger/pull/1> SMP PREEMPT Mon Dec 15 17:37:57 CET 2014 armv7l
> gcc ist die richtige [...]

Gruß

Matthias

mbehr1, 2017-02-13:
Änder mal

> /root/vzlogger/src/api/Volkszaehler.cpp 
> 
> Zeile 149 in

if (_values.size() <= (size_t) MAX_CHUNK_SIZE) {

(btw: du solltest nicht als root compilieren…)

> Alsooo: Das System ist ein älteres Bananian:
> Linux bananapi 3.4.103+ #1 <https://github.com/volkszaehler/vzlogger/pull/1> SMP PREEMPT Mon Dec 15 17:37:57 CET 2014 armv7l
> gcc ist die richtige [...]

Gruß

Matthias

J-A-U, 2017-02-14:
> (btw: du solltest nicht als root compilieren…)

In der Installationsanleitung stehen die ganzen sudo auch mit drin.  :/

UdoSchake, 2017-02-14:
> In der Installationsanleitung stehen die ganzen sudo auch mit drin. :/
Das ist auch richtig so. Nur bei 'Building vzlogger' wird unterschieden.

Das ich bei dem alten Bananian hier mit root arbeite liegt daran, dass kein user eingerichtet ist. Ich weiß, dass das nicht ok ist.

Gruß
Udo

UdoSchake, 2017-02-14:
> Änder mal
>  /root/vzlogger/src/api/Volkszaehler.cpp
>  Zeile 149 in
> if (_values.size() <= (size_t) MAX_CHUNK_SIZE) {

Ja, das war es. Damit läuft make durch.

Gruß
Udo

J-A-U, 2017-02-14:
> > In der Installationsanleitung stehen die ganzen sudo auch mit drin. :/
> Das ist auch richtig so. Nur bei 'Building vzlogger' wird unterschieden.

Wenn man das richtige Verzeichnis wählt (in dem man Schreibrechte hat) dann ist (geschätzt) die Hälfte davon unnötig. Vergleiche mal mit install.sh. Da stehen nur 6 sudo drin.

andig, 2017-02-14:
> In der Installationsanleitung stehen die ganzen sudo auch mit drin. :/
> Das ist auch richtig so. Nur bei 'Building vzlogger' wird unterschieden.
>
> Wenn man das richtige Verzeichnis wählt (in dem man [...]

Deshalb finde ich es ganz gut wenn wir im Wiki nur die Sachen dokumentieren
die nicht ohnehin schon automatisiert sind.

vg
Andreas

closed 2017-02-14 by UdoSchake