volkszaehler/vzlogger #310: Send Zero

Sineos, 2017-03-19:
Same settings as #309 but `send_zero = true` for the gasmeter makes the readings completely erratic:

[image]

My understanding / hope was that  `send_zero = true` would pull the reading to 0 when no gas is consumed instead of calculating a small left-over between "long-distance impulses".

frankrichter, 2017-03-19:
It doesn't make sense to use `send_zero` without `aggtime` set.

Sineos, 2017-03-19:
Thanks @frankrichter, good to know. I somehow did not really find a documentation on this parameter.

The vzlogger config editor suggest sum as `aggmode` for S0. This leads me to two further questions:
- How do I come up with a reasonable `aggtime` in this case
- How does `aggtime` actually work with sum? Lets assume I set `aggtime` to 10, will it return the sum of the impules over 10 minutes? (which sounds somehow weird to me) 

Actually my goal is the highest possible precision while having a true 0 when no gas is consumed.

J-A-U, 2017-03-19:
> Lets assume I set aggtime to 10, will it return the sum of the impules over 10 minutes?

10 seconds

> Actually my goal is the highest possible precision while having a true 0 when no gas is consumed.

I say its impossible to get a real 0 with Impulses.

frankrichter, 2017-03-19:
> How do I come up with a reasonable aggtime in this case

I would start testing with aggtime slightly longer than the usual timespan between 2 pulses.

> How does aggtime actually work with sum? Lets assume I set aggtime to 10, will it return the sum of the impules over 10 minutes? (which sounds somehow weird to me)

aggtime is set in seconds, but apart from that it works exactly this way.

> Actually my goal is the highest possible precision while having a true 0 when no gas is consumed.

That's not technically not possible with impulses.

Sineos, 2017-03-19:
Ok, then I will stay with the distinct values without aggregation and live with the mathematical "error" from the impulse logic.

Many thanks for the explanations.

closed 2017-03-19 by Sineos