volkszaehler/vzlogger #311: S0 meters and send_zero

Sineos, 2017-03-20:
I think it would be incredibly useful if S0 meters would return a defined "off" state without having to use aggregate functions.

Just as a brainstorming how this could be achieved, following idea:
Every monitored system usually have some kind of "working range" consisting of
- a maximum value
- a minimum value
- an "off" state

After some time of monitoring a user usually knows quite well the working range of a given system.
Now for send_zero, the challenge is to differentiate between the minimum of the working range and the off state.

How it could work:
- Define a `minimum` point in the settings of a channel
- With the `resolution` setting and the above minimum value, calculate the rate of impulses needed to "sustain" this minimum value, i.e. "What rate (time) of impulses would be needed for a minimum reading?"
- With each impulse arriving for this channel, set a time-out corresponding to the rate calculated above
- Each new impulse would reset / retrigger this time-out
- When the time-out is reached, call send_zero

Example:
[image]

The graph above shows:
- My gas consumption collected with an S0 meter
- The power regulation of my central heating, taken directly out of the heating control system

When the heating powers down, it can clearly be seen on the gas consumption and when the heating is off, the "lagging" of the S0 becomes visible.

Now for the idea above, I would define the minimum of the gas consumption somewhere around 200 liters.
- The meter has a resolution of 10 liters / Impulse 
- Every second 1 Impulse --> 3600 Impulses / h --> 36000 liters / h
- 200 / 36000 * 3600 = 20 seconds 

So if my math was correct, I would need 1 Impulse every 20 seconds to sustain a rate of 200 liters. This in turn would lead to a time-out value of 20 seconds after which a Zero would be send should no new Impulse arrive.

Does this makes sense?

frankrichter, 2017-03-20:
Try `send_zero` with `"aggtime": 20` and `"aggmode": "none"`.

Sineos, 2017-03-20:
As the documentation requires "sum" as aggmode for S0, would you please explain the difference / effect of an aggmode "sum" versus "none".
Just trying to understand some of vzlogger concepts ;-)

frankrichter, 2017-03-20:
aggmode "none" should transmit all buffered tuples (without loss of resolution) when aggtime is over.

J-A-U, 2017-03-21:
> The graph above shows:
•My gas consumption collected with an S0 meter

The graph for the gas consumtion is wrong, it should be a stepped line like the power regulation.

Sineos, 2017-03-21:
@J-A-U 
Why do you think the graph is wrong? The resolution / speed of the gas meter should definitively be lower than the power control of the heating. In addition this might be a zoom factor issue as well. See here:

[image]

As for the send_zero topic, I think something is wrong here. I do not get any sane values when `send_zero` is set to true:

1. First try
```
"aggmode": "none"
"aggtime": 20
"send_zero": true
```
 
[image]

2nd try
```
"aggmode": "sum"
"aggtime": 20
"send_zero": true
```
[image]

3rd try
```
"aggmode": "avg"
"aggtime": 20
"send_zero": true
```
[image]

The graph in Volkszaehler.org looks exactly the same as in Grafana

frankrichter, 2017-03-21:
Your graphs show that the timespan between 2 pulses seems to be longer than 20s. Please retry with 60s or 120s aggtime.

frankrichter, 2017-03-21:
> Why do you think the graph is wrong?

Compare the graph in your first post with volkszaehler frontend when set to style = steps in channel properties.

frankrichter, 2017-03-21:
> Now for the idea above, I would define the minimum of the gas consumption somewhere around 200 liters.
> The meter has a resolution of 10 liters / Impulse
> Every second 1 Impulse --> 3600 Impulses / h --> 36000 liters / h
> 200 / 36000 * 3600 = 20 seconds

Your calculation looks somehow weird. 1 pulse every 20 seconds would result in 1800l/h consumption.
Consumption of 200l/h means 20 pulses per hour or 180s between 2 pulses.

Sineos, 2017-03-22:
> Your calculation looks somehow weird. 1 pulse every 20 seconds would result in 1800l/h consumption.
> Consumption of 200l/h means 20 pulses per hour or 180s between 2 pulses.

Of course you are right, brain fart on my side. Thanks for the correction

> Compare the graph in your first post with volkszaehler frontend when set to style = steps in channel properties.

Ok, I don't get it. For me the graph looks right.
The following screenshot from Grafana shows:
- Gas consumption via vzlogger Influxdb API
- Gas consumption via Volkszaehler push-server (identical to Influxdb API)
- Power regulation of the heating.

The second screenshot is the same time span in Volkszaehler.org. All graphs are set to "step" style now. No aggtime, no send_zero, aggmode: "sum"

[image]
[image]

frankrichter, 2017-03-22:
> Ok, I don't get it. For me the graph looks right.

I (and JAU, afaik) referred to the screenshot in your first post in this issue. This graph is not in steps style and therefore clearly wrong in periods with no gas consumption (21:30 - 22:45 for example).
The area under power charts should represent the energy or consumption, which is only true when you plot  it in steps style.
Your latest graph looks reasonably better, but can you explain the difference between Volkszaehler and Grafana between 03:18 and 03:33? Volkszaehler chart looks better for this period.

>The second screenshot is the same time span in Volkszaehler.org. All graphs are set to "step" style now. No aggtime, no send_zero, aggmode: "sum"

aggmode does nothing with no aggtime set. 

Did you try with aggtime around 180s (200l/h) and send_zero: true? I think this could be a valid solution. If you need the resolution below 3 minutes you can use aggmode: none to get all tuples.

Please keep in mind that aggregation settings don't affect push server data.

J-A-U, 2017-03-22:
Darfs auch deutsch sein?
Ich hab den Eindruck wir spreichen beide nicht in unserer Muttersprache. Und ich bekomms in deutsch besser ausgedrückt.

> Ok, I don't get it. For me the graph looks right.

Ich hol ein wenig weiter aus und nehme, um den Unterschied zu verdeutlichen, erstmal einen Momentanwert Temperatur an.
Problem bei Momentanwerten ist immer das wir nie Wissen was zwischen zwei Messpunkten vor sich ging. Da Temperaturen eher nicht wild umherspringen ist die beste Annahme eine Linie die beide Punkte steigend oder fallen miteinander verbindet. Wir nehemen das auch gerne für Werte an die sich flinker ändern können (Strom z.B.). Die Annahme ist auch garnicht schlecht weil wir eins sicher Wissen: Unser Momentanwert zum Zeitpunkt der Messung ist korrekt.

Bei Impulsen (bzw. Zählerstände im Allgemeinen) ist das ein wenig anders. Wir Wissen nämlich nur das sich zum Zeitpuntk des Impulses (bzw Flanke) der Arbeitssumme geändert hat. Wir Wissen aber nicht ob die Leistung zu dem Zeitpunkt der Flanke nur so dahintröppelt oder relative steil gestiegen ist. Wir haben also keinen gesicherten Momentanwert mit dem wir arbeiten können!
Daher ist die beste Annhame die wir treffen können eine gleichmäßige Leistung über den Zeitraum zum vorletzten Impuls. Das ist was der vzlogger für dich macht wenn du identifier power wählst. Er ermittelt die mittlere Leistung für den Zeitraum.
Wichtig furs Verständnis warum wir das so handhaben: Unser Graph hat in der horizontalen die Zeit, wir können daher in der Vertikalen eigentlich gmessene Arbeit nicht darstellen sondern nur Leistung.
Und das ist bei einer geraden Linie zwischen den Zeitpunkten auch mathematisch korrekt: Die Fläche unter dem Strich ist nämlich Zeit lang und Leistung hoch. Leistung * Zeit = Arbeit.

Wenn du die Wert die dir vzlogger liefert mit einer steigenden oder fallenden Linie verbindest ist diese Fläche unter der Linie falsch.

Ich hoffe das klärt auch ein Stück weit warum ein "send zero" bei Impulszählern inkorrekt ist. Denn wir generieren dabei willkürlich zusätzliche Zeitpunkte die in der Darstellung zu falschen Ergebnissen führen.

Sineos, 2017-03-22:
> I (and JAU, afaik) referred to the screenshot in your first post in this issue. This graph is not in steps style and therefore clearly wrong in periods with no gas consumption (21:30 - 22:45 for [...]

Ah, now I get it. It was about the step style

> Your latest graph looks reasonably better, but can you explain the difference between Volkszaehler and Grafana between 03:18 and 03:33? Volkszaehler chart looks better for this period.

Actually I can't. Going in the details it seems that the measurements are "shifted":

```
Grafana:
03:18:18 to 03:33:36 219
03:33:36 to 03:34:33 39
03:34:33 to 03:36:56 626
```

```
Volkszaehler.org
03:15:37 to 03:18:17 219
03:18:17 to 03:33:37 39
03:33:37 to 03:34:28 626
```

Checking the values in the InfluxDB this matches to the graph.
vzlogger Influxdb API
`SELECT "value" FROM "Consumption_Gas" WHERE time > '2017-03-22T02:15:00.000Z' AND time < '2017-03-22T02:40:00.000Z'`

```
time	value
2017-03-22T02:15:34.665Z	218.479745
2017-03-22T02:18:18.822Z	219.302253
2017-03-22T02:33:36.483Z	39.230173
2017-03-22T02:34:33.954Z	626.402882
2017-03-22T02:36:56.344Z	252.826743
2017-03-22T02:39:30.093Z	234.147865
```

Volkszaehler push-server
`SELECT "value" *1000 FROM "raw_Gas" WHERE time > '2017-03-22T02:15:00.000Z' AND time < '2017-03-22T02:40:00.000Z'`

```
time	value
2017-03-22T02:15:34.665Z	218.47974510696
2017-03-22T02:18:18.822Z	219.30225333065002
2017-03-22T02:33:36.483Z	39.230173233906996
2017-03-22T02:34:33.954Z	626.40288145325
2017-03-22T02:36:56.344Z	252.82674345109
2017-03-22T02:39:30.093Z	234.14786437635
```

The values via vzlogger InfluxDB API and push-server (via Node-Red to MQTT to InfluxDB) are identical.

> Did you try with aggtime around 180s (200l/h) and send_zero: true? I think this could be a valid solution. If you need the resolution below 3 minutes you can use aggmode: none to get all tuples.
> 
> Please keep in mind that aggregation settings don't affect push server data.

Following Settings are used:
```
 "meters": [
    { // Gas
      "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": 180,
      "aggfixedinterval": false,
      "channels": [
        {
          "api": "volkszaehler",
          "uuid": "edc95ef0-dfd0-11e6-a86b-5d27a2655935",
          "identifier": "Impulse",
          "middleware": "http://192.168.178.8/volkszaehler/middleware.php",
          "aggmode": "sum",
          "duplicates": 0
        },
        {
          "api": "influxdb",
          "uuid": "edc95ef0-dfd0-11e6-a86b-5d27a2655935",
          "identifier": "Power",
          "host": "http://192.168.178.8:8086",
          "database": "vzlogger",
          "measurement_name": "Consumption_Gas",
          "username": "",
          "password": "",
          "max_batch_inserts": 4500,
          "max_buffer_size": 450000,
          "timeout": 30,
          "aggmode": "none"
        }
      ],
      "protocol": "s0",
      "gpio": 4,
      "gpio_dir": -1,
      "configureGPIO": true,
      "resolution": 100,
      "send_zero": true,
      "debounce_delay": 30,
      "nonblocking_delay": 100000
    },
```
Note that Volkszaehler API is set to "sum" while Influx to "none". Just wanted to check if this affects either push-server and / or Volkszaehler.

Here is the result:

[image]
[image]

For InfluxDB, both push-server and InfluxDB API are identical.

frankrichter, 2017-03-22:
Volkszaehler frontend (set to steps) plots the value of a tuple left from the timestamp (which is correct in this case). Grafana seems to handle this different, therefore you see this shift by 1 timestamp which makes the Grafana graph incorrect. This style exists also in vz (states), but it is not usable for energy measurements. Can you change the drawing style?

In my opinion the vz graph with aggregation settings doesn't look too bad: you get the "real zero" you were aiming for. aggmode: none should give you the full pulse resolution if 3 minutes aren't enough.

frankrichter, 2017-03-23:
I took a look in the code, `send_zero: true` and `aggmode: none` will most probably not work as i expected.
This is because send_zero pushes 0 to the buffer every second, and with aggmode: none all buffered readings are transmitted. I thought send_zero produces one zero-reading at the end off aggtime, but this is not how it really works. Sorry, i have no practical experience with S0-meters.
IMHO it would be a nice feature to make this timespan configurable as suggested in your first post.
The approach to calculate this timespan from a minimum value and resolution seems a bit overengineered to me. Most vzlogger configs don't even need the resolution option.
But what about configuring the timespan directly like we do with the duplicates option?

In the meantime, please try your influxdb channel with `aggtime: 180` and `aggmode:  avg`.

Sineos, 2017-03-23:
@J-A-U 
Vielen Dank für die Erklärung.

@frankrichter 
> Grafana seems to handle this different, therefore you see this shift by 1 timestamp which makes the Grafana graph incorrect. This style exists also in vz (states), but it is not usable for energy measurements. Can you change the drawing style?

This matches to @J-A-U explanation above. I will try to setup a show case and approach the Grafana Devs with it. Afaik there is no option to change it.

Now for something really weird:
Both Volkszaehler API and InfluxDB API are set to
```
"aggmode": "sum"
"aggtime": 180
"send_zero": true
```
[image]

In Volkszaehler the Graph looks OK, the values are sane.

[image]

In Grafana, the green graph (push-server) shows the effect as described by @frankrichter :

> This is because send_zero pushes 0 to the buffer every second, and with aggmode: none all buffered readings are transmitted. 

The yellow graph is totally off limits and this is also proven by the DB values:
```
time	value
2017-03-23T20:02:01.407Z	68746.206114
2017-03-23T20:05:01.387Z	108267.107076
2017-03-23T20:08:01.626Z	71512.528648
2017-03-23T20:11:01.034Z	104859.332321
2017-03-23T20:14:01.164Z	83262.807451
2017-03-23T20:17:01.268Z	78611.166968
```

LOL, this is really driving me crazy. But anyway, I already learned a lot and really thank you to @frankrichter and @J-A-U for driving me through this and your patience 👍

frankrichter, 2017-03-23:
You have to use aggmode: avg when using S0 meters with identifier "power". aggmode: sum makes only sense on impulse readings.

Sineos, 2017-03-25:
> Volkszaehler frontend (set to steps) plots the value of a tuple left from the timestamp (which is correct in this case). Grafana seems to handle this different, therefore you see this shift by 1 [...]

Just to verify that I understood @frankrichter and @J-A-U german explanation correctly:
For impulse measurement the "test right" graph would be correct. The logic behind is:
- At time 2 the impulses give a power value of 10
- We assume between time 1 and time 2 we had: Work = Delta(t) x 10
- The "test left" graph is wrong here, since for the same time it would give a Work of 0
- "test left" would be suitable, e.g. to plot temperature over time 

[image]

> You have to use aggmode: avg when using S0 meters with identifier "power". aggmode: sum makes only sense on impulse readings.

aggmode set to 'avg' for the InfluxDB API:
[image]

In Volkszaehler.org the graph looks good, but for Grafana it just makes no sense

frankrichter, 2017-03-25:
> Just to verify that I understood @frankrichter and @J-A-U german explanation correctly:
For impulse measurement the "test right" graph would be correct. The logic behind is:
At time 2 the impulses give a power value of 10
We assume between time 1 and time 2 we had: Work = Delta(t) x 10
The "test left" graph is wrong here, since for the same time it would give a Work of 0

Exactly.

> "test left" would be suitable, e.g. to plot temperature over time

Not really, in case of temperature i would connect the data points directly without steps (vz style "lines").  We use "states" for working hour counters or valve states or in general for sensors which only transmit their value when it has changed.

> aggmode set to 'avg' for the InfluxDB API:
grafik
In Volkszaehler.org the graph looks good, but for Grafana it just makes no sense

Perhaps send_zero and identifier power can't be used together in a sensible way, alltough i can't explain why at the moment. 
Can you use impulse readings in InfluxDB directly (without conversion to power)?

J-A-U, 2017-03-25:
> > Just to verify that I understood @frankrichter and @J-A-U german explanation correctly:
For impulse measurement the "test right" graph would be correct. The logic behind is:
At time 2 the impulses give a power value of 10
We assume between time 1 and time 2 we had: Work = Delta(t) x 10
The "test left" graph is wrong here, since for the same time it would give a Work of 0

> Exactly.

No!

Test left is the correct one because the power value got calculated from the the actual work value, the actual timestamp and the _previous_ timestamp .
So the graph has to draw the a line of power value (10) from the previus timestamp (1) to the actual timestamp (2).

frankrichter, 2017-03-25:
Ich glaub wir meinen eigentlich das gleiche. Hast du bemerkt dass "test left" im Bild der rechte ist? IMHO ist der linke Graph mit der Bezeichnung "test right" richtig.

J-A-U, 2017-03-25:
Ne, war mir entgangen. :/

Sineos, 2017-03-25:
Sorry für die Verwirrung ;-)
In highchart heißt der Parameter für die jeweiligen Step Darstellungen left bzw. right

andig, 2017-03-25:
Yep. Step before und step after. Wir haben dafür steps und states.

andig, 2017-08-29:
@Sineos gibts hier noch was zu diskutieren? Sieht mir "erledigt" aus?

Sineos, 2018-01-24:
Danke für die Unterstützung

closed 2018-01-24 by Sineos