volkszaehler/vzlogger #315: Does not compile due to Werror

kbabioch, 2017-06-13:
The current version does not compile with a recent toolchain due to the -Werror flag:

    cd /home/kbabioch/vzlogger-git/src/vzlogger/build/src/protocols && /usr/bin/c++  -DHAVE_CONFIG_HPP -I/home/kbabioch/vzlogger-git/src/vzlogger/build -I/home/kbabioch/vzlogger-git/src/vzlogger/include -I/home/kbabioch/vzlogger-git/src/vzlogger  -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++11 -fpermissive -Wno-error=redundant-decls -Wno-ignored-qualifiers   -g3 -o CMakeFiles/proto.dir/MeterFile.cpp.o -c /home/kbabioch/vzlogger-git/src/vzlogger/src/protocols/MeterFile.cpp
    /home/kbabioch/vzlogger-git/src/vzlogger/src/protocols/MeterFile.cpp: In constructor 'MeterFile::MeterFile(std::__cxx11::list<Option>)':
    /home/kbabioch/vzlogger-git/src/vzlogger/src/protocols/MeterFile.cpp:81:25: error: this statement may fall through [-Werror=implicit-fallthrough=]
           scanf_format[j++] = '%'; // add double %% to escape a conversion identifier
           ~~~~~~~~~~~~~~~~~~^~~~~
    /home/kbabioch/vzlogger-git/src/vzlogger/src/protocols/MeterFile.cpp:83:6: note: here
          default:
          ^~~~~~~
    cc1plus: all warnings being treated as errors
    make[2]: *** [src/protocols/CMakeFiles/proto.dir/build.make:162: src/protocols/CMakeFiles/proto.dir/MeterFile.cpp.o] Error 1
    make[2]: Leaving directory '/home/kbabioch/vzlogger-git/src/vzlogger/build'
    make[1]: *** [CMakeFiles/Makefile2:1173: src/protocols/CMakeFiles/proto.dir/all] Error 2
    make[1]: Leaving directory '/home/kbabioch/vzlogger-git/src/vzlogger/build'
    make: *** [Makefile:166: all] Error 2

kbabioch, 2017-06-13:
[kbabioch@antares build]$ LANG=C c++ --version
c++ (GCC) 7.1.1 20170528
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

r00t-, 2017-06-14:
is this the ONLY warning/error?

google says:
https://dzone.com/articles/implicit-fallthrough-in-gcc-7

that lists all the possible ways to cancel the warning (error),
even adding a *comment* (possibly mentioning "fall through" will do)...

r00t-, 2017-06-14:
maybe you could mention which distribution you are using that already has gcc-7,
that might help with verifying/fixing this.
debian unstable does not have it packaged.
(also, we would surely accept a pull request fixing this.)

mbehr1, 2017-06-14:
I'll fix it tomorrow.  

Gruß
Matthias 

Sent from a mobile device.

kbabioch, 2017-06-14:
I was able to fix the issue with the implicit fallthroughs by adding an attribute, which is described [here][https://developers.redhat.com/blog/2017/03/10/wimplicit-fallthrough-in-gcc-7/]. However I'm not sure how to fix this in a generic way without requiring C++17 (you are using C++11 right now).

I'm using Arch Linux, which has GCC available for quite some time already.

Ampelbein, 2017-10-02:
How about just a check if it is compiled under GCC7.1 and then using -Wimplicit-fallthrough=0? Like:

`if(CMAKE_COMPILER_IS_GNUCC AND NOT (CMAKE_CXX_COMPILER_VERSION LESS 7.1))`
`  message("Disable implicit-fallthrough on GCC7.1 and higher")`
`  set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wimplicit-fallthrough=0")`
`endif()`

That way it will still compile with older GCCs or different compilers.

kbabioch, 2017-10-02:
@Ampelbein Seems more than a workaround and not like a real fix, but should be good enough for the time being.

mbehr1, 2017-10-03:
I guess a simple
// fallthrough

comment should fix it.
diff --git a/src/protocols/MeterFile.cpp b/src/protocols/MeterFile.cpp
index ccf68ed..b158eef 100644
--- a/src/protocols/MeterFile.cpp
+++ b/src/protocols/MeterFile.cpp
@@ -79,7 +79,7 @@ MeterFile::MeterFile(std::list<Option> options)
 
                                        case '%':
                                                scanf_format[j++] = '%'; // add double %% to escape a conversion identifier
-                                               // nobreak;
+                        // fallthrough
                                        default:
                                                scanf_format[j++] = config_format[i]; // just copying
                        }

I’ll give it a try now by installing gcc 7.1 on my machine (might take some time…).

> 
> @Ampelbein <https://github.com/ampelbein> Seems more than a workaround and not like a real fix, but should be good enough for the time being.

Gruß

Matthias

mbehr1, 2017-10-03:
yep. that works.

I’ll add travis-ci support for gcc-7. json-c fails with similar problems. Fixing those as well.

> 
> I guess a simple
> // fallthrough
> 
> comment should fix it.
> diff --git a/src/protocols/MeterFile.cpp b/src/protocols/MeterFile.cpp
> index ccf68ed..b158eef 100644
> --- a/src/protocols/MeterFile.cpp
> +++ b/src/protocols/MeterFile.cpp
> @@ -79,7 +79,7 @@ MeterFile::MeterFile(std::list<Option> options)
>  
>                                         case '%':
>                                                 scanf_format[j++] = '%'; // add double %% to escape a conversion identifier
> -                                               // nobreak;
> +                        // fallthrough
>                                         default:
>                                                 scanf_format[j++] = config_format[i]; // just copying
>                         }
> 
> I’ll give it a try now by installing gcc 7.1 on my machine (might take some time…).
> 
>> 
>> @Ampelbein <https://github.com/ampelbein> Seems more than a workaround and not like a real fix, but should be good enough for the time being.
>> 
>> 
> 
> Gruß
> 
> Matthias

Gruß

Matthias

mbehr1, 2017-10-03:
See #332

r00t-, 2017-10-22:
@mbehr1:
i love how i suggested that fix five months ago...

andig, 2017-10-22:
@r00t- hattest Du einen PR gestellt der uns durchgerutscht ist? Dann tuts mir leid :(

r00t-, 2017-10-22:
@andig:
kein PR, nur weiter oben hier im thread:
> [...] all the possible ways to cancel the warning (error),
> even adding a comment (possibly mentioning "fall through") will do...

ich hatte auch keinen gcc7 zum testen zur hand, und haette gehofft der teporter macht das fuer uns,

kbabioch, 2017-11-07:
Since the latest versions can be compiled again, I'm closing this issue.

closed 2017-11-07 by kbabioch