volkszaehler/vzlogger #316: Compiler error with GCC 7

kbabioch, 2017-06-13:
cd /home/kbabioch/vzlogger-git/src/vzlogger/build/src/api && /usr/bin/c++  -DHAVE_CONFIG_HPP -I/home/kbabioch/vzlogger-git/src/vzlogger/build -I/home/kbabioch/vzlogger-git/src/vzlogger/include -I/home/kbabioch/vzlogger-git/src/vzlogger  -W -Wall -Wextra -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++11 -fpermissive -Wno-error=redundant-decls -Wno-ignored-qualifiers   -g3 -o CMakeFiles/vz-api.dir/MySmartGrid.cpp.o -c /home/kbabioch/vzlogger-git/src/vzlogger/src/api/MySmartGrid.cpp
    /home/kbabioch/vzlogger-git/src/vzlogger/src/api/MySmartGrid.cpp: In member function 'void vz::api::MySmartGrid::hmac_sha1(char*, const unsigned char*, size_t)':
    /home/kbabioch/vzlogger-git/src/vzlogger/src/api/MySmartGrid.cpp:578:11: error: aggregate 'HMAC_CTX hmacContext' has incomplete type and cannot be defined
      HMAC_CTX hmacContext;
               ^~~~~~~~~~~
    /home/kbabioch/vzlogger-git/src/vzlogger/src/api/MySmartGrid.cpp:573:6: warning: 'snprintf' output truncated before the last format character [-Wformat-truncation=]
     void vz::api::MySmartGrid::hmac_sha1(
          ^~
    /home/kbabioch/vzlogger-git/src/vzlogger/src/api/MySmartGrid.cpp:591:11: note: 'snprintf' output 4 bytes into a destination of size 3
       snprintf(s, 3, "%02x:", out[i]);
       ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
    make[2]: *** [src/api/CMakeFiles/vz-api.dir/build.make:90: src/api/CMakeFiles/vz-api.dir/MySmartGrid.cpp.o] Error 1
    make[2]: Leaving directory '/home/kbabioch/vzlogger-git/src/vzlogger/build'
    make[1]: *** [CMakeFiles/Makefile2:1118: src/api/CMakeFiles/vz-api.dir/all] Error 2
    make[1]: Leaving directory '/home/kbabioch/vzlogger-git/src/vzlogger/build'
    make: *** [Makefile:166: all] Error 2

    [kbabioch@antares build]$ LANG=C c++ --version
    c++ (GCC) 7.1.1 20170528
    Copyright (C) 2017 Free Software Foundation, Inc.
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

kbabioch, 2017-06-13:
[kbabioch@antares build]$ LANG=C c++ --version
    c++ (GCC) 7.1.1 20170528
    Copyright (C) 2017 Free Software Foundation, Inc.
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

andig, 2017-08-29:
ping @mbehr1 raspbian is still on 6.3 but I guess this problem will hurt us sooner or later?

mbehr1, 2017-10-03:
see #332

closed 2019-03-14 by andig