volkszaehler/vzlogger #317: Compiler issues with OpenSSL 1.1.0

kbabioch, 2017-06-13:
cd /home/kbabioch/vzlogger/build/src/protocols && /usr/bin/c++  -DHAVE_CONFIG_HPP -I/home/kbabioch/vzlogger/build -I/home/kbabioch/vzlogger/include -I/home/kbabioch/vzlogger  -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++11 -fpermissive -Wno-error=redundant-decls -Wno-ignored-qualifiers   -g3 -o CMakeFiles/proto.dir/MeterOMS.cpp.o -c /home/kbabioch/vzlogger/src/protocols/MeterOMS.cpp
    /home/kbabioch/vzlogger/src/protocols/MeterOMS.cpp: In constructor 'MeterOMS::MeterOMS(const std::__cxx11::list<Option>&, MeterOMS::OMSHWif*)':
    /home/kbabioch/vzlogger/src/protocols/MeterOMS.cpp:145:21: error: 'void OPENSSL_config(const char*)' is deprecated [-Werror=deprecated-declarations]
      OPENSSL_config(NULL);
                         ^
    In file included from /usr/include/openssl/crypto.h:32:0,
                     from /usr/include/openssl/bio.h:20,
                     from /usr/include/openssl/conf.h:13,
                     from /home/kbabioch/vzlogger/src/protocols/MeterOMS.cpp:13:
    /usr/include/openssl/conf.h:92:1: note: declared here
     DEPRECATEDIN_1_1_0(void OPENSSL_config(const char *config_name))
     ^
    cc1plus: all warnings being treated as errors
    make[2]: *** [src/protocols/CMakeFiles/proto.dir/build.make:258: src/protocols/CMakeFiles/proto.dir/MeterOMS.cpp.o] Error 1
    make[2]: Leaving directory '/home/kbabioch/vzlogger/build'
    make[1]: *** [CMakeFiles/Makefile2:1173: src/protocols/CMakeFiles/proto.dir/all] Error 2
    make[1]: Leaving directory '/home/kbabioch/vzlogger/build'
    make: *** [Makefile:166: all] Error 2

closed 2017-06-18 by mbehr1

okrause, 2017-08-20:
Just clean installed a Raspi (Raspbian based on stretch) and compiling vzlogger from scratch. Install documentation (dependencies) need update, but thats another topic.

When trying to compile commit d30f922d89da0c6b8a63f0d68d94199903f84653, I still get this error:

`/home/pi/vzlogger/src/protocols/MeterOMS.cpp:145:2: error: ‘void OPENSSL_config(const char*)’ is deprecated [-Werror=deprecated-declaration]
  OPENSSL_config(NULL);
  ^~~~~~~~~~~~~~
In file included from /usr/include/openssl/crypto.h:32:0,
                 from /usr/include/openssl/bio.h:20,
                 from /usr/include/openssl/conf.h:13,
                 from /home/pi/vzlogger/src/protocols/MeterOMS.cpp:13:
/usr/include/openssl/conf.h:92:1: note: declared here
 DEPRECATEDIN_1_1_0(void OPENSSL_config(const char *config_name))
 ^
/home/pi/vzlogger/src/protocols/MeterOMS.cpp:145:21: error: ‘void OPENSSL_config(const char*)’ is deprecated [-Werror=deprecated-declarations]
  OPENSSL_config(NULL);
                     ^
In file included from /usr/include/openssl/crypto.h:32:0,
                 from /usr/include/openssl/bio.h:20,
                 from /usr/include/openssl/conf.h:13,
                 from /home/pi/vzlogger/src/protocols/MeterOMS.cpp:13:
/usr/include/openssl/conf.h:92:1: note: declared here
 DEPRECATEDIN_1_1_0(void OPENSSL_config(const char *config_name))
 ^
cc1plus: all warnings being treated as errors
`

andig, 2017-08-20:
> When trying to compile commit d30f922, I still get this error:

Sieht aus wie openssl Versionsproblem. Stretch?

Viele Grüße,
Andreas

okrause, 2017-08-20:
https://www.heise.de/newsticker/meldung/Raspbian-Version-Stretch-auf-Basis-von-Debian-9-veroeffentlicht-3807824.html

okrause, 2017-08-20:
pi@volkszaehler:~ $ dpkg -l | grep ssl
ii  libgnutls-openssl27:armhf        3.5.8-5+deb9u2               armhf        GNU TLS library - OpenSSL wrapper
ii  libssl-dev:armhf                 1.1.0f-3                     armhf        Secure Sockets Layer toolkit - development files
ii  libssl-doc                       1.1.0f-3                     all          Secure Sockets Layer toolkit - development documentation
ii  libssl1.0.2:armhf                1.0.2l-2                     armhf        Secure Sockets Layer toolkit - shared libraries
ii  libssl1.1:armhf                  1.1.0f-3                     armhf        Secure Sockets Layer toolkit - shared libraries
ii  openssl                          1.1.0f-3                     armhf        Secure Sockets Layer toolkit - cryptographic utility

l3d00m, 2017-08-20:
Gerade durch Googeln gefunden, dass ihr hier auch schon über das Problem diskutiert. Ich hatte vorher schon #322, eröffnet und link das hiermit mal

okrause, 2017-08-20:
Ja, das isses. Hab grad das hier gefunden:

https://forum.nginx.org/read.php?29,265783,265783

Denke ein Fix fuer vzlogger waere dem nicht unähnlich.

okrause, 2017-08-20:
https://manpages.debian.org/testing/libssl-doc/OPENSSL_config.3ssl.en.html

sagt: 

> The OPENSSL_no_config() and OPENSSL_config() functions were deprecated in OpenSSL 1.1.0 by OPENSSL_init_crypto().
> 
https://www.openssl.org/docs/man1.1.0/crypto/OPENSSL_init_crypto.html