volkszaehler/vzlogger #319: Pull up/down resistors not configured for GPIO based S0-meters

kbabioch, 2017-06-27:
I'm currently looking into using vzlogger to log impulses from my S0 meters and the readings are jumping all over the place. Looking at the code and the GPIO configuration (see [here](https://github.com/volkszaehler/vzlogger/blob/master/src/protocols/MeterS0.cpp)), I realized that no pull-up or pull-down resistors are configured (on the Raspberry Pi). This is unfortunate, since it requires me to use external resistors, which is a PITA for many channels. Am I missing something, or how are you dealing with this?

All in all the code seems to be quite inflexible. Only active high with rising edges is supported, so it is not possible to reliable detect active low circuits :'(.

J-A-U, 2017-06-27:
> Am I missing something, or how are you dealing with this?

We recommend additional hardware to fit the S0 specifications. From these standard logic "1" is 15mA and higher. Impedance of most meters is about 1000 Ohm. You can't get a sure logic "1" with only 3,3V supply voltage. Even 5V often don't work safe.
[https://wiki.volkszaehler.org/hardware/controllers/raspberry_pi_erweiterung_mit_schaltausgaengen_rev.1](url)
[https://wiki.volkszaehler.org/hardware/controllers/raspberry_pi_erweiterung_klein](url)

J-A-U, 2017-06-28:
One other thing:
We already have the keys "Impulse" and "Impulse_neg" in parameter "Identifier" of channel-configuration. I thougth this selects the detection of rising/fallin edge...

mbehr1, 2017-06-28:
Nee. Impulses_neg ermöglicht über einen 2. GPIO sozusagen die Drehrichtung des Impulsgebers anzugeben. Also: dreht sich das Rad links oder rechtsherum ist anhand der reinen Impulse ja nicht zu unterscheiden. 

Gruß
Matthias 

Sent from a mobile device.

andig, 2017-07-05:
Die Benennung Impulses_neg ist vielleicht etwas unglücklich und resultiert aus der konkreten Physik meines Zählers der eben Bezug und Einspeisung kann wobei ein Pin die Richtung anzeigt und der andere Pin die Impulse liefert. Besser gewesen wäre vielleicht "Impulse_alt".

J-A-U, 2017-07-06:
Ich glaube nicht das es besser wäre. "alt" könnte als Alternativ interpretiert werden, das wäre noch unpräziser. Aber selbst alternierend wäre nicht ganz korrekt, bedeutet es doch eher periodisch oder wechselseitig als umgekehrt.

Für mich wäre interessanter was das für ein Zähler ist, bislang haben wir in der Zählerliste im Wiki noch kein Modell mit Richtungsausgang.
Und eine Beispielkonfig würde mich interessieren, hab nämlich noch keine rechte Vorstellung wie das praktisch aussehen soll.

andig, 2017-07-06:
> Ich glaube nicht das es besser wäre. "alt" könnte als Alternativ interpretiert werden, das wäre noch unpräziser. 

So hatte ich das gemeint. Es werden primäre und alternative Impulse erzeugt.

> Und eine Beispielkonfig wäre super

Im Moment mache ich das noch mit Python da das schon lief und ich von der Verfügbarkeit des APIs unabhängig bin. Anscheinend hat außer mir auch niemand so einen Zähler...

andig, 2017-10-03:
Can we close this PR and move discussions to ML?

closed 2018-03-05 by kbabioch