volkszaehler/vzlogger #322: Compilation error caused through deprecation under raspbian stretch

l3d00m, 2017-08-20:
I'm getting the following error if I'm running the install script under raspbian stretch.

```
[  9%] Building CXX object src/protocols/CMakeFiles/proto.dir/MeterOMS.cpp.o
cd /home/pi/vzlogger/build/src/protocols && /usr/bin/c++   -DHAVE_CONFIG_HPP -I/usr/local/include -I/home/pi/vzlogger/build -I/home/pi/vzlogger/include -I/home/pi/vzlogger  -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++11 -fpermissive -Wno-error=redundant-decls -Wno-ignored-qualifiers   -g3 -o CMakeFiles/proto.dir/MeterOMS.cpp.o -c /home/pi/vzlogger/src/protocols/MeterOMS.cpp
/home/pi/vzlogger/src/protocols/MeterOMS.cpp: In constructor ‘MeterOMS::MeterOMS(const std::__cxx11::list<Option>&, MeterOMS::OMSHWif*)’:
/home/pi/vzlogger/src/protocols/MeterOMS.cpp:145:2: error: ‘void OPENSSL_config(const char*)’ is deprecated [-Werror=deprecated-declarations]
  OPENSSL_config(NULL);
  ^~~~~~~~~~~~~~
In file included from /usr/include/openssl/crypto.h:32:0,
                 from /usr/include/openssl/bio.h:20,
                 from /usr/include/openssl/conf.h:13,
                 from /home/pi/vzlogger/src/protocols/MeterOMS.cpp:13:
/usr/include/openssl/conf.h:92:1: note: declared here
 DEPRECATEDIN_1_1_0(void OPENSSL_config(const char *config_name))
 ^
/home/pi/vzlogger/src/protocols/MeterOMS.cpp:145:21: error: ‘void OPENSSL_config(const char*)’ is deprecated [-Werror=deprecated-declarations]
  OPENSSL_config(NULL);
                     ^
In file included from /usr/include/openssl/crypto.h:32:0,
                 from /usr/include/openssl/bio.h:20,
                 from /usr/include/openssl/conf.h:13,
                 from /home/pi/vzlogger/src/protocols/MeterOMS.cpp:13:
/usr/include/openssl/conf.h:92:1: note: declared here
 DEPRECATEDIN_1_1_0(void OPENSSL_config(const char *config_name))
 ^
cc1plus: all warnings being treated as errors
src/protocols/CMakeFiles/proto.dir/build.make:257: recipe for target 'src/protocols/CMakeFiles/proto.dir/MeterOMS.cpp.o' failed
make[2]: *** [src/protocols/CMakeFiles/proto.dir/MeterOMS.cpp.o] Error 1
make[2]: Leaving directory '/home/pi/vzlogger/build'
CMakeFiles/Makefile2:1172: recipe for target 'src/protocols/CMakeFiles/proto.dir/all' failed
make[1]: *** [src/protocols/CMakeFiles/proto.dir/all] Error 2
make[1]: Leaving directory '/home/pi/vzlogger/build'
Makefile:163: recipe for target 'all' failed
make: *** [all] Error 2
```

c++ version

```
pi@raspberrypi:~/vzlogger $ LANG=C c++ --version
c++ (Raspbian 6.3.0-18+rpi1) 6.3.0 20170516
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

mbehr1, 2017-08-20:
hmm. we need to adjust vzlogger for new openssl. Can you create an issue in github?

> I'm getting the following error if I'm running the install script under raspbian stretch.
>
> [  9%] Building CXX object src/protocols/CMakeFiles/proto.dir/MeterOMS.cpp.o
> cd [...]

Gruß

Matthias

l3d00m, 2017-08-20:
Ist doch ein issue, oder hab ich da was falsch? @mbehr1

andig, 2017-08-20:
> Ist doch ein issue, oder hab ich da was falsch? @mbehr1

Ja, ist es. Matthias hatte Dir vmtl. vorher schon per Mail geantwortet und das nicht gesehen.

mbehr1, 2017-08-20:
Ja, sorry. Siehe Fix in #323. Wäre gut, wenn du ( @l3d00m ) den schnell testen könntest.

mbehr1, 2017-08-20:
Hab den Fix noch mal geändert. Weiß zwar nicht, ob irgendjemand open ssl cfg files nutzt/nutzen muss, aber so sollte das Verhalten wie vorher sein.

l3d00m, 2017-08-20:
Heute schaff ich das leider nicht mehr, mache ich morgen früh direkt. Danke dir für die Mühe @mbehr1

PR #323 (2017-08-24): fix compilation with openssl 1.1
Should fix #322 . According to OPENSLL docs OPENSSL_config is no longer needed with v1.1.
Untested!
closed 2017-08-24 by andig