volkszaehler/vzlogger #324: Stretch: build fails without libmicrohttp

andig, 2017-08-29:
Compiling on fresh raspbian stretch gives:

	CMake Warning at CMakeLists.txt:219 (message):
	  microhttpd was not found.  Ignored ENABLE_LOCAL!

	  Install microhttpd or call cmake
	  -DMICROHTTPD_HOME=path_to_microhttpd_install

	CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
	Please set them or make sure they are set and tested correctly in the CMake files:
	MICROHTTPD_LIBRARY (ADVANCED)
	    linked by target "mock_metermap" in directory /home/pi/vzlogger/tests/mocks

Since `Ignored ENABLE_LOCAL!` the compile should imho continue but it fails in the tests due to missing but ignored library. 

@mbehr1 Would is be possible to make the tests as flexible as the compile itself if the lib is missing?

l3d00m, 2017-09-24:
related to this maybe? https://github.com/volkszaehler/vzlogger/pull/325

Quite a noob concerning packages, but I think you're just missing the `d`? Saw this issue after opening the PR unfortunately

andig, 2017-09-26:
Nope, the pr only concerned the readme.

mbehr1, 2017-10-03:
I'll provide a fix for that.

mbehr1, 2017-10-03:
See fix with #334. Can you test whether it works now? There might be some more instances of wrong linkage...

andig, 2017-10-03:
Will check.

mbehr1, 2018-08-05:
Can this be closed now?

andig, 2018-08-05:
Wir machen es einfach mal zu, danke 😁

closed 2018-08-05 by andig