volkszaehler/vzlogger #329: Question abouit Unit Tests in cross-compile environment

oe5hpm, 2017-09-26:
U build the vzlogger with the buildroot toolchain and i have troubles there with the unit tests.

Actually i simple disable these tests there with setting
'set(ENABLE_GOOGLEMOCK FALSE)' within CMakeLists.txt

I'm not sure if this is the 100% right way.
Fact is, that we cannot run the produced binary on the build host.

Would it be an idea for having some 'if cross-compile' within the CMakeLists for switching off the unit tests ?

cheers,
Hannes

oe5hpm, 2017-09-26:
buildroot today passes these arguments per default to the cmake build:

DCMAKE_COLOR_MAKEFILE=OFF -DBUILD_DOC=OFF -DBUILD_DOCS=OFF -DBUILD_EXAMPLE=OFF -DBUILD_EXAMPLES=OFF -DBUILD_TEST=OFF -DBUILD_TESTS=OFF -DBUILD_TESTING=OFF -DBUILD_SHARED_LIBS=ON

maybe we can hook on something there ... 

 -DBUILD_TEST=OFF

might be an idea.

cheers,
Hannes

andig, 2017-09-26:
For "how to" questions please contact the developers mailing list.

oe5hpm, 2017-09-26:
moving that to the mailinglist

closed 2017-09-26 by oe5hpm