volkszaehler/vzlogger #331: Please add a license exemption to use vzlogger with openssl

Ampelbein, 2017-10-03:
Hi,

I'm in the process of packaging vzlogger for Debian and the last remaining blocker is the licensing of vzlogger. The GPL is considered to be incompatible with the OpenSSL License (see https://people.gnome.org/~markmc/openssl-and-the-gpl.html for an explanation).

One way to make the licenses compatible would be to add a exemption to the vzlogger license, allowing it to be linked with OpenSSL, like: "This program is released under the GPL with the additional exemption that compiling, linking, and/or using OpenSSL is allowed." (Text taken from https://www.openssl.org/docs/faq.html#LEGAL2)

If all relevant copyright holders agree, would you please consider adding this exemption?

Thanks.

andig, 2017-10-03:
We'd need to get the list of all committers? Is consent required or just not objecting?

Ampelbein, 2017-10-03:
I'm not a copyright law expert, but yes, getting all committers to consent would be the best thing. But I guess it is enough if the "major" contributors agree and those with "minor" code changes don't object.

mbehr1, 2017-10-03:
we could change to use gnu tls instead. Might be easier/safer.

> 
> I'm definitely not a lawyer, but I'm remembering the amount of effort it
> took for the bootstrap css framework team to move from GPLv2->MIT.
> You will need the explicit agreement from each and every contributor and
> rewrite the contributions of everyone who either doesn't agree or not
> respond.
> License changes are hard without a CLA...

Gruß

Matthias

Ampelbein, 2017-10-18:
You are right, it might be easier to convert to gnutls. I have started a branch that uses GnuTLS instead of OpenSSL at https://github.com/Ampelbein/vzlogger/tree/gnutls - though I don't yet have a mySmartGrid account to test it. Once I (or anyone else) have verified that it still works with the GnuTLS library, I'll open a PR.

andig, 2017-10-22:
> You are right, it might be easier to convert to gnutls.

https://news.ycombinator.com/item?id=7347500:

> The annoying thing about GnuTLS is that it normally might not be very widely used, except that the Debian project initiated a huge push to make software linkable with GnuTLS instead of OpenSSL, because of issues with the OpenSSL license[1]. So if you're a Debian or Ubuntu user, you're probably relying on GnuTLS a lot more than users of any other distribution, or people who compile the upstream sources themselves. (Not that OpenSSL is a panacea, but at least it gets more attention than GnuTLS).

> One way to make the licenses compatible would be to add a exemption to the vzlogger license, allowing it to be linked with OpenSSL

Thinking about this: why not try? Our committers list is not _that_ long?

andig, 2017-10-22:
Pinging all committers to consent to adding an OpenSSL license exception:

> "This program is released under the GPL with the additional exemption that compiling, linking, and/or using OpenSSL is allowed."

Please confirm below, this comment tracks the signatures.

  - [ ] @mbehr1
  - [ ] @kaikrueger
  - [ ] @stv0g
  - [ ] @andig
  - [ ] @justinotherguy
  - [ ] @r00t-
  - [ ] @peterevertz
  - [ ] @J-A-U
  - [ ] @flyingflo
  - [ ] @nuccleon
  - [ ] @gitka
  - [ ] @okrause
  - [ ] @Stefan-Code
  - [ ] @asdil12
  - [ ] @schnello
  - [ ] @griesi
  - [ ] @homagnussen
  - [ ] @l3d00m
  - [ ] @Ampelbein
  - [ ] @cmorty
  - [ ] @zobelhelas
  - [ ] @amenk
  - [ ] @JT-DE
  - [ ] @mwulz

amenk, 2017-10-22:
Confirm

Stefan-Code, 2017-10-22:
:+1:
> "This program is released under the GPL with the additional exemption that compiling, linking, and/or using OpenSSL is allowed."

l3d00m, 2017-10-22:
confirm

Ampelbein, 2017-10-22:
On Sun, Oct 22, 2017 at 11:51:12AM -0700, andig wrote:
> Pinging all committers to consent to adding an OpenSSL license exception:
> 
> > "This program is released under the GPL with the additional exemption that compiling, linking, and/or using OpenSSL is allowed."

confirm

-- 
PGP-encrypted mails preferred
PGP Fingerprint: 74CD D9FE 5BCB FE0D 13EE 8EEA 61F3 4426 74DE 6624

flyingflo, 2017-10-22:
Confirm

andig, 2017-10-22:
Stark- ihr seid _richtig_ schnell, vielen Dank :)

asdil12, 2017-10-22:
Confirm

JT-DE, 2017-10-22:
- @JT-DE <https://github.com/jt-de>

zobelhelas, 2017-10-22:
Confirm. Works for me!

schnello, 2017-10-22:
confirmed

J-A-U, 2017-10-23:
> "This program is released under the GPL with the additional exemption that compiling, linking, and/or using OpenSSL is allowed."

:+1:

griesi, 2017-10-23:
confirm

justinotherguy, 2017-10-23:
confirm

stv0g, 2017-10-23:
confirm

nuccleon, 2017-10-23:
confirm

okrause, 2017-10-23:
confirm

mbehr1, 2017-10-23:
confirm

> Pinging all committers to consent to adding an OpenSSL license exception:
>
> "This program is released under the GPL with the additional exemption that compiling, linking, and/or using OpenSSL is [...]

Gruß

Matthias

gitka, 2017-10-23:
confirm

J-A-U, 2017-12-22:
*push*
@cmorty
@homagnussen
@kaikrueger
@mwulz
@peterevertz

cmorty, 2017-12-22:
confirm

That's what you get for making drive-by contributions.... :)

andig, 2017-12-22:
> That's what you get for making drive-by contributions.... :)

Yep, then they push you :) *thank you and Merry Christmas*

peterevertz, 2017-12-22:
confirm

Ampelbein, 2018-01-22:
Apologies for being absent for a while, I started a new job and had to deal with moving to a different city, @homagnussen
@kaikrueger
@mwulz
@peterevertz
Is there anything I can do to convince you to provide a licensing exemption? Do you have any concerns about it?

peterevertz, 2018-01-22:
I already confirmed in Dezember ...

r00t-, 2019-05-29:
obviously the remaining authors are not reading their github notifications...

@mwulz has only one trivial commit, maybe we can just remove that?
https://github.com/volkszaehler/vzlogger/commits?author=mwulz
has not been active on github since 2013:
https://github.com/mwulz?tab=overview&from=2013-12-01&to=2013-12-31
he was active on vz-users until 2015: https://marc.info/?a=135884082100001

@homagnussen only did "Use inotify for MeterFile when interval is empty (#335)"
https://github.com/volkszaehler/vzlogger/commits?author=homagnussen
and only ever used his github account for that one commit.
https://github.com/homagnussen?tab=overview&from=2017-09-01&to=2017-09-30
there is a vz-dev thread that has his email address:
https://marc.info/?l=volkszaehler-dev&m=150568262213408

but @kaikrueger did the C++ port...
https://github.com/volkszaehler/vzlogger/commits?author=kaikrueger
https://marc.info/?l=volkszaehler-dev&m=134060505102743

i contacted all three of them via email at the addresses they used on the mailing lists,
will report results (ideally they will respond here themselves.)

kaikrueger, 2019-05-29:
Confirm

narc-Ontakac2, 2023-01-03:
This is really sad. @homagnussen did not confirm (his commit is not trivial) and the gnutls change was also not done. So nothing happened.

narc-Ontakac2, 2023-01-03:
I have sent another email to @homagnussen explaining the situation.

r00t-, 2023-01-03:
@narc-Ontakac2:
i don't think the feature that @homagnussen contributed is critical or widely used,
so if that's the final blocker, we can just revert it.
(and possibly re-implement later.)

narc-Ontakac2, 2023-01-03:
It is at least not easily doable:
`jo@pause:~/projects/debian/vzlogger$ git revert 5342f0b096b13424c25327881bfd2b0537ecc180
automatischer Merge von include/protocols/MeterFile.hpp
KONFLIKT (Inhalt): Merge-Konflikt in include/protocols/MeterFile.hpp
automatischer Merge von src/Meter.cpp
KONFLIKT (Inhalt): Merge-Konflikt in src/Meter.cpp
automatischer Merge von src/protocols/MeterFile.cpp
KONFLIKT (Inhalt): Merge-Konflikt in src/protocols/MeterFile.cpp
automatischer Merge von tests/ut_MeterFile.cpp
KONFLIKT (Inhalt): Merge-Konflikt in tests/ut_MeterFile.cpp
Fehler: Konnte "revert" nicht auf 5342f0b... (Use inotify for MeterFile when interval is empty (#335)) ausführen
`

narc-Ontakac2, 2023-01-03:
There are also new contributors since 2017-10-03. How was the list with the checkboxes created?

narc-Ontakac2, 2023-01-03:
Adding a USE_OPENSSL option with gnutls as fallback is relatively easy. MySmartGrid.cpp could be adapted this afternoon (not yet tested, compiles and looks right). The only other place where openssl is used is in MeterOMS.cpp and requires libmbus.  So  -DUSE_OPENSSL=off -DENABLE_OMS=off should give a debian packageable build.

narc-Ontakac2, 2023-01-04:
Actually parts of the code are copyrighted by Fraunhofer ITWM, if I can trust the headers:

`$ grep -R Fraunhofer src/ include/
src/MeterMap.cpp: * (C) Fraunhofer ITWM
src/api/MySmartGrid.cpp: * (C) Fraunhofer ITWM
src/api/Volkszaehler.cpp: * (C) Fraunhofer ITWM
include/VZException.hpp: * Copyright (c) 2011 Fraunhofer ITWM
include/MeterMap.hpp: * (C) Fraunhofer ITWM
include/shared_ptr.hpp: * Copyright (c) 2011 Fraunhofer ITWM
include/api/Volkszaehler.hpp: * (C) Fraunhofer ITWM
include/api/MySmartGrid.hpp: * (C) Fraunhofer ITWM
include/ApiIF.hpp: * (C) Fraunhofer ITWM
`
This complicates the situation further. I think that gnutls is the way to go.

r00t-, 2023-01-04:
@kaikrueger added those headers (most or all of them) here:
https://github.com/volkszaehler/vzlogger/commit/a39e2ab2329011bccb19f9980c49f229a5e3f289
i don't think this is a big issue, but we might need additional confirmation from him that his employer is agreeing, or that he added the headers accidentially.

narc-Ontakac2, 2023-01-04:
@kaikrueger In case these headers have been added accidentally please remove them.

Gnutls looks however quite charming:
` ` `
void hmac_sha1(char *digest, const unsigned char *data, size_t dataLen,
			   const unsigned char *secretKey, size_t secretLen) {
	// compile time digest size for HMAC-SHA1 
	const unsigned int len = 20;
	unsigned char out[len];
	gnutls_hmac_fast(GNUTLS_MAC_SHA1,
					 secretKey, secretLen,
					 data, dataLen,
					 out);

	size_t ret_len = 2 * len + 1; 
	char ret[ret_len];
	const gnutls_datum_t d_out = {out, len};
	gnutls_hex_encode(&d_out, ret, &ret_len);

	snprintf(digest, 255 /*sizeof(digest)*/, "X-Digest: %s", ret);
}
```
So I currently think this is a good way to go.

narc-Ontakac2, 2023-01-04:
The good news is that this is a temporary problem that is just going away. [OpenSSL 3 is licensed Apache v2](https://www.openssl.org/source/license.html).

narc-Ontakac2, 2023-01-08:
We now also have an answer from @homagnussen . Since mbox format is not supported I'll fall back to copy & paste:
Hallo Joachim,

sorry wegen der späten Rückmeldung, du warst in meinem Spam-Ordner 
gelandet. Aber der Anruf bei meiner Frau hat geholfen ;-)

Natürlich bin ich einverstanden!

-holger

On 03.01.23 08:55, Joachim Zobel wrote:
> Hi.
>
> You contributed to vzlogger years ago. Since the vzlogger project is
> doing a small change in its licensing we would appreciate if you would
> confirm you agree with the following change:
>
> "This program is released under the GPL with the additional exemption
> that compiling, linking, and/or using OpenSSL is allowed."
>
> See
>
> https://github.com/volkszaehler/vzlogger/issues/331
>
> Since you are the last one missing this would be really helpful. In
> case you do not have access to your github account any more an answer
> via email should do.
>
> Sincerely,
> Joachim

r00t-, 2023-01-08:
> There are also new contributors since 2017-10-03. 
> How was the list with the checkboxes created?
if this is not obvious yet: it's simply some emojis, and copy/paste from the list of commiters,
this can even be filtered by date:
https://github.com/volkszaehler/vzlogger/graphs/contributors?from=2017-10-03&to=2023-01-31&type=c

narc-Ontakac2, 2023-07-12:
Since this is a nonissue with openssl 3 I am closing this.

closed 2023-07-12 by narc-Ontakac2