volkszaehler/vzlogger #333: Disable tests/google mock in default build?

andig, 2017-10-03:
@mbehr1 I'm not sure about best practices but I've noticed that compiling vzlogger seems to spend large parts of its time in the tests. Is it necessary to compile those by default or would it be sufficient to do so for travis?

mbehr1, 2017-10-03:
We could change default to be disabled. I'd prefer that whoever changes the code at least runs the provided tests locally as well. But I don't really mind if we change this and rely on travis.

andig, 2017-10-03:
> We could change default to be disabled. I'd prefer that whoever changes the code at least runs the provided tests locally as well.

Would another make target for the tests be sufficient for that?

narc-Ontakac2, 2023-01-14:
This is partially resolved by #560. Without downloading and compiling googletest the time for the tests alone is IMHO acceptable.