volkszaehler/vzlogger #337: "Cannot allocate memory" wenn "push" aktiv

M0ebiu5, 2017-10-08:
vzlogger läuft bei mir auf einem Orange Pi Zero unter Ubuntu 16.04.3. Installiert habe ich über github und das install Script. 

Alles läuft super, wenn ich nur die influxdb api verwende - wenn ich aber die "push" funktion aktiviere, crasht vzlogger. Im Log kommt der fehler "Cannot allocate memory".

J-A-U, 2017-10-08:
Ein paar Details könnten hilfreich sein: Wohin pushst du, wie sieht die Config aus?
Und der Abschnitt aus dem Log bitte um die zeitlich Abfolge etwas einordnen zu können.

M0ebiu5, 2017-10-08:
Push auf node-red, dort kommt aber nichts an. Habe die config schon nur auf das "push" reduziert - ohne Erfolg.

**Log:**
[Oct 08 15:20:02][mtr0] Got 4 new readings from meter:
[Oct 08 15:20:02][mtr0] Reading: id=255-255:1.8.0*255/ObisIdentifier:255-255:1.8.0*255 value=2664328.00 ts=1507468802000
[Oct 08 15:20:02][mtr0] Reading: id=255-255:2.8.0*255/ObisIdentifier:255-255:2.8.0*255 value=5125569.00 ts=1507468802000
[Oct 08 15:20:02][mtr0] Reading: id=255-255:1.7.0*255/ObisIdentifier:255-255:1.7.0*255 value=195.00 ts=1507468802000
[Oct 08 15:20:02][mtr0] Reading: id=255-255:2.7.0*255/ObisIdentifier:255-255:2.7.0*255 value=7.00 ts=1507468802000
[Oct 08 15:20:02][chn0] Adding reading to queue (value=195.00 ts=1507468802000)
[Oct 08 15:20:02][push] added to uuid 01234567-9abc-def0-1234-56789abcdeff
[Oct 08 15:20:02][chn1] Adding reading to queue (value=7.00 ts=1507468802000)
[Oct 08 15:20:02][push] added to uuid 01234567-9abc-def0-1234-56789abcdfff
[Oct 08 15:20:02][push] push: { "data": [ { "uuid": "01234567-9abc-def0-1234-56789abcdeff", "tuples": [ [ 1507468802000, 195 ] ] } ] }
[Oct 08 15:20:02][chn2] Adding reading to queue (value=2664328.00 ts=1507468802000)
[Oct 08 15:20:02][push] added to uuid 01234567-9abc-def0-1234-56789abcdaaa
[Oct 08 15:20:02][chn3] Adding reading to queue (value=5125569.00 ts=1507468802000)
[Oct 08 15:20:02][push] added to uuid 01234567-9abc-def0-1234-56789abcdfaa
[Oct 08 15:20:02]       Cannot allocate memory
 

**vzlogger.conf:**
`/**
 * vzlogger configuration example for InfluxDB
 *
 * use proper encoded JSON with javascript comments
 *
 * take a look at the wiki for detailed information:
 
*/ 

  {
	"retry": 30,
	"daemon": true,
	"log": "/tmp/vzlogger.log",
	"push": [{ "url": "http://192.168.4.221:1880/vzpush"}],
    "local": {
	"enabled": false,
	"port": 8080,
	"index": true,
	"timeout": 30,
	"buffer": -1
    },
    // ... for general vzlogger settings see vzlogger.conf
    "meters": 
[
        // examples for InfluxDB as storage
        {
            // See vzlogger.conf for complete meter configuration options
	"enabled" : true,
	"skip": false,
	"protocol": "oms",
	"device": "/dev/ttyUSB0",
	"baudrate": 9600, // optional default 9600
	"aggtime": 10,
	"mbus_debug": false,
	"key": "0287945FDB4CD804E926810A9F2D8126", // AES key in hex without spaces. Needs to be exactly 32 chars"mbus_debug": false, 
					//optional provide additional debug output from libmbus on the console/stderr/stdout
            "channels": [{
                "uuid": "01234567-9abc-def0-1234-56789abcdeff", // use the uuid command to generate this
                "identifier" : "1.7.0", // OBIS code
		"middleware": "http://127.0.0.1/middleware.php",
		"api": "influxdb",
		"host": "192.168.4.221:8086",
		"database": "infra",
		"measurement_name": "strom",
		"duplicates": 10,
		"aggmode": "avg"
            },{
                "uuid": "01234567-9abc-def0-1234-56789abcdfff", // use the uuid$
                "identifier" : "2.7.0", // OBIS code
                "middleware": "http://127.0.0.1/middleware.php",
                "api": "influxdb",
                "host": "192.168.4.221:8086",
                "database": "infra",
                "measurement_name": "strom",
                "duplicates": 10,
                "aggmode": "avg"
            },{
                "uuid": "01234567-9abc-def0-1234-56789abcdaaa", // use the uuid command to generate this
                "identifier" : "1.8.0", // OBIS code
		"middleware": "http://127.0.0.1/middleware.php",
		"api": "influxdb",
		"host": "192.168.4.221:8086",
		"database": "infra",
		"measurement_name": "strom"
            },{
                "uuid": "01234567-9abc-def0-1234-56789abcdfaa", // use the uuid$
                "identifier" : "2.8.0", // OBIS code
                "middleware": "http://127.0.0.1/middleware.php",
                "api": "influxdb",
                "host": "192.168.4.221:8086",
                "database": "infra",
                "measurement_name": "strom"
        }]
	} 
]
} `

J-A-U, 2017-10-08:
> Habe die config schon nur auf das "push" reduziert - ohne Erfolg.

Es werden nur die Daten gepusht für die ein channel deklariert wurde. Wenn du diese nur pushen möchtest  muss 
`"api": "none", ` 
parametriert werden.

Das hilft dir aber sicherlich nur den push testweise in Betrieb zu nehmen, ich gehe davon aus das du auch weiterhin Daten in influxdb haben möchtest?

M0ebiu5, 2017-10-08:
nö, das wäre schon ok - kann die Daten in node-red gleich für mqtt und influxdb nehmen.

Auch mit "api": "none" bekomme ich den Fehler (hab jetzt in der config alles von "influxdb" auf "none" umgestellt).

**Log:**
[Oct 08 15:51:02][mtr0] Got 4 new readings from meter:
[Oct 08 15:51:02][mtr0] Reading: id=255-255:1.8.0*255/ObisIdentifier:255-255:1.8.0*255 value=2664437.00 ts=1507470662000
[Oct 08 15:51:02][mtr0] Reading: id=255-255:2.8.0*255/ObisIdentifier:255-255:2.8.0*255 value=5125569.00 ts=1507470662000
[Oct 08 15:51:02][mtr0] Reading: id=255-255:1.7.0*255/ObisIdentifier:255-255:1.7.0*255 value=292.00 ts=1507470662000
[Oct 08 15:51:02][mtr0] Reading: id=255-255:2.7.0*255/ObisIdentifier:255-255:2.7.0*255 value=0.00 ts=1507470662000
[Oct 08 15:51:02][chn0] Adding reading to queue (value=292.00 ts=1507470662000)
[Oct 08 15:51:02][push] added to uuid 01234567-9abc-def0-1234-56789abcdeff
[Oct 08 15:51:02][chn1] Adding reading to queue (value=0.00 ts=1507470662000)
[Oct 08 15:51:02][push] added to uuid 01234567-9abc-def0-1234-56789abcdfff
[Oct 08 15:51:02][push] push: { "data": [ { "uuid": "01234567-9abc-def0-1234-56789abcdeff", "tuples": [ [ 1507470662000, 292 ] ] } ] }
[Oct 08 15:51:02][chn2] Adding reading to queue (value=2664437.00 ts=1507470662000)
[Oct 08 15:51:02][push] added to uuid 01234567-9abc-def0-1234-56789abcdaaa
[Oct 08 15:51:02][chn3] Adding reading to queue (value=5125569.00 ts=1507470662000)
[Oct 08 15:51:02][push] added to uuid 01234567-9abc-def0-1234-56789abcdfaa
[Oct 08 15:51:02]       Cannot allocate memory

J-A-U, 2017-10-08:
Ok, danke. Damit ist leider auch der Punkt erreicht ab dem ich dir nicht mehr weiterhelfen kann. Da muss jemand mit Tiefenkenntnis nachschauen was falsch läuft...

M0ebiu5, 2017-10-08:
Hab mir das log nochmal angeschaut - da schreibt er das "none" keine bekannte api ist...hast du "null" gemeint?
Leider kommt nach Umstellung auf "api":"null" aber der selbe Fehler...

mbehr1, 2017-10-08:
kommt der Fehler sofort oder nach paar Minuten?

Mit freundlichen Grüßen / Best Regards
Matthias Behr

Sent from a mobile device.

M0ebiu5, 2017-10-08:
Sofort - es kommt nichts am Zielrechner an.

J-A-U, 2017-10-08:
> hast du "null" gemeint?

Hoppla, Fehler in der Doku. :/

mbehr1, 2017-10-08:
kannst du mal folgende Änderung anwenden und erneut testen?

bug337 <https://github.com/mbehr1/vzlogger/tree/bug337> 

https://github.com/mbehr1/vzlogger/commit/18c0eca6f630d21319cf8912f10c84300f17dd3a <https://github.com/mbehr1/vzlogger/commit/18c0eca6f630d21319cf8912f10c84300f17dd3a> 

> 
> Sofort - es kommt nichts am Zielrechner an.

Gruß

Matthias

M0ebiu5, 2017-10-08:
Hi Matthias,
hier das log:

[Oct 08 22:00:02][push] push: { "data": [ { "uuid": "01234567-9abc-def0-1234-56789abcdeff", "tuples": [ [ 1507492802000, 463 ] ] } ] }
[Oct 08 22:00:02][push] added to uuid 01234567-9abc-def0-1234-56789abcdaaa
[Oct 08 22:00:02][chn3] Adding reading to queue (value=5125569.00 ts=1507492802000)
[Oct 08 22:00:02][push] added to uuid 01234567-9abc-def0-1234-56789abcdfaa
[Oct 08 22:00:02][push] Cannot allocate memory response->size=-1092164904 realsize=146

mbehr1, 2017-10-08:
hmm. kannst du den noch mal versuchen:
more debug infos in case of no mem <https://github.com/mbehr1/vzlogger/commit/18c0eca6f630d21319cf8912f10c84300f17dd3a>

https://github.com/mbehr1/vzlogger/commit/18c0eca6f630d21319cf8912f10c84300f17dd3a <https://github.com/mbehr1/vzlogger/commit/18c0eca6f630d21319cf8912f10c84300f17dd3a> 

> Hi Matthias,
> hier das log:
>
> [Oct 08 22:00:02][push] push: { "data": [ { "uuid": "01234567-9abc-def0-1234-56789abcdeff", "tuples": [ [ 1507492802000, 463 ] ] } ] }
> [Oct 08 22:00:02][push] added to [...]

Gruß

Matthias

mbehr1, 2017-10-08:
oder nimm gleich den: https://github.com/mbehr1/vzlogger/commit/e587a8df9108c85c15e5bb0220262364f1cc27aa <https://github.com/mbehr1/vzlogger/commit/e587a8df9108c85c15e5bb0220262364f1cc27aa>

size wurde nicht initialisiert. 

> 
> hmm. kannst du den noch mal versuchen:
> more debug infos in case of no mem <https://github.com/mbehr1/vzlogger/commit/18c0eca6f630d21319cf8912f10c84300f17dd3a>
> 
> https://github.com/mbehr1/vzlogger/commit/18c0eca6f630d21319cf8912f10c84300f17dd3a <https://github.com/mbehr1/vzlogger/commit/18c0eca6f630d21319cf8912f10c84300f17dd3a> 
> 
>> 
>> Hi Matthias,
>> hier das log:
>> 
>> [Oct 08 22:00:02][push] push: { "data": [ { "uuid": "01234567-9abc-def0-1234-56789abcdeff", "tuples": [ [ 1507492802000, 463 ] ] } ] }
>> [Oct 08 22:00:02][push] added to uuid 01234567-9abc-def0-1234-56789abcdaaa
>> [Oct 08 22:00:02][chn3] Adding reading to queue (value=5125569.00 ts=1507492802000)
>> [Oct 08 22:00:02][push] added to uuid 01234567-9abc-def0-1234-56789abcdfaa
>> [Oct 08 22:00:02][push] Cannot allocate memory response->size=-1092164904 realsize=146
>> 
>> 
> 
> Gruß
> 
> Matthias

Gruß

Matthias

mbehr1, 2017-10-08:
siehe #338

M0ebiu5, 2017-10-08:
Hi Matthias,
das push funktioniert jetzt - debug output gibts deshalb keinen :)
Übrigens läuft push und influxdb hier parallel...
Danke für die schnelle Hilfe!
LG
Walter

PR #338 (2017-10-09): init size and more debug output for PushData
should fix #337
closed 2017-10-09 by andig