volkszaehler/vzlogger #342: YPort+ binary

DennisVonDerBey, 2017-11-07:
Hey guys,
I just received my YPort+ with 2 IR readers and I just don't get it right.
I am running a Pi with OpenHab and one with multiple services including influxdb + Grafana.

The best way to integrate the readers into my architecture would be:

Readers 1&2 to YPort+ -> Socat proxy -> Socat mount of the serial devices on the OpenHab Pi -> OpenHab binding accessing the serial port -> persistence through OpenHab with influxdb

But this one won't work because I cannot use socat on the current distributed OpenWRT version. It just says "socat not found" even though it's in the PATH. I found an issue which said it would need to be recompiled due to a change in the C environment.

Then I tried directly injecting into influxdb on my other Pi with VZlogger but the version of the binary is too old and the influxdb API isn't included in it.

Searching for these issues is really damn hard because of old mailing list systems and no real centralized forum so I thought I'd try here. Is there any place where current binaries of this tool for OpenWRT are downloadable? Or maybe a tutorial on how to compile it for the YPort+ environment?

Thanks for your help.

J-A-U, 2017-11-08:
Hello,

> It just says "socat not found" even though it's in the PATH.

chmod +x?

> Then I tried directly injecting into influxdb on my other Pi with VZlogger but the version of the binary is too old and the influxdb API isn't included in it.

Just a few days earlier somebody (forum) asked for yport and i mentioned this problem.

> Searching for these issues is really damn hard because of old mailing list systems

[Full-Text search](https://marc.info/?l=volkszaehler-dev)

> so I thought I'd try here. 

Our hardware-guy is not very much into github, you would have a better chance for support in the mailinglist. Sorry.

> Or maybe a tutorial on how to compile it for the YPort+ environment?

Nope, but what i found (read the full thread):
[Yportnode](https://marc.info/?l=volkszaehler-dev&m=140197582023026&w=2)

mfg JAU

DennisVonDerBey, 2017-11-08:
> chmod +x?

The binary is already executable.

> Just a few days earlier somebody (forum) asked for yport and i mentioned this problem.

Sorry but which forum do you mean?

> Nope, but what i found (read the full thread):
> Yportnode

Thanks for the link. This guy tries to compile his own firmware but there are a few interesting attachments I will look into.

Thanks for your fast help!

closed 2020-03-26 by andig