volkszaehler/vzlogger #351: wrong timestamp when inserting in influxdb

artcrime, 2018-03-19:
Hello,
 I just updated vzlogger to 0.61 to use the influxdb api. Metering data is addedinto influx db but the relative values have a wrong timestamp:

[Mar 19 13:54:24][mtr2] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=5073278.90 ts=1521464064000
[Mar 19 13:54:24][mtr2] Reading: id=1-0:16.7.0*255/ObisIdentifier:1-0:16.7.0*255 value=380.40 ts=45180567000
[Mar 19 13:54:24][chn4] Adding reading to queue (value=5073278.90 ts=1521464064000)
**[Mar 19 13:54:24][chn5] Adding reading to queue (value=380.40 ts=45180567000)
[Mar 19 13:54:24][chn6] Adding reading to queue (value=380.40 ts=45180567000)**
[Mar 19 13:54:24][chn7] Adding reading to queue (value=5073278.90 ts=1521464064000)

It is someday in july 1971 and not today. Is this a bug or your intention?
Although all collected values are written to influxdb and they are not aggregated as configured in the channel. Is this behaviour a misconfiguration for the channel?

frankrichter, 2018-03-19:
The wrong timestamp is read from your meters (SML type?) internal clock which is not set correctly.
Use config option use_local_time as workaround.

Please use the mailing list for further user questions.

andig, 2018-03-19:
I’ve had the same problem with some MME meters. Funny enough it was sending different timestamps per OBIS code…

`use_local_time` does the trick.

Cheers,
Andi

artcrime, 2018-03-21:
Yes it is SML type and the workaround workes for me. Intersting thet the same meter delivers two different timestamps. It is a EMH ED300S meter delivered from EWE.

andig, 2018-03-21:
Solution is `use_local_time`. Looks like a bug in the meter.

artcrime, 2018-03-21:
But I seem to get right timestamps when writing the data to mysql via the middleware. So the time stamps are generated by the middleware in case of  mysql. Why do you write explicit timestamps when data is written to influxdb? If you dont send a timestamp it will be generated by influxdb like if you would use the middleware.

andig, 2018-03-21:
Sorry, I misunderstood. So you‘re using the option and still vet wrong timestamps with influx?

andig, 2018-03-22:
ping @mbehr1 could you check where the timestamps are generated?

closed 2018-06-13 by andig