volkszaehler/vzlogger #352: Add support for user count

kbabioch, 2018-03-27:
I'm currently logging how much users are at home and want to visualize this along with the other graphs (water, power, etc.). I'm getting this data from a script (based on the network devices currently loggined into the WiFi, etc.), but I'm not sure what counter type suits best for such a channel. There is a "user" type, but this seems to be only an aggregate, which cannot have any values attached to it. Basically I need "gauge" kind of type, which is an absolute value and can be incremented / decremented from 0 to x.

frankrichter, 2018-03-27:
That's not at all related to vzlogger. Please use the mailing list for this
kind of questions.

You can use channel type "Sensor".

andig, 2018-06-13:
Off-topic

closed 2018-06-13 by andig