volkszaehler/vzlogger #356: Different aggtimes per channels

Gitsaibot, 2018-06-11:
Ich möchte den Istwert und den Zählerwert mit unterschiedlichen timings abfragen um die Datenbank zu entlasten. Das scheint aber nicht möglich zu sein da ich pro meter nur eine aggtime festlegen kann!? Den Istwert würde ich gerne im Intervall von 5sec erfassen und den Zählerstand nur einmal am Tag. 

> "enabled": true,               // disabled meters will be ignored (default)
>  "skip": false,                  // errors when opening meter may be ignored if enabled
>  "protocol": "sml",              // meter protocol, see 'vzlogger -h' for full list
>  "device": "/dev/ttyAMA0",       // meter device
>  "aggtime": 5,            

> > 
> >             "channels": [{
> >                "api": "volkszaehler",      // middleware api, default volkszaehler
> >                 "uuid": "68c338a0-68fb-11e8-8e6e-a9c8f91b23ff",
> >                 "middleware": "http://localhost/middleware.php",
> >                 "identifier": "1-0:16.7.0",    // OBIS identifier
> >                 "aggmode":"avg",
> >                 }, {
> >                 "uuid": "76291be0-68fd-11e8-8fb7-1dea64015521",
> >                 "middleware": "http://localhost/middleware.php",
> >                 "identifier": "1-0:1.8.0",    // OBIS identifier
> >                 "aggmode":"max",
> >

J-A-U, 2018-06-11:
Wie viele Stellen nach dem Komma gibt dein Zähler den bei 1.8.0 aus?
Grundsätzlich sind die Daten im Frontend ja nachher redundant, es ist
daher in der Regel unnötig den Aktualwert Leistung zu loggen.

mfg Daniel

andig, 2018-06-11:
Davon unabhängig- hast Du mal versucht aggtime in den Kanal zu schreiben? Und "wie kann ich" Anfragen gehören auf die Mailingliste M)

Gitsaibot, 2018-06-11:
Der Zähler hat 2 Nachkommastellen. Den Zählerstand will ich in Tabellenform verarbeiten und nicht auf einer Zeitachse. Quasi eine einfach Ansicht des Zählerstandes am Ende jeden Monats. 

@andig Ja das mit aggtime im "channels" hab ich als erstes versucht aber hat aber nichts gebracht. Hab auch versucht meters zweimal für das gleiche device aufzurufen aber das geht auch nicht. 
An die Mailingliste habe ich garnicht gedacht tut mir Leid.

J-A-U, 2018-06-12:
> Der Zähler hat 2 Nachkommastellen.

Bissl wenig, könnte aber hinkommen. Ich rate die von VNB die PIN zum
Zähler anzufordern, vielleicht springt nach der Freischaltung noch
ein bisschen mehr raus.

>  Den Zählerstand will ich in
> Tabellenform verarbeiten und nicht auf einer Zeitachse.

Dafür ist es dann aber egal ob die Zählerstände alle 5s oder 24h
in der DB stehen.

>  auch versucht meters zweimal für
> das gleiche device aufzurufen aber das geht auch nicht.

Weil vzlogger nach der Auswertung den FiFo-Puffer für das device
leert. Ruft man das device nochmal auf sind die Daten schon weg.

andig, 2018-06-12:
Es sieht leider aus als hätten wir `aggtime` tatsächlich nur per meter (https://github.com/volkszaehler/vzlogger/blob/5342f0b096b13424c25327881bfd2b0537ecc180/src/Meter.cpp#L121) und nicht per Channel. Ist also "works as designed" :/

> ...um die Datenbank zu entlasten

Ist halt die Frage was das bringen soll. Wenn Aggregation eingeschaltet ist spielt es für Performance keine Rolle und auf SD geschrieben werden muss ohnehin für den anderen Kanal.

Gitsaibot, 2018-06-12:
Okay, das erklärt warum ich mit meinen Versuchen keinen Erfolg hatte ;). Für mich würde es ~35 Mio Datenbankeinträge weniger im Jahr bedeuten. Deshalb wäre eine Einstellung pro Kanal schon eine feine Sache. Dazu kommt das ich nur die Hälfte an Speicherplatz brauche. Als Hardware habe ich momentan einen kleinen Raspberry Pi Zero W am laufen. 

Wenn ich das mit der Aggregation richtig verstanden habe werden einzelne Sekundenwerte zu einer Minute zusammengefasst usw. Ist das nicht das Gleiche wie aggtime = 1 min und aggmode = avg ? Oder kann ich dann trotzdem noch auf eine Sekunde genau zoomen ?

andig, 2018-06-12:
> Oder kann ich dann trotzdem noch auf eine Sekunde genau zoomen ?

Kannst Du- die Aggregationstabelle kommt nur bei grossen Zeiträumen zum Einsatz wo ansonsten Millionen von Datensätzen gelesen werden müssten (Leistung über 1 Jahr :)

Ich mache das Issue mal zu da Problem gelöst. Viele Grüße, Andreas