volkszaehler/vzlogger #359: Error compiling vzlogger from scratch on debian x86_64 and armbian armhf

griesi, 2018-10-08:
Hey Guys,

I get an error while compiling vzlogger from scratch with the following commands:

`sudo apt-get install build-essential git-core cmake pkg-config subversion libcurl4-openssl-dev libgnutls28-dev libsasl2-dev uuid-dev libtool libssl-dev libgcrypt20-dev libmicrohttpd-dev libltdl-dev libjson-c-dev libleptonica-dev libunistring-dev dh-autoreconf
git clone https://github.com/volkszaehler/vzlogger.git
cd vzlogger
./install.sh
`
The error comes after 

> Building CXX object tests/CMakeFiles/vzlogger_unit_tests.dir/__/src/protocols/MeterOMS.cpp.o

and is:

`make[2]: *** Keine Regel vorhanden, um das Ziel „gmock/src/googlemock-build/googlemock/gtest/libgtest.a“,
  benötigt von „tests/vzlogger_unit_tests“, zu erstellen.  Schluss.
make[2]: Verzeichnis „/usr/src/vzlogger/build“ wird verlassen
CMakeFiles/Makefile2:1284: die Regel für Ziel „tests/CMakeFiles/vzlogger_unit_tests.dir/all“ scheiterte
make[1]: *** [tests/CMakeFiles/vzlogger_unit_tests.dir/all] Fehler 2
make[1]: Verzeichnis „/usr/src/vzlogger/build“ wird verlassen
Makefile:163: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2
`
I tried this on my debian box with debian 9.5 box and on my bananapipro with latest armbian. Same error on both machines.

Maybe disabling the tests as mentioned here #333 would help compiling vzlogger?

Thanks for the help.

J-A-U, 2018-10-08:
This should dissable the tests:

`cd vzlogger/build/`
`cmake BUILD_TEST=false ..`
`make`
`sudo make install`

Please notice: The script failed, so you have to setup up systemd manually.
https://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version#systemd_startscript

J-A-U, 2018-10-12:
Tested myself on Armbian (Debian Stretch), didn't work out.

Suggestions anybody?

J-A-U, 2018-10-12:
Same problem on the vz-image!

griesi, 2018-10-12:
Adding "BUILD_TEST=false" to the cmake command didn't do the job.

For me I changed the makefile generated by cmake directly to state BUILD_TEST=OFF (added a sleep 60 in install.sh ;-) and changed the makefile while install.sh was sleeping)

Then the make was successful. 

I guess there are 2 problems: One with cmake not recognizing "BUILD_TEST=false" and the other Problem with gmock.

J-A-U, 2018-10-13:
> One with cmake not recognizing "BUILD_TEST=false"

Cachefile... :/

mbehr1, 2018-10-13:
this should disable:
…
cd vzlogger/build/
cmake -DBUILD_TEST=off ..
make
sudo make install

> This should dissable the tests:
>
> cd vzlogger/build/
> cmake BUILD_TEST=false ..
> make
> sudo make install
>
> Please notice: The script failed, so you have to setup up systemd manually. [...]

Gruß

Matthias

J-A-U, 2018-10-13:
Strange behavior on my system.

1) changes in ./CMakeFiles.txt only apply when no ./build/CMakeCache.txt exist
2) make sometimes does its own cmake - with standard parameters from ./CMakeLists.txt

And gmock dosn't work...
Workaround by disabling the tests?

J-A-U, 2018-10-13:
> make sometimes does its own cmake - with standard parameters from ./CMakeLists.txt

or (even better) once the cmake ran with tests enabled they will always be performed.

You need to:
`cd vzlogger/build/`
`make clean-all`
`cmake -DBUILD_TEST=off ..`
`make`

griesi, 2018-10-14:
OK ... now we have the correct syntax to disable the tests ... this works for me when changing "install.sh".

The Problem with building the target „gmock/src/googlemock-build/googlemock/gtest/libgtest.a“ of "tests/vzlogger_unit_tests“ still remains.

J-A-U, 2018-12-31:
https://github.com/volkszaehler/vzlogger/pull/370/commits

closed 2019-01-01 by J-A-U