volkszaehler/vzlogger #361: ./install.sh fails on Ubuntu 16.04 (Banana Pi M2 Berry)

mklhmnn, 2018-11-04:
I'm following the installation steps at http://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version : after installing all dependencies and cloning the latest master head (39ee0be844d554), I'm building using `./install.sh` and get following error:

```
...
[ 31%] Linking CXX executable vzlogger
cd /home/pi/vzlogger/build/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/vzlogger.dir/link.txt --verbose=1
/usr/bin/c++    -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++11 -fpermissive -Wno-error=redundant-decls -Wno-ignored-qualifiers   CMakeFiles/vzlogger.dir/vzlogger.cpp.o CMakeFiles/vzlogger.dir/ltqnorm.cpp.o CMakeFiles/vzlogger.dir/Meter.cpp.o CMakeFiles/vzlogger.dir/__/gitSha1.cpp.o CMakeFiles/vzlogger.dir/CurlSessionProvider.cpp.o CMakeFiles/vzlogger.dir/PushData.cpp.o  -o vzlogger  -L/usr/local/lib -rdynamic protocols/libproto.a libvz.a api/libvz-api.a /usr/local/lib/libjson-c.a -lrt -Wl,-Bstatic -lsml -Wl,-Bdynamic -lrt /usr/local/lib/libmbus.so -lm -Wl,-Bstatic -lmicrohttpd -Wl,-Bdynamic -lrt -lgcrypt -lpthread -lm -luuid -ldl -lcurl -lunistring -Wl,-Bstatic -lgnutls -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lgmp -lhogweed -lgmp -lnettle -ltasn1 -lidn -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lz -Wl,-Bdynamic -lsasl2 -lgcrypt -lssl -lcrypto -Wl,-Bstatic -lsml -Wl,-Bdynamic /usr/local/lib/libmbus.so -lm -Wl,-Bstatic -lmicrohttpd -Wl,-Bdynamic -lgcrypt -lpthread -lm -luuid -ldl -lcurl -lunistring -Wl,-Bstatic -lgnutls -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lgmp -lhogweed -lnettle -ltasn1 -lidn -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lgmp -lhogweed -lnettle -ltasn1 -lidn -lz -Wl,-Bdynamic -lsasl2 -lgcrypt -lssl -lcrypto -Wl,-rpath,/usr/local/lib
/usr/bin/ld: protocols/libproto.a(MeterD0.cpp.o): undefined reference to symbol 'socket@@GLIBC_2.4'
//lib/arm-linux-gnueabihf/libc.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
src/CMakeFiles/vzlogger.dir/build.make:267: recipe for target 'src/vzlogger' failed
make[2]: *** [src/vzlogger] Error 1
make[2]: Leaving directory '/home/pi/vzlogger/build'
CMakeFiles/Makefile2:1025: recipe for target 'src/CMakeFiles/vzlogger.dir/all' failed
make[1]: *** [src/CMakeFiles/vzlogger.dir/all] Error 2
make[1]: Leaving directory '/home/pi/vzlogger/build'
Makefile:163: recipe for target 'all' failed
make: *** [all] Error 2
```

J-A-U, 2018-11-04:
I guess there's something wrong with your libraries. Check the dependencies.

mklhmnn, 2018-11-04:
Thanks for the quick answer. What dependencies I should look for? I'm no C++ developer, I haven't written any build file, but just copy-pasted the instructions.

mklhmnn, 2018-11-04:
Running

```
vzlogger $ cd build
vzlogger/build $cmake ..

        ***** Configuration parameters *****
             prefix: /usr/local
             json: -L/usr/local/lib/libjson-c.a;-lrt -I/usr/local/include
             sml:  -L/usr/lib/libsml.a;-lrt -I/usr/include
             microhttpd: -L/usr/lib/arm-linux-gnueabihf/libmicrohttpd.a;-lrt -I/usr/include
             libmbus: -L/usr/local/lib/libmbus.so;-lm -I/usr/local/include
-- Configuring done
CMake Warning at src/CMakeLists.txt:45 (add_executable):
  Cannot generate a safe linker search path for target vzlogger because files
  in some directories may conflict with libraries in implicit directories:

    link library [libz.a] in /usr/lib/arm-linux-gnueabihf may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.

CMake Warning at src/CMakeLists.txt:45 (add_executable):
  Cannot generate a safe runtime search path for target vzlogger because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libssl.so.1.0.0] in /usr/lib/arm-linux-gnueabihf may be hidden by files in:
      /usr/local/lib
    runtime library [libcrypto.so.1.0.0] in /usr/lib/arm-linux-gnueabihf may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.

-- Generating done
-- Build files have been written to: /home/pi/vzlogger/build

```
Could this CMake Warning be somehow related to the above problem?

J-A-U, 2018-11-05:
> Could this CMake Warning be somehow related to the above problem?

Maybe.

>  I'm no C++ developer,

Me, neither.

> just copy-pasted the instructions. 

You have always to look out for failures that may occure.

> How to fix that?

If its not a first time install you might need to
`cd vzlogger/build/`
`make clean-all`

If not, you could test by comment-out lines 64-72 in vzlogger/src/CMakeLists.txt
`if( TARGET )`
`  if( ${TARGET} STREQUAL "ar71xx")`
`    set_target_properties(vzlogger PROPERTIES LINK_FLAGS "-static")`
`    target_link_libraries(vzlogger ${OPENWRT_TARGET_DIR}/usr/lib/libssl.a)`
`    target_link_libraries(vzlogger ${OPENWRT_TARGET_DIR}/usr/lib/libcrypto.a)`
`    target_link_libraries(vzlogger ${OPENWRT_TARGET_DIR}/usr/lib/libz.a)`
`    target_link_libraries(vzlogger dl)`
`  endif( ${TARGET} STREQUAL "ar71xx")`
`endif( TARGET )`

Instead, if you're not hell-bound to Ubuntu, i suggest to use Armbian instead, i compiled vzlogger on it a few days ago.

mklhmnn, 2018-11-05:
Thank you for answering. Commenting out did not help any further.
I've tried Ubuntu because I want to get long support. I'm currently trying Debian 9 stretch lite and if this doesn't work, Raspbian Jessie. This takes a while until everything is ready to work from SATA disk - until I can start compiling volkszaehler.

mklhmnn, 2018-11-05:
I already was happy to see so good compile progress, but at the end it fails on _Debian 9 stretch lite_ with the same error as above. :(

mklhmnn, 2018-11-10:
I've finally managed to install Armbian - nice system - and successfully compile vzlogger. 😃

closed 2018-11-18 by J-A-U