volkszaehler/vzlogger #362: handle read eror / eof from libsml

r00t-, 2018-11-04:
since https://github.com/volkszaehler/libsml/pull/51 libsml will return 0 from reads on eror or EOF (before the change, it would run into a loop and never return).

MeterSML should be adjusted to report those errors to the user, and either abort or try to re-open the device.

the code in question is:
https://github.com/volkszaehler/vzlogger/blob/39ee0be8/src/protocols/MeterSML.cpp#L217

r00t-, 2018-11-04:
it appears that the current code will log "in it's current state, it will log ""short message from sml_transport_read" in a tight loop." in a tight loop.

mbehr1, 2018-12-31:
I'd like to fix the real problem in libsml but what I could add here is that e.g. if libsml read returns 0 I reopen the device. Sadly I can't test that change at all. But if the solution/workaround is ok, I'll add a PR.

mklhmnn, 2018-12-31:
I'm happy to give it a try. It would be helpful if you could log something in that case, so we know this sporadic error really happened again.

mbehr1, 2018-12-31:
@mklhmnn see PR#370 with a potential fix. Should at least provide some more useful debugging output.

andig, 2019-05-31:
I feel this might have been fixed by #370.

closed 2019-05-31 by andig