volkszaehler/vzlogger #364: Timestamps  from OMS meters and time zone

mley, 2018-11-18:
I'm using vzlogger to read data from a Siemens TD-3511 and write the data to influxdb. As seeing from MeterOMS.cpp, it looks like the local time value from the meter is used. It seems, there is no timezone involved and the time is sent to influx as is. That means influx uses the time value from the meter (in whatever timezone that is) without any timezone information.

InfluxDB recommends to store all time values in UTC. 

I'm already storing other data in influxdb in UTC and that works best with all the other tools I'm using (chronograf). 

Expected behaviour: One can configure a timezone in meter configuration, stating the timezone that the meter is in. The time of the data sent to influxdb is in UTC

Actual behaviour: The time of the data sent to influxdb is not in UTC, but the timezone of the meter.

J-A-U, 2018-11-18:
You might use
`"use_local_time": true`
and select a timeconfiguration on the Rpi that suits your needs best.

mley, 2018-11-18:
Thanks for the hint, but unfortunately it's not working as expected:

`[Nov 18 08:30:44][chn3] request body is power,uuid=c0cfd381-1466-4ffc-899c-a87de1a03378,source=home value=404.000000 1542533444000`

Where 1542533444 is 2018-11-18T09:30:44 in UTC. The timezone on the raspberry (and docker container containing vzlogger) is set to UTC.

After a quick grep through the code, it looks, like the `use_local_time` option is only implemented for SML meters. But that implementation gives me a hint how to change MeterOMS.

J-A-U, 2018-11-18:
Excellent, thank you!