volkszaehler/vzlogger #369: Error "short message from sml_transport_read len=0"

mklhmnn, 2018-12-23:
I'm using vzlogger v0.6.1 based on heads/master-0-ga68f7ff8c4 and from time to time messages like

`[Dec 22 12:21:10][sml]  short message from sml_transport_read len=0`

flood my log file.

`$ sudo systemctl restart vzlogger`

fixes this bug. Should I really restart the vzlogger that way each night using cron?

J-A-U, 2018-12-23:
This message is typical for reading errors from your meter. Check the function of your IR-reader and the position on your meter to prevent it.

andig, 2018-12-23:
Are you saying the flooding only stops when restarted? Meanwhile no meter data is being received?

mklhmnn, 2018-12-23:
Yes, after restarting the vzlogger it works fine again for a couple of weeks. IMHO this rather speaks for some software problem.

J-A-U, 2018-12-23:
It may be a software issue. Depending on the IR-reader you are using my best guess would be something with the serial driver.

For further help: Please post a logfile.

mklhmnn, 2018-12-23:
I would have posted the log file, but it is 2048M large and only contains the above line. BTW, a few minutes ago it happened again.

mklhmnn, 2018-12-23:
I'm using log-level 3. Should I increase it?

J-A-U, 2018-12-23:
Yes, 15.

mklhmnn, 2018-12-23:
OK, increased to 15 and now the log file is again at 2GB. Here are the few seconds before the error:

```
...
[Dec 23 16:39:59][chn0] Adding reading to queue (value=13163300.60 ts=1545579599820)
[Dec 23 16:39:59][chn1] Adding reading to queue (value=22075824.70 ts=1545579599820)
[Dec 23 16:39:59][mtr1] Got 4 new readings from meter:
[Dec 23 16:39:59][mtr1] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=28378543.80 ts=1545579599822
[Dec 23 16:39:59][mtr1] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=0.00 ts=1545579599822
[Dec 23 16:39:59][mtr1] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=28378543.80 ts=1545579599822
[Dec 23 16:39:59][mtr1] Reading: id=1-0:15.7.0*255/ObisIdentifier:1-0:15.7.0*255 value=0.00 ts=1545579599822
[Dec 23 16:39:59][chn2] Adding reading to queue (value=28378543.80 ts=1545579599822)
[Dec 23 16:40:01][mtr1] Got 4 new readings from meter:
[Dec 23 16:40:01][mtr1] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=28378543.80 ts=1545579601166
[Dec 23 16:40:01][mtr1] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=0.00 ts=1545579601166
[Dec 23 16:40:01][mtr1] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=28378543.80 ts=1545579601166
[Dec 23 16:40:01][mtr1] Reading: id=1-0:15.7.0*255/ObisIdentifier:1-0:15.7.0*255 value=0.00 ts=1545579601166
[Dec 23 16:40:01][chn2] Adding reading to queue (value=28378543.80 ts=1545579601166)
[Dec 23 16:40:01][mtr0] Got 6 new readings from meter:
[Dec 23 16:40:01][mtr0] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=13163300.80 ts=1545579601606
[Dec 23 16:40:01][mtr0] Reading: id=1-0:2.8.0*255/ObisIdentifier:1-0:2.8.0*255 value=22075824.70 ts=1545579601606
[Dec 23 16:40:01][mtr0] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=0.00 ts=1545579601606
[Dec 23 16:40:01][mtr0] Reading: id=1-0:2.8.1*255/ObisIdentifier:1-0:2.8.1*255 value=22075824.70 ts=1545579601606
[Dec 23 16:40:01][mtr0] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=13163300.80 ts=1545579601606
[Dec 23 16:40:01][mtr0] Reading: id=1-0:15.7.0*255/ObisIdentifier:1-0:15.7.0*255 value=405.90 ts=1545579601606
[Dec 23 16:40:01][chn0] Adding reading to queue (value=13163300.80 ts=1545579601606)
[Dec 23 16:40:01][chn1] Adding reading to queue (value=22075824.70 ts=1545579601606)
[Dec 23 16:40:02][mtr1] Got 4 new readings from meter:
[Dec 23 16:40:02][mtr1] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=28378543.80 ts=1545579602500
[Dec 23 16:40:02][mtr1] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=0.00 ts=1545579602500
[Dec 23 16:40:02][mtr1] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=28378543.80 ts=1545579602500
[Dec 23 16:40:02][mtr1] Reading: id=1-0:15.7.0*255/ObisIdentifier:1-0:15.7.0*255 value=0.00 ts=1545579602500
[Dec 23 16:40:02][chn2] Adding reading to queue (value=28378543.80 ts=1545579602500)
[Dec 23 16:40:03][mtr0] Got 6 new readings from meter:
[Dec 23 16:40:03][mtr0] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=13163301.00 ts=1545579603443
[Dec 23 16:40:03][mtr0] Reading: id=1-0:2.8.0*255/ObisIdentifier:1-0:2.8.0*255 value=22075824.70 ts=1545579603443
[Dec 23 16:40:03][mtr0] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=0.00 ts=1545579603443
[Dec 23 16:40:03][mtr0] Reading: id=1-0:2.8.1*255/ObisIdentifier:1-0:2.8.1*255 value=22075824.70 ts=1545579603443
[Dec 23 16:40:03][mtr0] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=13163301.00 ts=1545579603443
[Dec 23 16:40:03][mtr0] Reading: id=1-0:15.7.0*255/ObisIdentifier:1-0:15.7.0*255 value=387.80 ts=1545579603443
[Dec 23 16:40:03][chn0] Adding reading to queue (value=13163301.00 ts=1545579603443)
[Dec 23 16:40:03][chn1] Adding reading to queue (value=22075824.70 ts=1545579603443)
[Dec 23 16:40:03][mtr1] Got 4 new readings from meter:
[Dec 23 16:40:03][mtr1] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=28378543.80 ts=1545579603843
[Dec 23 16:40:03][mtr1] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=0.00 ts=1545579603843
[Dec 23 16:40:03][mtr1] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=28378543.80 ts=1545579603843
[Dec 23 16:40:03][mtr1] Reading: id=1-0:15.7.0*255/ObisIdentifier:1-0:15.7.0*255 value=0.00 ts=1545579603843
[Dec 23 16:40:03][chn2] Adding reading to queue (value=28378543.80 ts=1545579603843)
[Dec 23 16:40:05][mtr1] Got 4 new readings from meter:
[Dec 23 16:40:05][mtr1] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=28378543.80 ts=1545579605198
[Dec 23 16:40:05][mtr1] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=0.00 ts=1545579605198
[Dec 23 16:40:05][mtr1] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=28378543.80 ts=1545579605198
[Dec 23 16:40:05][mtr1] Reading: id=1-0:15.7.0*255/ObisIdentifier:1-0:15.7.0*255 value=0.00 ts=1545579605198
[Dec 23 16:40:05][chn2] Adding reading to queue (value=28378543.80 ts=1545579605198)
[Dec 23 16:40:05][mtr0] Got 6 new readings from meter:
[Dec 23 16:40:05][mtr0] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=13163301.20 ts=1545579605292
[Dec 23 16:40:05][mtr0] Reading: id=1-0:2.8.0*255/ObisIdentifier:1-0:2.8.0*255 value=22075824.70 ts=1545579605292
[Dec 23 16:40:05][mtr0] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=0.00 ts=1545579605292
[... cut ...]
[Dec 23 16:40:05][mtr0] Reading: id=1-0:15.7.0*255/ObisIdentifier:1-0:15.7.0*255 value=388.40 ts=1545579605292
[Dec 23 16:40:05][chn0] Adding reading to queue (value=13163301.20 ts=1545579605292)
[Dec 23 16:40:05][chn1] Adding reading to queue (value=22075824.70 ts=1545579605292)
[Dec 23 16:40:06][sml]  short message from sml_transport_read len=0
[Dec 23 16:40:06][mtr1] Got 0 new readings from meter:
[Dec 23 16:40:06][sml]  short message from sml_transport_read len=0
[Dec 23 16:40:06][mtr1] Got 0 new readings from meter:
...
```
The last 2 lines repeat again and again (of course with different times).

r00t-, 2018-12-23:
note that this might relate to https://github.com/volkszaehler/libsml/pull/51

r00t-, 2018-12-24:
this is esentially a duplicate of volkszaehler/vzlogger#362

r00t-, 2018-12-24:
@mklhmnn:
we have an important question for you here:
would vzlogger work reliably previously?
because according to my understanding of the code, the condition that leads to "short message from sml_transport_read len=0" should previously have led to vzlogger stopping to log data, with no message being logged.

J-A-U, 2018-12-25:
> because according to my understanding of the code, the condition that leads to "short message from sml_transport_read len=0" should previously have led to vzlogger stopping to log data, with no message being logged.

I'm pretty sure there's more than 1 instance that leads to this message. I have them appearing from time to time. Vzlogger keeps working, just this single, incomplete message gets dropped.

mklhmnn, 2018-12-28:
Happened again. Attached you'll find a screenshot from the web frontend. As you can see, the yellow line simply stops while the other lines continue. IIRC, it is always the yellow line (the single PV sensor) that dies.
[image]

J-A-U, 2018-12-28:
> "allowskip": true,

?

andig, 2018-12-28:
@J-A-U `allowskip` is only evaluated at startup!

mklhmnn, 2018-12-29:
Happened again. What can I do to help to resolve this bug? Is there some debug option that might give you clues what could cause the problem?

r00t-, 2018-12-29:
@andig, @J-A-U:
can you please make sure you understand https://github.com/volkszaehler/libsml/pull/51 and it's implications before you comment any more on this?

r00t-, 2018-12-29:
@mklhmnn:
the issue is that reading from your meter fails. with an error. after some time.    
vzlogger does not currently handle this correctly.    
the solution is to re-open the meter. either by restarting vzlogger - which you are doing manually. or, preferably, by having vzlogger re-open the meter without restarting when the eror occurs.

before https://github.com/volkszaehler/libsml/pull/51 , vzlogger would simply "hang" and stop logging data from that sensor, because libsml would ignore the error and hang.    
since the change in libsml, it will, instead of hanging, return 0 from `sml_transport_read()`, which is exactly what vzlogger is reporting to be happening.   

when i made the change in libsml, i already created a ticket to fix the handling in vzlogger: https://github.com/volkszaehler/vzlogger/issues/362    
but this has not been a priority, because it's just an improvement, not a bug - vzlogger never handled this case correctly.    
(as you confirmed: "But it always stopped working after some time - sometimes it stops multiple times a day, sometimes it runs without problem for weeks")

i will try to write a fix sometime soon, and will get back to you for testing it.

i am however very glad to have actually found someone (you) who is affected by those issues, and can help us test the fix (once it's written).

r00t-, 2018-12-29:
and now i'll just close this because it is a duplicate of #362 .

but feel free to leave further comments.

andig, 2018-12-29:
@r00t- 

> can you please make sure you understand volkszaehler/libsml#51 and it's implications before you comment any more on this?

Feels like you assume I'm not aware of it which is funny since I had attempted to solve it myself as you are well aware. Not sure how you got that notion as I only pointed out what `allowskip` actually does. This is of cause unrelated to the root cause of this issue.

I would appreciate a more friendly conversational tone here, especially when addressing @J-A-U who's helping out whereever he can.

That being said, #362 still needs a C dev to loo into or we should rather rewind the libsml commit since right now we're in inconsistent state regarding vzlogger.

ping @mbehr1 could you kindly take a look at #362?

andig, 2018-12-29:
Depends on #362

mbehr1, 2018-12-30:
> @r00t- <https://github.com/r00t->
> can you please make sure you understand volkszaehler/libsml#51 <https://github.com/volkszaehler/libsml/pull/51> and it's implications before you comment any more on [...]

yep, will take a look at #362.

Gruß

Matthias

mbehr1, 2018-12-31:
sadly libsml prints errors to stderr only. can you start vzlogger from console and reproduce the issue? I'd like to see whether there are any fprintf(stderr,"libsml...")) outputs visible describing more the real root cause.

andig, 2019-05-31:
I feel this might have been fixed by #370.

closed 2019-05-31 by andig