volkszaehler/vzlogger #372: build on rpi jessie doesn't work

bmshine, 2019-01-27:
Hi if i try to build the vzlogger on rpi / jessie i am getting the following error:

[ 54%] Building CXX object src/CMakeFiles/vz.dir/mqtt.cpp.o
cd /root/vzlogger/build/src && /usr/bin/c++   -DHAVE_CONFIG_HPP -I/usr/local/include -I/root/vzlogger/build -I/root/vzlogger/include -I/root/vzlogger  -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++11 -fpermissive -Wno-error=redundant-decls -Wno-ignored-qualifiers   -g3 -o CMakeFiles/vz.dir/mqtt.cpp.o -c /root/vzlogger/src/mqtt.cpp
/root/vzlogger/src/mqtt.cpp: In constructor âMqttClient::MqttClient(json_object*)â:
/root/vzlogger/src/mqtt.cpp:139:47: error: âmosquitto_threaded_setâ was not declared in this scope
    int res = mosquitto_threaded_set(_mcs, true);
                                               ^
src/CMakeFiles/vz.dir/build.make:281: recipe for target 'src/CMakeFiles/vz.dir/mqtt.cpp.o' failed
make[2]: *** [src/CMakeFiles/vz.dir/mqtt.cpp.o] Error 1
make[2]: Leaving directory '/root/vzlogger/build'
CMakeFiles/Makefile2:1062: recipe for target 'src/CMakeFiles/vz.dir/all' failed
make[1]: *** [src/CMakeFiles/vz.dir/all] Error 2
make[1]: Leaving directory '/root/vzlogger/build'
Makefile:163: recipe for target 'all' failed
make: *** [all] Error 2

I think for a succesfull build a newer version of mosquitto is required.

Is there still a way to compile it?

Best regards
Markus

mbehr1, 2019-01-27:
Hi,

you can ask cmake to not build mqtt:
cmake -DENABLE_MQTT=Off ..

Mit freundlichen Grüßen / Best Regards
Matthias Behr

Sent from a mobile device.

bmshine, 2019-01-27:
Hallo, vielen Dank für die prompte Antwort. Ich habe es einfach im makefile abgeschaltet. Gibt es eine Möglichkeit die passenden Libs "reinzufummeln" damit auch MQTT funktioniert?
Gruss Markus

andig, 2019-01-27:
Hast Du schonmal einen Blick ins Wiki geworfen?

bmshine, 2019-01-28:
Hi, wenn du auf den Umweg mit Node rede anspielst, dann habe ich das Wiki gelesen.  
Das funktioniert bei mir auch einwandfrei. Es wäre nur ein weiterer Schritt um das Ganze einfacher zu halten.
Gruss Markus

kassi, 2020-05-05:
@bmshine @andig Ich habe das gleiche Problem, finde aber zu den keywords nichts im Wiki, was mir weiter hilft, es zu compilieren. Zu "node red" sehe ich hier keinen Zusammenhang. Vielleicht kann mich jemand aufklären? Das wäre super.

r00t-, 2021-03-26:
which version of mosquitto are you people trying to build with?

J-A-U, 2022-05-08:
Just did a build on Bullseye with:
```
$ apt show libmosquitto-dev
Package: libmosquitto-dev
Version: 1.4.10-3+deb9u5
Priority: optional
Section: libdevel
Source: mosquitto
Maintainer: Roger A. Light <roger@atchoo.org>
Installed-Size: 115 kB
Depends: libmosquitto1 (<< 1.4.10-3+deb9u5.1~), libmosquitto1 (>= 1.4.10-3+deb9u5)
Breaks: libmosquitto0-dev (<< 1.2-1~)
Replaces: libmosquitto0-dev (<< 1.2-1~)
Homepage: http://mosquitto.org/
Download-Size: 42,2 kB
APT-Manual-Installed: yes
APT-Sources: http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
Description: MQTT version 3.1/3.1.1 client library, development files
 This is the header and man page for the libmosquitto1 C library, which is a
 library for implementing MQTT version 3.1/3.1.1 clients. This package is needed to do
 development with libmosquitto1.
```
Works so far...