volkszaehler/vzlogger #373: GCC 8.3 / buster build error

andig, 2019-03-14:
This is from running raspbian buster:

    [ 28%] Linking CXX executable vzlogger
    cd /home/pi/vzlogger/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/vzlogger.dir/link.txt --verbose=1
    /usr/bin/c++   -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++11 -fpermissive -Wno-error=redundant-decls -Wno-ignored-qualifiers    -rdynamic CMakeFiles/vzlogger.dir/vzlogger.cpp.o CMakeFiles/vzlogger.dir/ltqnorm.cpp.o CMakeFiles/vzlogger.dir/Meter.cpp.o CMakeFiles/vzlogger.dir/__/gitSha1.cpp.o CMakeFiles/vzlogger.dir/CurlSessionProvider.cpp.o CMakeFiles/vzlogger.dir/PushData.cpp.o  -o vzlogger -Wl,-rpath,/usr/local/lib protocols/libproto.a libvz.a api/libvz-api.a /usr/local/lib/libjson-c.a -lrt -Wl,-Bstatic -lsml -Wl,-Bdynamic -lrt /usr/local/lib/libmbus.so -lm -Wl,-Bstatic -lmicrohttpd -Wl,-Bdynamic -lrt -lgcrypt -lpthread -lm -luuid -ldl -lcurl -lunistring -Wl,-Bstatic -lgnutls -lz -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lgmp -lhogweed -lgmp -lnettle -ltasn1 -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lz -Wl,-Bdynamic -lsasl2 -lgcrypt -lssl -lcrypto -Wl,-Bstatic -lsml -Wl,-Bdynamic /usr/local/lib/libmbus.so -lm -Wl,-Bstatic -lmicrohttpd -Wl,-Bdynamic -lgcrypt -lpthread -lm -luuid -ldl -lcurl -lunistring -Wl,-Bstatic -lgnutls -lz -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lgmp -lhogweed -lnettle -ltasn1 -lz -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lgmp -lhogweed -lnettle -ltasn1 -Wl,-Bdynamic -lsasl2 -lgcrypt -lssl -lcrypto
    /usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libgnutls.a(str-idna.o): undefined reference to symbol 'idn2_to_unicode_8z8z@@IDN2_0.0.0'
    /usr/bin/ld: //usr/lib/arm-linux-gnueabihf/libidn2.so.0: error adding symbols: DSO missing from command line

Looks a lot like https://stackoverflow.com/questions/19901934/strange-linking-error-dso-missing-from-command-line but I have no idea what/how to fix.

mbehr1, 2019-03-14:
i‘ll take a look

Mit freundlichen Grüßen / Best Regards
Matthias Behr

Sent from a mobile device.

J-A-U, 2019-04-09:
Build worked out fine over her with an up to date buster and gcc 8.3.0-4.

fis1497, 2019-04-09:
on a fresh checkout and compile everything is fine. i had the same error on old sources

andig, 2019-04-09:
Not working for me:

    pi@server:~/vzlogger $ uname -a
    Linux server 4.9.46-v7+ #1032 SMP Wed Aug 30 12:09:14 BST 2017 armv7l GNU/Linux

    pi@server:~/vzlogger $ git pull
    Already up to date.

    pi@server:~/vzlogger $ gcc -v
    Using built-in specs.
    COLLECT_GCC=gcc
    COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/8/lto-wrapper
    Target: arm-linux-gnueabihf
    Configured with: ../src/configure -v --with-pkgversion='Raspbian 8.3.0-4+rpi1' --with-bugurl=file:///usr/share/doc/gcc-8/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-8 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --disable-libquadmath-support --enable-plugin --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --disable-werror --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
    Thread model: posix
    gcc version 8.3.0 (Raspbian 8.3.0-4+rpi1)

    pi@server:~/vzlogger $ make
    [ 34%] Linking CXX executable vzlogger
    cd /home/pi/vzlogger/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/vzlogger.dir/link.txt --verbose=1
    /usr/bin/c++   -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++11 -fpermissive -Wno-error=redundant-decls -Wno-ignored-qualifiers    -rdynamic CMakeFiles/vzlogger.dir/vzlogger.cpp.o CMakeFiles/vzlogger.dir/ltqnorm.cpp.o CMakeFiles/vzlogger.dir/Meter.cpp.o CMakeFiles/vzlogger.dir/__/gitSha1.cpp.o CMakeFiles/vzlogger.dir/CurlSessionProvider.cpp.o CMakeFiles/vzlogger.dir/PushData.cpp.o  -o vzlogger -Wl,-rpath,/usr/local/lib protocols/libproto.a libvz.a api/libvz-api.a /usr/local/lib/libjson-c.a -lrt -Wl,-Bstatic -lsml -Wl,-Bdynamic -lrt /usr/local/lib/libmbus.so -lm -Wl,-Bstatic -lmicrohttpd -Wl,-Bdynamic -lrt -lgcrypt -lpthread -lm -luuid -ldl -lcurl -lunistring -Wl,-Bstatic -lgnutls -lz -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lgmp -lhogweed -lgmp -lnettle -ltasn1 -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lz -Wl,-Bdynamic -lsasl2 -lgcrypt -lssl -lcrypto -Wl,-Bstatic -lsml -Wl,-Bdynamic /usr/local/lib/libmbus.so -lm -Wl,-Bstatic -lmicrohttpd -Wl,-Bdynamic -lgcrypt -lpthread -lm -luuid -ldl -lcurl -lunistring -Wl,-Bstatic -lgnutls -lz -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lgmp -lhogweed -lnettle -ltasn1 -lz -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lgmp -lhogweed -lnettle -ltasn1 -Wl,-Bdynamic -lsasl2 -lgcrypt -lssl -lcrypto
    /usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libgnutls.a(str-idna.o): undefined reference to symbol 'idn2_to_unicode_8z8z@@IDN2_0.0.0'
    /usr/bin/ld: //usr/lib/arm-linux-gnueabihf/libidn2.so.0: error adding symbols: DSO missing from command line
    collect2: error: ld returned 1 exit status
    make[2]: *** [src/CMakeFiles/vzlogger.dir/build.make:201: src/vzlogger] Error 1
    make[2]: Leaving directory '/home/pi/vzlogger'
    make[1]: *** [CMakeFiles/Makefile2:1031: src/CMakeFiles/vzlogger.dir/all] Error 2
    make[1]: Leaving directory '/home/pi/vzlogger'
    make: *** [Makefile:166: all] Error 2

Would appreciate to keep this open.

J-A-U, 2019-04-10:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

pi@raspberrypi:~/vzlogger $ git pull
Already up to date.

pi@raspberrypi:~/vzlogger $ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/8/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 8.3.0-4+rpi1' --with-bugurl=file:///usr/share/doc/gcc-8/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-8 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --disable-libquadmath-support --enable-plugin --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --disable-werror --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 8.3.0 (Raspbian 8.3.0-4+rpi1)

pi@raspberrypi:~/vzlogger $ apt show libgnutls28-dev
Package: libgnutls28-dev
Version: 3.6.6-2
[...]
```

frankrichter, 2019-04-10:
@andig: dein Kernel ist uralt. Reboot vergessen?

andig, 2019-04-12:
Same thing after reboot with newer kernel:

    pi@server:~/vzlogger $ uname -a
    Linux server 4.19.32-v7+ #1210 SMP Tue Apr 2 14:07:04 BST 2019 armv7l GNU/Linux

J-A-U, 2019-04-13:
I managed to reproduce this failure.
1) build vzlogger on raspian stretch
2) update system to buster
3) build vzlogger again

Solution: delete the build directory and run install.sh again.
Maybe solution: `cp clean-all-cmake.txt build/`, `cd build` and run `make clean-all`.

andig, 2019-04-13:
Thanks, better. I had tried `make clean` before to no avail. After `make clean-all` the build proceeds further:

    make[2]: *** No rule to make target 'gmock/src/googlemock-build/lib/libgtest.a', needed by 'tests/vzlogger_unit_tests'.  Stop.

But this is a separate issue.

closed 2019-04-13 by andig