volkszaehler/vzlogger #374: test suite fails on debian buster: vzlogger.conf not copied to build dir

Zugschlus, 2019-05-28:
Hi,

when I use the Debian packaging (stand by for a PR) to built vzlogger, the test suite fails since in the build subdir etc/vzlogger.conf is missing. Manually coping the file to the subdir makes make test succeed. 

I'm not proficient enough to fix this in Cmake. it should be a trivial fix though.

Greetings
Marc

r00t-, 2019-05-28:
@ampelbein:
just FYI... maybe you have this fixed already?

( he's working on debian packaging, https://github.com/volkszaehler/libsml/issues/72 ,  https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=864255 )

Zugschlus, 2019-05-28:
Currently just working for my private installation, I am aware of the blocking licensing issue.

If that one is solved, I might be able to help, I am a DD, just in case you need a sponsor for an initial upload.

Zugschlus, 2019-05-28:
A similiar issue happens when one finally tries to install the built package, postinst fails because there is no /etc/vzlogger.conf in the package.

btw, the package in this repo still has "luft nach oben" before it can be uploaded to Debian. For example, it uses an ancient debhelper compat level and standards-version.

andig, 2019-05-31:
Should be partially fixed by #379

claashk, 2020-01-15:
I have the same issue with the current build. I am not sure, whether running the unit tests in the source dir as suggested by #379 is a good idea. Why not copy the config using [configure_file](https://cmake.org/cmake/help/v3.2/command/configure_file.html) ? If you are interested, I could create a pull request.

claashk, 2020-03-26:
In my opinion this should have been fixed by PR #415. Can you confirm this ?

andig, 2020-03-26:
Please reopen if this remains to be an issue.

closed 2020-03-26 by andig

Zugschlus, 2020-04-10:
I don't see any tests being run any more when the package is built. I am not sure whether not using the tests that come with the source is the intended behavior. Also, the built package still doesn't have a vzlogge.conf.

claashk, 2020-04-10:
That is strange. Could you check, whether BUILD_TEST is set to False in your cmake config ? Which compiler are you using ?

Zugschlus, 2020-04-11:
I apologize. I git pulled into my work tree where I had a workaround in place to get the package build. That included BUILD_TEST=false :-(

I can confirm that the test suite not actually runs withour errors. Sorry again.

Greetings
Marc