volkszaehler/vzlogger #376: should log its PID on startup

Zugschlus, 2019-05-29:
Hi,

vzlogger should, at least in the more verbose log levels, write its pid to the log. That would make debugging way easier.

Greetings
Marc

r00t-, 2021-03-12:
do we also want a pidfile, or is logging sufficient?

r00t-, 2021-03-12:
also, now that we have `--foreground`, this is much less critical.

Zugschlus, 2021-03-14:
A pidfile would allow to write a way more robust sysv init script. In the systemd world, that is not really necessary since systemd puts each service in its own process group and simply kills off everything that is in the process group. Since I am using systemd, I don't have a strong opinion about that.

Logging the pid is most probably a trivial one-line change, that is really worth doing.