volkszaehler/vzlogger #382: please consider allowing # as a comment character

Zugschlus, 2019-06-15:
Hi,

please consider allowing # as a comment character.

Greetings
Marc

J-A-U, 2019-06-15:
Where? In config.json?

Zugschlus, 2019-06-15:
ah, sorry. in /etc/vzlogger.conf

J-A-U, 2019-06-15:
That would be a syntax violation.

Actually JSON (as standard) doesn't allow _any_ comments. The structure should be enough to describe whats going on.
Well, it's not. And we are already off by allowing JavaScript comments. Not far, because: **J**ava**S**cript **O**bject **N**otation.

Zugschlus, 2019-06-15:
Well, not willing to begin a discussion about JSON probably not being the best choice for a configuration file that is intended to be written by humans, i am merely doing you a service by pointing out where an experienced UNIX system administrator might (and actually did) struggle when using your software. Feel free to close this issue.

J-A-U, 2019-06-15:
> JSON probably not being the best choice for a configuration file that is intended to be written by humans

I'm with you on this one. Our wish would be to automate and get rid of configuration at all.
But, we actually don't have someone implement it. I'm not even sure we could get hjson as short term solution.

closed 2019-06-15 by andig