volkszaehler/vzlogger #383: init script issues resulting in strange behavior with systemd generator

Zugschlus, 2019-06-15:
Hi,

on Debian systems, init scripts are run via systemd and a systemd generator. This mechanism expects the init script to be extraordinary clean in reporting its results back to the caller.

This leads to quite a number of issues including:

- restart not working since the init script returns before the process  has fully died, resulting in the start faliing because the old instance is still around
- no reload option offered (which would, for example, be handy for log rotation
- systemd thinking the process is running after a failed start (for example, when a syntax error in the configuration file has prevented the daemon from coming up), resulting in the need to do systemctl stop on a not-running daemon before another start can be attempted

Greetings
Marc

andig, 2019-06-18:
Can this be achieved by the init script or modification of vzlogger? As we don‘t have any debian experts here- do you want to contribute a PR?

Zugschlus, 2019-07-03:
I think this can be properly addressed by the init script. Those things are also not Debian specific, it's just shell scripting.

Any no, I cannot help here, I have moved to using a systemd unit for my vzlogger, sorry.

J-A-U, 2019-08-17:
Nach der Diskussion in #384 verstehe ich um was es geht.
Bleibt aber die Frage: Genügt es den Abschnitt
```
do_reload() {
	#
	# If the daemon can reload its configuration without
	# restarting (for example, when it is sent a SIGHUP),
	# then implement that here.
	#
	start-stop-daemon --stop --signal 1 --quiet  --name $NAME
	return 0
}
```

in debian/vzlogger.init zu entfernen?

Zugschlus, 2019-08-17:
Lies mal die Debian policy 9.3.2, da wird ganz gut erklärt, was die Mechanismen um ein Initscript herum erwarten.

Kurz gefasst:

- restart muss restarten
- force-reload kann die Konfiguration neu laden, wenn der daemon das kann, sonst muss es restarten
- reload darf nur eine Funktion haben, wenn der Daemon seine Konfiguration neu laden kann. Der vzlogger kann das offensichtlich (leider) nicht, also weg damit.

Ich persönlich fände es schöner, wenn der vzlogger auf ein SIGHUP hin sein Logfile schließen und neu öffnen würde (damit man ihm das Signal nach Logrotation schicken kann, ohne potenziell Daten zu verlieren) und die Konfiguration neu einläse, aber das sind umfangreiche Änderungen am Daemon selbst.

J-A-U, 2019-08-18:
Was den restart angeht müsste man halt mal testen ob die aktuelle Version zum Initscript passt oder nicht. Ich hab aber auch kein System auf dem noch mit Init gearbeitet wird.

r00t-, 2021-03-11:
i don't think reloading the configuration on sighup is sensible to implement.
we either need to tear down and rebuild all state (might aswell restart),
or we would need a LOT of logic to detect changes in the config, to only stop/reconfigure/restart affected threads.
the cost/use factor does not work out at all.

only re-opening the logfile on a signal should be within reach.
does that make sense by itself?
we even have a lock in the logging code already, that could be used to pause writes while the file is being reopened.

Zugschlus, 2021-03-11:
I think that it might be worth going through the motions of tears down and rebuild, that is still quicker than stop, wait for "really dead" and restart. In that time, vzlogger might miss pulses from S0 meters.

If there is only a function to re-opening the log file, that should not be tied to SIGHUP but rather, for example, to SIGUSR1 so that one can process log rotation without restarting (again to avoid missing S0 pulses).

r00t-, 2021-03-12:
@Zugschlus: please check #458

r00t-, 2021-03-12:
i also think i came up with an easy-to-implement solution for "reloading" on SIGHUP:
i would simply tear down vzlogger like on exit, but then instead of exiting, do an `execvp(argv+"-f")`.
that is easy enough to implement, and should suffice for the requirements.
seems ok?

r00t-, 2021-03-12:
[wrote this before the previous post but had not submitted:]
even when only internally stopping and restarting the metering thread, we may lose s0 pulses.
and we will not be able to calculate power for the first pulse after the restart.

there are some programs that provide high-availability restarting,
including magic like passing open FDs and runtime data over to the new instance.
but we should be realistic with the requirements of vzlogger.
nobody will get hurt if we lose one or two seconds of data on a restart.

Zugschlus, 2021-03-14:
> @Zugschlus: please check #458

I cannot really comment on the actual change, my C is not fluent enough. I didn't see something blatantly wrong.

Zugschlus, 2021-03-14:
> i also think i came up with an easy-to-implement solution for "reloading" on SIGHUP:
> i would simply tear down vzlogger like on exit, but then instead of exiting, do an `execvp(argv+"-f")`.
> that is easy enough to implement, and should suffice for the requirements.
> (if we get fancy, parse the config into a temporary config object to validate it. if we're lazy document that the user has to ensure this.)
> seems ok?

I would probably not do this. It's just "having reload" without actually having reload. Better document the missing reload feature more clearly and fix the init script.

Zugschlus, 2021-03-14:
> [wrote this before the previous post but had not submitted:]
> even when only internally stopping and restarting the metering thread, we may lose s0 pulses.
> and we will not be able to calculate power [...]

Agreed. Probably not worth doing.

Zugschlus, 2021-03-14:
> (i need to add that it seems sad that one can't trust the wrapper to restart vzlogger quickly.)

The easy fix would be to have do_stop in the init script wait until the process is really gone and only return then. Having a pidfile would make that even more easy ;-)

r00t-, 2021-03-14:
> >@Zugschlus: please check #458
> I cannot really comment on the actual change, my C is not fluent enough. I didn't see something blatantly wrong.

i guess i was hoping you could try it and check if it works for you.

Zugschlus, 2021-03-14:
> > > @Zugschlus: please check #458
> > 
> > 
> > I cannot really comment on the actual change, my C is not fluent enough. I didn't see something blatantly wrong.
> 
> i guess i was hoping you could try it and check if it works for you.

I can do that but it is unlikely that I will find the time to do so before easter. Please dont't hold your breath, but I'll give feedback one I have it.

r00t-, 2021-03-17:
it would seem that any *proper* fixing of the initscript requires that we have a pidfile.
i may look into implementing this, but i'm afraid it will lead to lots of details and corner cases needing to be taken care of.
i wonder if there's some best-practice documenation or a library...

r00t-, 2021-03-17:
shouldn't it be worth improving the situation by getting systemd setups to use a proper systemd unit instead of the initscript via the generator?
(it seems to me that that's a quicker path to success than getting the initscript to be fully compliant.)

i see that we have a basic systemd unit file in https://github.com/volkszaehler/vzlogger/blob/master/etc/vzlogger.service
but it seems very basic indeed.

especially with the newly added `-f`, it should be possible to write a better one.

> Any no, I cannot help here, I have moved to using a systemd unit for my vzlogger, sorry.

can you share yours?

Zugschlus, 2021-03-18:
> > Any no, I cannot help here, I have moved to using a systemd unit for my vzlogger, sorry.
> 
> can you share yours?

Sure:

```
[Unit]
Description=vzlogger
After=syslog.target network.target
After=mysql.service ntp.service

[Service]
ExecStart=/usr/bin/vzlogger -c /etc/vzlogger.conf
#reload doesn't work, https://github.com/volkszaehler/vzlogger/issues/384
#ExecReload=/bin/kill -HUP $MAINPID
StandardOutput=null
User=vzlogger
Group=vzlogger

[Install]
WantedBy=multi-user.target

```

Zugschlus, 2021-03-18:
> shouldn't it be worth improving the situation by getting systemd setups to use a proper systemd unit instead of the initscript via the generator?
> (it seems to me that that's a quicker path to success than getting the initscript to be fully compliant.)

systemd has not yet been completely accepted and is unlikely to be, there are many gallische Dörfer who refuse using systemd. A proper init script is, in my opinion, still needed for a proper piece of software. The most prominent issue with the init script, the prematurely returning do_stop, will also manifest in a proper sysv environment because restart won't work. I'd recommend mutating from a well-known-good init script from some other package.

PR #458 (2021-03-17): re-open logfile on SIGUSR1
partially fixes #383