volkszaehler/vzlogger #384: systemd unit reload fails

Zugschlus, 2019-06-15:
Hi,

vzlogger git contains etc/vzlogger.service which looks like a systemd unit. I took this to get rid of the init script, and promptly found out that the reload option offered by the unit doesn't work:

```
[52/3970]mh@lapis:~ $ pgrep vzlogger
21459
[53/3971]mh@lapis:~ $ sudo systemctl reload vzlogger
[54/3972]mh@lapis:~ $ pgrep vzlogger
(returned with exit code 1)
[56/3973]mh@lapis:~ $ sudo systemctl status vzlogger
● vzlogger.service - vzlogger
   Loaded: loaded (/etc/systemd/system/vzlogger.service; disabled; vendor preset: enabled)
   Active: failed (Result: signal) since Sat 2019-06-15 12:14:28 CEST; 5s ago
  Process: 21459 ExecStart=/usr/bin/vzlogger -c /etc/vzlogger.conf (code=killed, signal=SEGV)
  Process: 21500 ExecReload=/bin/kill -HUP $MAINPID (code=exited, status=0/SUCCESS)
 Main PID: 21459 (code=killed, signal=SEGV)

Jun 15 12:14:15 lapis systemd[1]: Started vzlogger.
Jun 15 12:14:27 lapis systemd[1]: Reloading vzlogger.
Jun 15 12:14:27 lapis systemd[1]: Reloaded vzlogger.
Jun 15 12:14:28 lapis systemd[1]: vzlogger.service: Main process exited, code=killed, status=11/
Jun 15 12:14:28 lapis systemd[1]: vzlogger.service: Failed with result 'signal'.

130 [57/3974]mh@lapis:~ $ 
```

Greetings
Marc

J-A-U, 2019-06-15:
Thats why we always used stop and start instead of reload.

And two issues to the same problem. Which one do you like to close?

Zugschlus, 2019-06-15:
None. Those are two different issues. One for the init script, and one for the systemd unit.

Also, stop/start might lose data.

Btw, I am hereby pulling back my offer to assist with the Debian packaging of vzlogger. You might also have to find another DD to sponsor your software into Debian. I am not interested to work with an upstream this arrogant.

andig, 2019-06-15:
> I am not interested to work with an upstream this arrogant.

Hallo Marc, bitte überlege Dir wie Du hier auftreten willst. Ich denke GitHub issues sind nicht dafür geeignet sich gegenseitig Vorwürfe zu machen.

mbehr1, 2019-08-14:
This issue (systemd reload failing) might be fixed with v 0.8.0 after integration of PR #388 as well. @Zugschlus are you willing to have another look into it? (I'd like to get this working properly but some testing support would be appreciated!)

Zugschlus, 2019-08-15:
Sure, just tell me which branch/tag/foo you want me to test.

frankrichter, 2019-08-15:
PR #388 is already merged, so git pull should be sufficient.

Zugschlus, 2019-08-15:
and the next release is going to be 0.8.0? Asking because of the version number of the package I build and install.

mbehr1, 2019-08-15:
hmm. I changed vzlogger version to 0.8.0 but I have no clue about how the debian packages work. but I assume it’s using the same version. 

> 
> and the next release is going to be 0.8.0? Asking because of the version number of the package I build and install.

Gruß

Matthias

Zugschlus, 2019-08-16:
Nosireebob.

```
[39/4402]mh@lapis:~/vzlogger $ sudo systemctl status vzlogger
[sudo] password for mh on lapis: 
● vzlogger.service - vzlogger
   Loaded: loaded (/lib/systemd/system/vzlogger.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2019-08-16 17:35:06 CEST; 30min ago
 Main PID: 14848 (vzlogger)
    Tasks: 23 (limit: 2366)
   Memory: 3.9M
   CGroup: /system.slice/vzlogger.service
           ├─14848 /usr/bin/vzlogger -c /etc/vzlogger.conf
           ├─18204 sh -c /home/mh/volkszaehler/exec-pvwr pvwr01.ka51.zugschlus.de 6 Gesamtenergie aktuell
           ├─18205 /bin/bash /home/mh/volkszaehler/exec-pvwr pvwr01.ka51.zugschlus.de 6 Gesamtenergie aktuell
           ├─18219 sleep 30
           ├─18220 sh -c /home/vz/volkszaehler/exec-pvwr pvwr02.ka51.zugschlus.de 6 Gesamtenergie aktuell
           ├─18221 /bin/bash /home/vz/volkszaehler/exec-pvwr pvwr02.ka51.zugschlus.de 6 Gesamtenergie aktuell
           └─18251 sleep 30

Aug 16 17:35:06 lapis systemd[1]: Started vzlogger.
[40/4402]mh@lapis:~/vzlogger $ sudo systemctl reload vzlogger
[41/4403]mh@lapis:~/vzlogger $ sudo systemctl status vzlogger
● vzlogger.service - vzlogger
   Loaded: loaded (/lib/systemd/system/vzlogger.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Fri 2019-08-16 18:05:53 CEST; 694ms ago
  Process: 14848 ExecStart=/usr/bin/vzlogger -c /etc/vzlogger.conf (code=exited, status=0/SUCCESS)
  Process: 18274 ExecReload=/bin/kill -HUP $MAINPID (code=exited, status=0/SUCCESS)
 Main PID: 14848 (code=exited, status=0/SUCCESS)

Aug 16 17:35:06 lapis systemd[1]: Started vzlogger.
Aug 16 18:05:53 lapis systemd[1]: Reloading vzlogger.
Aug 16 18:05:53 lapis systemd[1]: Reloaded vzlogger.
Aug 16 18:05:53 lapis systemd[1]: vzlogger.service: Succeeded.
3 [42/4404]mh@lapis:~/vzlogger $ sudo systemctl status vzlogger
● vzlogger.service - vzlogger
   Loaded: loaded (/lib/systemd/system/vzlogger.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Fri 2019-08-16 18:05:53 CEST; 5s ago
  Process: 14848 ExecStart=/usr/bin/vzlogger -c /etc/vzlogger.conf (code=exited, status=0/SUCCESS)
  Process: 18274 ExecReload=/bin/kill -HUP $MAINPID (code=exited, status=0/SUCCESS)
 Main PID: 14848 (code=exited, status=0/SUCCESS)

Aug 16 17:35:06 lapis systemd[1]: Started vzlogger.
Aug 16 18:05:53 lapis systemd[1]: Reloading vzlogger.
Aug 16 18:05:53 lapis systemd[1]: Reloaded vzlogger.
Aug 16 18:05:53 lapis systemd[1]: vzlogger.service: Succeeded.
3 [43/4404]mh@lapis:~/vzlogger $ pgrep vzlogger
1 [44/4405]mh@lapis:~/vzlogger $ sudo systemctl stop vzlogger
[45/4406]mh@lapis:~/vzlogger $ sudo systemctl start vzlogger
[46/4407]mh@lapis:~/vzlogger $ sudo systemctl status vzlogger
● vzlogger.service - vzlogger
   Loaded: loaded (/lib/systemd/system/vzlogger.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2019-08-16 18:06:17 CEST; 2s ago
 Main PID: 18795 (vzlogger)
    Tasks: 29 (limit: 2366)
   Memory: 4.8M
   CGroup: /system.slice/vzlogger.service
           ├─18795 /usr/bin/vzlogger -c /etc/vzlogger.conf
           ├─18826 sh -c /home/mh/volkszaehler/exec-pvwr pvwr01.ka51.zugschlus.de 6 Gesamtenergie aktuell
           ├─18827 /bin/bash /home/mh/volkszaehler/exec-pvwr pvwr01.ka51.zugschlus.de 6 Gesamtenergie aktuell
           ├─18848 sh -c /home/vz/volkszaehler/exec-modbus --tty /dev/USBserialModbus --baudrate 9600 --parity E --stopbits 2 --addr 0x47 0x4001 0x0000 0x0002 0x0004 0x0006 0x0008
           ├─18849 sh -c /home/vz/volkszaehler/exec-pvwr pvwr02.ka51.zugschlus.de 6 Gesamtenergie aktuell
           ├─18850 /bin/bash /home/vz/volkszaehler/exec-modbus --tty /dev/USBserialModbus --baudrate 9600 --parity E --stopbits 2 --addr 0x47 0x4001 0x0000 0x0002 0x0004 0x0006 0x0008
           ├─18852 /bin/bash /home/vz/volkszaehler/exec-pvwr pvwr02.ka51.zugschlus.de 6 Gesamtenergie aktuell
           ├─18862 /bin/bash /home/vz/volkszaehler/exec-modbus --tty /dev/USBserialModbus --baudrate 9600 --parity E --stopbits 2 --addr 0x47 0x4001 0x0000 0x0002 0x0004 0x0006 0x0008
           ├─18863 mbrtu -d/dev/USBserialModbus -b9600 -pE -s2 -Q -P -a0x47 -f3 -tint32 -n2 -r0x4001
           ├─18864 /bin/bash /home/vz/volkszaehler/exec-pvwr pvwr02.ka51.zugschlus.de 6 Gesamtenergie aktuell
           ├─18865 cut -d  -f3
           ├─18866 sleep 30
           ├─18867 curl --silent http://pvserver:thooW1ueKu@pvwr02.ka51.zugschlus.de/index.fhtml
           ├─18868 html2text
           └─18869 grep --max-count=1 aktuell.*Gesamtenergie

Aug 16 18:06:17 lapis systemd[1]: Started vzlogger.
[47/4408]mh@lapis:~/vzlogger $ 
```

mbehr1, 2019-08-16:
hmm. ok. also das stop (kill …) scheint jetzt zu gehen, aber manchmal startet systemd vzlogger dann nicht?
Schaue ich mir mal an.

> Nosireebob.
>
> [39/4402]mh@lapis:~/vzlogger $ sudo systemctl status vzlogger
> [sudo] password for mh on lapis:
> ● vzlogger.service - vzlogger
>    Loaded: loaded (/lib/systemd/system/vzlogger.service; [...]

Gruß

Matthias

Zugschlus, 2019-08-16:
So detailliert habe ich das nicht angeschaut, aber nach reload ist der Daemon tot. Wie reagiert der vzloggerd denn auf ein SIGHUP? Wenn Du das nicht abfängst ist der Daemon tot.

mbehr1, 2019-08-16:
SIGINT, SIGHUP, SIGTERM führen alle zum Ende.
Das erklärt das Problem schon mal.

Jetzt können wir entweder:
SIGHUP anders implementieren oder

das systemd recipe entsprechend anpassen.

Ich wäre für letzteres ;-)

> 
> So detailliert habe ich das nicht angeschaut, aber nach reload ist der Daemon tot. Wie reagiert der vzloggerd denn auf ein SIGHUP? Wenn Du das nicht abfängst ist der Daemon tot.

Gruß

Matthias

J-A-U, 2019-08-17:
SIGHUP ist bei daemons ja wohl dazu gedacht nicht den Prozess zu beenden sondern die Konfiguration neu zu laden.

Soweit ich weiß ist die Funktion reload nach systemd-Philosophie optional, unser Fehler ist das wir systemd anweisen es doch zu tun. Lösung in vzlogger.service:
`ExecReload=`

Auf Seiten vzlogger den SIGHUP abzufangen und einfach einen restart zu versuchen wird systemd auch durcheinander bringen weil die PID wechselt.
Die Anwendung reagiert korrekt wenn über systemd statt reload ein restart befohlen wird  (Alternativer Universalbefehl: systemctl reload-or-restart).

Zugschlus, 2019-08-17:
HUP war ursprünglich das Signal, das ein Prozess bekommen hat, wenn das Modem aufgelegt hat (Hangup). Das ist ein Überbleibsel aus der Zeit, als es noch der Normalfall war, ein Unix-System über eine Wählleitung anzusprechen. Seit ~ 25 Jahren (also länger als ich selbst Unix mache) ist die Konvention, das SIGHUP für den Reload der Konfiguration verwendet wird. Das muss aber im daemon selbst implementiert sein, indem man das Signal abfängt und die entsprechende Aktion ausführt.

Ein nicht abgefangenes Signal gleich welcher Art führt zum sofortigen Ableben des Prozesses. Wenn vzlogger HUP nicht abfängt, sollte zum Erreichen des erwartungsgemäßen Verhaltens auch niemand HUP an den Prozess schicken. => systemd unit und initscript anpassen und reload entfernen.

Übrigens empfiehlt systemd selbst, den reload-Prozess nicht durch SIGHUP durchzuführen, siehe man systemd.service und nach ExecReload suchen.

reload-or-restart oder noch schlimmer try-reload-or-restart werden meine Finger nie lernen, auch wenn es das richtige ist.

Grüße
Marc

J-A-U, 2019-08-17:
Dann würde ich sagen wir tun so. Mein vzlogger ist hier zwar noch nicht ganz aktuell, aber mit leerem ExecReload bekomm ich immerhin eine anständige Meldung statt eines gecrashten Daemon.

closed 2019-08-18 by J-A-U