volkszaehler/vzlogger #385: systemctl stop vzlogger results in timeout

frankrichter, 2019-06-26:
Stopping vzlogger (79402aeb96) via systemd takes 90s before it runs in a timeout. Here are two outputs of `systemctl status vzlogger` while waiting for systemctl stop to finish and after finishing with timeout:
```
pi@raspberrypi:~ $ systemctl status vzlogger
● vzlogger.service - vzlogger
   Loaded: loaded (/etc/systemd/system/vzlogger.service; enabled; vendor preset: enabled)
   Active: deactivating (stop-sigterm) since Wed 2019-05-15 22:58:49 CEST; 8s ago
 Main PID: 27331 (vzlogger)
   CGroup: /system.slice/vzlogger.service
           └─27331 /usr/local/bin/vzlogger -c /etc/vzlogger.conf

May 15 21:54:04 raspberrypi systemd[1]: Started vzlogger.
May 15 22:58:49 raspberrypi systemd[1]: Stopping vzlogger...

pi@raspberrypi:~ $ systemctl status vzlogger
● vzlogger.service - vzlogger
   Loaded: loaded (/etc/systemd/system/vzlogger.service; enabled; vendor preset: enabled)
   Active: failed (Result: timeout) since Wed 2019-05-15 23:00:19 CEST; 38s ago
  Process: 27331 ExecStart=/usr/local/bin/vzlogger -c /etc/vzlogger.conf (code=killed, signal=KILL)
 Main PID: 27331 (code=killed, signal=KILL)

May 15 21:54:04 raspberrypi systemd[1]: Started vzlogger.
May 15 22:58:49 raspberrypi systemd[1]: Stopping vzlogger...
May 15 23:00:19 raspberrypi systemd[1]: vzlogger.service: State 'stop-sigterm' timed out. Killing.
May 15 23:00:19 raspberrypi systemd[1]: vzlogger.service: Killing process 27331 (vzlogger) with signal SIGKILL.
May 15 23:00:19 raspberrypi systemd[1]: vzlogger.service: Main process exited, code=killed, status=9/KILL
May 15 23:00:19 raspberrypi systemd[1]: vzlogger.service: Failed with result 'timeout'.
May 15 23:00:19 raspberrypi systemd[1]: Stopped vzlogger.
```

mbehr1, 2019-06-27:
> Stopping vzlogger (79402ae) via systemd takes 90s before it runs in a timeout. Here are two outputs of systemctl status vzlogger while waiting for systemctl stop to finish and after finishing with timeout:

can you add the conf file? we have some blocking code (e.g. waiting for the next impulse/data) that might cause this. 

Mit freundlichen Grüßen / Best Regards
Matthias Behr

Sent from a mobile device.

andig, 2019-06-27:
Klingt wie das issue von Zugschluss zum systemd Skript bevor er schmollend abgedampft ist?

frankrichter, 2019-06-27:
Here is the conf file, but i use similar configurations on systems with older vzlogger builds which dont't show this problem.
```
{
    "retry": 30,            // how long to sleep between failed requests, in seconds
    "daemon": true,         // run periodically
    "verbosity": 0,        // between 0 and 15
    "log": "/var/log/vzlogger.log",     // path to logfile, optional

    "local": {
        "enabled": false,   // should we start the local HTTPd for serving live readings?
        "port": 8080,       // the TCP port for the local HTTPd
        "index": true,      // should we provide a index listing of available channels if no UUID was requested?
        "timeout": 30,      // timeout for long polling comet requests, 0 disables comet, in seconds
        "buffer": 600       // how long to buffer readings for the local interface, in seconds
    },

    "meters": [{
        "enabled": true,
        "skip": false,
        "protocol": "d0",
        "device": "/dev/lesekopf1",
        "aggtime": 60,
        "aggfixedinterval": true,
        "channels": [{
            "uuid": "",
            "middleware": "http://localhost/middleware.php",
            "identifier": "1-0:1.8.1",
            "aggmode": "MAX"
        },{
            "uuid": "",
            "middleware": "http://localhost/middleware.php",
            "identifier": "1-0:2.8.1",
            "aggmode": "MAX"
        }]
    },{
        "enabled": true,
        "skip": false,
        "protocol": "sml",
        "device": "/dev/lesekopf2",
        "aggtime": 60,
        "aggfixedinterval": true,
        "channels": [{
            "uuid": "",
            "middleware": "http://localhost/middleware.php",
            "identifier": "1-0:2.8.0",
            "aggmode": "MAX"
        }]
    }]
}
```

mbehr1, 2019-06-27:
hmm. so both meterS0 and meterSml wait blocking at least until the next package arrives from the meter.
So I assume your meter are sending data rarely only (once per min or even less)? Can you confirm this?
We could change the code to non-blocking but it’s a lot more difficult / error-prone.

> Here is the conf file, but i use similar configurations on systems with older vzlogger builds which dont't show this problem.
>
> {
>     "retry": 30,            // how long to sleep between failed [...]

Gruß

Matthias

frankrichter, 2019-06-27:
I can recheck this, but I'm quite sure both meters send packages every 2-3s.

frankrichter, 2019-06-27:
Both meters are sending new data about every 2s. Additionally I tried to disable one meter vice versa, but the behaviour didn't change noticeable.
 
1 minute Logfile: [vzlogger.log](https://github.com/volkszaehler/vzlogger/files/3336594/vzlogger.log)

frankrichter, 2019-06-27:
I compared the last logfile entries of a recent and an older vzlogger build. The newer version lacks the entries

```
[Jun 27 23:19:44][]     Server stopped.
[Jun 27 23:19:44][]     Trying to delete curlSessionProvider...
[Jun 27 23:19:44][]     deleted curlSessionProvider
```
The recent vzlogger is built and runs on raspbian buster (volkszaehler image). Could this be related to this issue? 
 
```
[Jun 27 20:13:04][main] vzlogger v0.7.0 based on heads/master-0-g79402aeb96 from Wed, 30 Jan 2019 15:56:19 +0100 started.
...
[Jun 27 20:14:24]       MapContainer::quit terminating on signal 15.
[Jun 27 20:14:24]       Closing connections to terminate
[Jun 27 20:14:24][main] MeterMap::cancel entered...
[Jun 27 20:14:24][main] MeterMap::cancel wait for readingthread
[Jun 27 20:14:24][main] MeterMap::cancel wait for meter::close
[Jun 27 20:14:24][main] MeterMap::cancel finished.
[Jun 27 20:14:24][main] MeterMap::cancel entered...
[Jun 27 20:14:24][main] MeterMap::cancel wait for readingthread
[Jun 27 20:14:24][main] MeterMap::cancel wait for meter::close
[Jun 27 20:14:24][main] MeterMap::cancel finished.
[Jun 27 20:14:24][main] MapContainer::quit finished.
```
```
[Jun 27 23:19:32][main] vzlogger v0.6.0 based on heads/master-0-g1484f9aad1 from Mon, 22 Aug 2016 22:32:51 +0200 started.
...
[Jun 27 23:19:44]       MapContainer::quit terminating on signal 15.
[Jun 27 23:19:44]       Closing connections to terminate
[Jun 27 23:19:44][main] MeterMap::cancel entered...
[Jun 27 23:19:44][main] MeterMap::cancel wait for readingthread
[Jun 27 23:19:44][main] MeterMap::cancel wait for meter::close
[Jun 27 23:19:44][main] MeterMap::cancel finished.
[Jun 27 23:19:44][main] MapContainer::quit finished.
[Jun 27 23:19:44][]     Server stopped.
[Jun 27 23:19:44][]     Trying to delete curlSessionProvider...
[Jun 27 23:19:44][]     deleted curlSessionProvider
```

mbehr1, 2019-06-28:
interesting. thx for the logs. will try to reproduce locally

Mit freundlichen Grüßen / Best Regards
Matthias Behr

Sent from a mobile device.

frankrichter, 2019-08-11:
Hallo Matthias,
weil es vom Thema her grad irgendwie passt: hast du mal versucht den Fehler nachzuvollziehen?
Viele Grüße
Frank

mbehr1, 2019-08-11:
Hallo Frank. Leider grad erst. Glaube aber, dass ich den Fehler verstanden habe. Siehe potentiellen Fix in PR#388. Wäre gut, wenn du den testen könntest.

frankrichter, 2019-08-11:
Hi Matthias,
eben getestet: leider keine offensichtliche Veränderung, weiterhin Timeout nach 90s. 
Viele Grüße
Frank

mbehr1, 2019-08-12:
hast du Logs?

Mit freundlichen Grüßen / Best Regards
Matthias Behr

Sent from a mobile device.

frankrichter, 2019-08-12:
Von gestern nicht, aber von eben. Einmal SML, einmal D0:
[vzlogger.log](https://github.com/volkszaehler/vzlogger/files/3494067/vzlogger.log)

mbehr1, 2019-08-12:
Hallo Frank, siehe mein Kommentar in #388. Habe noch ein größeres Problem gefunden. Werde das die nächsten Tage sauber überarbeiten.

mbehr1, 2019-08-12:
(Danke für neuen Log. Bestätigt meine Vermutung vom Kommentar in #388)

mbehr1, 2019-08-13:
@frankrichter habe den #388 noch mal aktualisiert mit "async-signal-safe" Lösung. Bei mir im Test beendet vzlogger sich jetzt bei killall vzlogger sauber wie gewünscht. Kannst du dein Setup noch mal damit testen? Danke!

frankrichter, 2019-08-13:
@mbehr1 sieht gut aus! vzlogger beendet sich jetzt ohne merkliche Verzögerung. Das Log endet mit

```
[Aug 13 23:31:18][]     main loop indicating all mappings to quit
[Aug 13 23:31:18][mtr0] MeterMap::cancel entered...
[Aug 13 23:31:18][mtr0] MeterMap::cancel wait for readingthread
[Aug 13 23:31:18][mtr0] MeterMap::cancel wait for meter::close
[Aug 13 23:31:18][mtr0] MeterMap::cancel finished.
[Aug 13 23:31:18][mtr1] MeterMap::cancel entered...
[Aug 13 23:31:18][mtr1] MeterMap::cancel wait for readingthread
[Aug 13 23:31:18][mtr1] MeterMap::cancel wait for meter::close
[Aug 13 23:31:18][mtr1] MeterMap::cancel finished.
[Aug 13 23:31:18][]     Server stopped.
[Aug 13 23:31:18][]     Trying to delete curlSessionProvider...
[Aug 13 23:31:18][]     deleted curlSessionProvider
```

mbehr1, 2019-08-14:
fixed with PR #388 in version 0.8.0.

mbehr1, 2019-08-14:
@frankrichter Danke fürs Testen!

closed 2019-08-14 by mbehr1