volkszaehler/vzlogger #386: vzlogger fails silently in curl'ing InfluxDB

dmesser, 2019-07-29:
I am running vzlogger in a container on a Raspberry Pi using debian-stretch:slim. Until recently everything has been fine reading from my ISKRA MT681 meter into InfluxDB running on another machine. But since a couple of days no data has been sent and the vzlogger container keeps crash-looping. Running interactively I get this with log-level 30:

```
sudo docker run --name vzlogger --rm -it --device=/dev/ttyUSB0:rw -v /etc/vzlogger.conf:/etc/vzlogger.conf:ro -v /var/log/vzlogger.log:/var/log/vzlogger.log:rw  dmesser/vzlogger:latest /usr/local/bin/vzlogger --config /etc/vzlogger.conf --verbose 30
[Jul 29 08:29:18][main] vzlogger v0.7.0 based on heads/master-0-g12e74ddd43 from Sun, 2 Jun 2019 20:48:14 +0200 started.
[Jul 29 08:29:18]       Start parsing configuration from /etc/vzlogger.conf
[Jul 29 08:29:18][main] log level is 30
[Jul 29 08:29:18][main] daemon=0, local=0
[Jul 29 08:29:18]       Process not  daemonized...
[Jul 29 08:29:18]       Opened logfile /var/log/vzlogger.log
[Jul 29 08:29:18][push] No pushDataServer defined.
[Jul 29 08:29:18][]     ===> Start meters
[Jul 29 08:29:18][mtr0] Meter connection established
[Jul 29 08:29:18][mtr0] Meter thread started
[Jul 29 08:29:18][mtr0] Number of readers: 32
[Jul 29 08:29:18][mtr0] Meter is opened. Starting channels.
[Jul 29 08:29:18][mtr0] Config.daemon: 0
[Jul 29 08:29:18][mtr0] Config.local: 0
[Jul 29 08:29:18][chn0] Logging thread started
[Jul 29 08:29:18][chn0] Start logging thread for influxdb-api. Running as daemon: no
[Jul 29 08:29:18][chn1] Logging thread started
[Jul 29 08:29:18][chn0] InfluxDB API initialize
[Jul 29 08:29:18][]     Startup done.
[Jul 29 08:29:18][chn0] api InfluxDB using host 192.168.1.35:8086
[Jul 29 08:29:18][chn1] Start logging thread for influxdb-api. Running as daemon: no
[Jul 29 08:29:18][chn0] api InfluxDB using username vzlogger-write
[Jul 29 08:29:18][chn1] InfluxDB API initialize
[Jul 29 08:29:18][chn0] api InfluxDB using database vzlogger
[Jul 29 08:29:18][chn1] api InfluxDB using host 192.168.1.35:8086
[Jul 29 08:29:18][chn1] api InfluxDB using username vzlogger-write
[Jul 29 08:29:18][chn0] api InfluxDB will use default measurement name "vzlogger"
[Jul 29 08:29:18][chn0] api InfluxDB using tags meter=gesamtverbrauch,unit=kwh
[Jul 29 08:29:18][chn0] api InfluxDB using curl timeout 30
[Jul 29 08:29:18][chn0] api InfluxDB will use default max_batch_inserts 4500
[Jul 29 08:29:18][chn0] api InfluxDB using max_buffer_size: 450000
[Jul 29 08:29:18][chn1] api InfluxDB using database vzlogger
[Jul 29 08:29:18][chn1] api InfluxDB will use default measurement name "vzlogger"
[Jul 29 08:29:18][chn0] api InfluxDB using url 192.168.1.35:8086/write?db=vzlogger&precision=ms
[Jul 29 08:29:18][chn1] api InfluxDB using tags meter=momentanverbrauch,unit=kw
[Jul 29 08:29:18][chn0] Using InfluxDB api
[Jul 29 08:29:18][chn1] api InfluxDB using curl timeout 30
[Jul 29 08:29:18][chn1] api InfluxDB will use default max_batch_inserts 4500
[Jul 29 08:29:18][chn1] api InfluxDB using max_buffer_size: 450000
[Jul 29 08:29:18][chn1] api InfluxDB using url 192.168.1.35:8086/write?db=vzlogger&precision=ms
[Jul 29 08:29:18][chn1] Using InfluxDB api
[Jul 29 08:29:19][mtr0] Got 4 new readings from meter:
[Jul 29 08:29:19][mtr0] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=6771455.10 ts=1564388959918
[Jul 29 08:29:19][mtr0] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=6771455.10 ts=1564388959918
[Jul 29 08:29:19][mtr0] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=0.00 ts=1564388959918
[Jul 29 08:29:19][mtr0] Reading: id=1-0:16.7.0*255/ObisIdentifier:1-0:16.7.0*255 value=363.00 ts=1564388959918
[Jul 29 08:29:19][chn0] Adding reading to queue (value=6771455.10 ts=1564388959918)
[Jul 29 08:29:19][chn1] Adding reading to queue (value=363.00 ts=1564388959918)
[Jul 29 08:29:19][mtr0] Stopped reading.
[Jul 29 08:29:19][chn0] Buffer has 1 items
[Jul 29 08:29:19][chn1] Buffer has 1 items
[Jul 29 08:29:19][chn0] Reading buffer: timestamp 1564388959918 value 6771455.100000
[Jul 29 08:29:19][chn1] Reading buffer: timestamp 1564388959918 value 363.000000
[Jul 29 08:29:19][chn0] request body is vzlogger,uuid=a1839c70-e946-11e8-8e20-b52cd95d444d,meter=gesamtverbrauch,unit=kwh value=6771455.100000 1564388959918

[Jul 29 08:29:19][chn1] request body is vzlogger,uuid=6c5a4660-e943-11e8-bb86-ad8d84144e21,meter=momentanverbrauch,unit=kw value=363.000000 1564388959918

[Jul 29 08:29:19][CURL] CURL:   Trying 192.168.1.35...
[Jul 29 08:29:19][CURL] CURL:   Trying 192.168.1.35...
[Jul 29 08:29:19][CURL] CURL: TCP_NODELAY set
[Jul 29 08:29:19][]     Server stopped.
[Jul 29 08:29:19][]     Trying to delete curlSessionProvider...
[Jul 29 08:29:19][CURL] CURL: Curl_http_done: called premature == 1
[Jul 29 08:29:19][CURL] CURL: stopped the pause stream!
[Jul 29 08:29:19][CURL] CURL: Closing connection 0
[Jul 29 08:29:19][CURL] CURL: Curl_http_done: called premature == 1
[Jul 29 08:29:19][CURL] CURL: stopped the pause stream!
[Jul 29 08:29:19][CURL] CURL: Closing connection 0
[Jul 29 08:29:19][]     deleted curlSessionProvider
```

This is my vzlogger.conf with redacted passwords:

```
{
"retry" : 3,
"daemon": false,
"verbosity" : 3,
"log" : "/var/log/vzlogger.log",

"local" : {
    "enabled" : false,
    "port" : 8080,
    "index" : true,
    "timeout" : 30,
    "buffer" : 600
},

"meters" : [
        {
        "enabled" : true,
        "device" : "/dev/ttyUSB0",
        "protocol" : "sml",
        "parity" : "8N1",
    "baudrate" : 9600,
    "aggtime" : -1,
    "aggfixedinterval" : true,
        "channels": [{
			"api" : "influxdb",
			"host": "192.168.1.35:8086",
			"database" : "vzlogger",
			"username" : "vzlogger-write",
			"password" : "*******",
			"max_buffer_size" : 450000,
			"timeout": 30,
			"uuid": "a1839c70-e946-11e8-8e20-b52cd95d444d",
			"identifier" : "1-0:1.8.0",
			"tags" : "meter=gesamtverbrauch,unit=kwh"
			},
            {
            "api" : "influxdb",
            "host": "192.168.1.35:8086",
            "database" : "vzlogger",
            "username" : "vzlogger-write",
            "password" : "*******",
            "max_buffer_size" : 450000,
            "timeout": 30,
            "uuid": "6c5a4660-e943-11e8-bb86-ad8d84144e21",
            "identifier" : "1-0:16.7.0",
			"tags" : "meter=momentanverbrauch,unit=kw"
            }]
        }
    ]
}
```

Parsing the source I narrowed it down to curl not working. I can curl the InfluxDB from the PI itself fine, and it appears in InfluxDB logs. Running vzlogger there are no log entries on the InfluxDB side and I presume it's failing silently somewhere trying to curl. Any ideas?

andig, 2019-07-29:
Can you curl from inside the vzlogger container?

Viele Grüße,
Andreas

dmesser, 2019-07-29:
Yep, when passing thru the binary from the host OS (because `curl` is not part of the image):

```
pi@raspberrypi:~ $ sudo docker run --name vzlogger --rm -it \
 --device=/dev/ttyUSB0:rw -v /etc/vzlogger.conf:/etc/vzlogger.conf:ro \
 -v /var/log/vzlogger.log:/var/log/vzlogger.log:rw -v /usr/bin/curl:/usr/bin/curl:ro  \
 dmesser/vzlogger:latest \
 /usr/bin/curl --user vzlogger-write:******* -X POST -d "vzlogger,uuid=6c5a4660-e943-11e8-bb86-ad8d84144e21,meter=momentanverbrauch,unit=kw value=419.000000 1564387513619" "http://192.168.1.35:8086/write?db=vzlogger&precision=ms"

pi@raspberrypi:~ $ echo $?
0
```

andig, 2019-07-29:
ping @mbehr1

dmesser, 2019-07-30:
Funny enough, repeated runs of above containers produce different output in regards to `curl`:

Examples:

```
[...]
[Jul 29 22:38:44][chn0] Adding reading to queue (value=6777258.30 ts=1564439924183)
[Jul 29 22:38:44][chn1] Adding reading to queue (value=446.00 ts=1564439924183)
[Jul 29 22:38:44][mtr0] Stopped reading.
[Jul 29 22:38:44][chn0] Buffer has 1 items
[Jul 29 22:38:44][chn0] Reading buffer: timestamp 1564439924183 value 6777258.300000
[Jul 29 22:38:44][chn0] request body is vzlogger,uuid=a1839c70-e946-11e8-8e20-b52cd95d444d,meter=gesamtverbrauch,unit=kwh value=6777258.300000 1564439924183

[Jul 29 22:38:44][CURL] CURL: Expire in 0 ms for 6 (transfer 0x74e05af0)
[Jul 29 22:38:44][chn1] Buffer has 1 items
[Jul 29 22:38:44][]     Server stopped.
[Jul 29 22:38:44][]     Trying to delete curlSessionProvider...
```

```
[...]
[Jul 29 22:38:58][mtr0] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=6777260.10 ts=1564439938617
[Jul 29 22:38:58][mtr0] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=6777260.10 ts=1564439938617
[Jul 29 22:38:58][mtr0] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=0.00 ts=1564439938617
[Jul 29 22:38:58][mtr0] Reading: id=1-0:16.7.0*255/ObisIdentifier:1-0:16.7.0*255 value=456.00 ts=1564439938617
[Jul 29 22:38:58][chn0] Adding reading to queue (value=6777260.10 ts=1564439938617)
[Jul 29 22:38:58][chn1] Adding reading to queue (value=456.00 ts=1564439938617)
[Jul 29 22:38:58][chn0] Buffer has 1 items
[Jul 29 22:38:58][chn0] Reading buffer: timestamp 1564439938617 value 6777260.100000
[Jul 29 22:38:58][chn0] request body is vzlogger,uuid=a1839c70-e946-11e8-8e20-b52cd95d444d,meter=gesamtverbrauch,unit=kwh value=6777260.100000 1564439938617

[Jul 29 22:38:58][CURL] CURL: Expire in 0 ms for 6 (transfer 0x74e05af0)
[Jul 29 22:38:58][CURL] CURL: Expire in 30000 ms for 8 (transfer 0x74e05af0)
[Jul 29 22:38:58][CURL] CURL:   Trying 192.168.1.35...
[Jul 29 22:38:58][CURL] CURL: TCP_NODELAY set
[Jul 29 22:38:58][CURL] CURL: Expire in 200 ms for 4 (transfer 0x74e05af0)
[Jul 29 22:38:58][mtr0] Stopped reading.
[Jul 29 22:38:58][CURL] CURL: Connected to 192.168.1.35 (192.168.1.35) port 8086 (#0)
[Jul 29 22:38:58][CURL] CURL: Server auth using Basic with user 'vzlogger-write'
[Jul 29 22:38:58][chn1] Buffer has 1 items
[Jul 29 22:38:58][]     Server stopped.
[Jul 29 22:38:58][]     Trying to delete curlSessionProvider...
[Jul 29 22:38:58][CURL] CURL: Closing connection 0
[Jul 29 22:38:58][]     deleted curlSessionProvider
```

```
[...]
[Jul 29 22:39:14][mtr0] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=6777262.10 ts=1564439954631
[Jul 29 22:39:14][mtr0] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=6777262.10 ts=1564439954631
[Jul 29 22:39:14][mtr0] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=0.00 ts=1564439954631
[Jul 29 22:39:14][mtr0] Reading: id=1-0:16.7.0*255/ObisIdentifier:1-0:16.7.0*255 value=440.00 ts=1564439954631
[Jul 29 22:39:14][chn0] Adding reading to queue (value=6777262.10 ts=1564439954631)
[Jul 29 22:39:14][chn1] Adding reading to queue (value=440.00 ts=1564439954631)
[Jul 29 22:39:14][mtr0] Stopped reading.
[Jul 29 22:39:14][chn0] Buffer has 1 items
[Jul 29 22:39:14][chn1] Buffer has 1 items
[Jul 29 22:39:14][chn0] Reading buffer: timestamp 1564439954631 value 6777262.100000
[Jul 29 22:39:14][chn0] request body is vzlogger,uuid=a1839c70-e946-11e8-8e20-b52cd95d444d,meter=gesamtverbrauch,unit=kwh value=6777262.100000 1564439954631
[Jul 29 22:39:14][chn1] Reading buffer: timestamp 1564439954631 value 440.000000
[Jul 29 22:39:14][chn1] request body is vzlogger,uuid=6c5a4660-e943-11e8-bb86-ad8d84144e21,meter=momentanverbrauch,unit=kw value=440.000000 1564439954631
[Jul 29 22:39:14][]     Server stopped.
[Jul 29 22:39:14][]     Trying to delete curlSessionProvider...
[Jul 29 22:39:14][]     deleted curlSessionProvider
```

```
[...]
[Jul 29 22:39:32][mtr0] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=6777264.30 ts=1564439972294
[Jul 29 22:39:32][mtr0] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=6777264.30 ts=1564439972294
[Jul 29 22:39:32][mtr0] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=0.00 ts=1564439972294
[Jul 29 22:39:32][mtr0] Reading: id=1-0:16.7.0*255/ObisIdentifier:1-0:16.7.0*255 value=446.00 ts=1564439972294
[Jul 29 22:39:32][chn0] Adding reading to queue (value=6777264.30 ts=1564439972294)
[Jul 29 22:39:32][chn1] Adding reading to queue (value=446.00 ts=1564439972294)
[Jul 29 22:39:32][mtr0] Stopped reading.
[Jul 29 22:39:32][chn1] Buffer has 1 items
[Jul 29 22:39:32][chn1] Reading buffer: timestamp 1564439972294 value 446.000000
[Jul 29 22:39:32][chn1] request body is vzlogger,uuid=6c5a4660-e943-11e8-bb86-ad8d84144e21,meter=momentanverbrauch,unit=kw value=446.000000 1564439972294

[Jul 29 22:39:32][CURL] CURL: Expire in 0 ms for 6 (transfer 0x74405af0)
[Jul 29 22:39:32][CURL] CURL: Expire in 30000 ms for 8 (transfer 0x74405af0)
[Jul 29 22:39:32][]     Server stopped.
[Jul 29 22:39:32][]     Trying to delete curlSessionProvider...
[Jul 29 22:39:32][]     deleted curlSessionProvider
```

Looks like some sort of race condition is kicking in.

dmesser, 2019-08-05:
The problem is also reproducible outside of the container environment, on an up-to-date Raspbian (Debian Stretch).

andig, 2019-08-05:
I‘m confused by the „server stopped“ messages too. Which server? Why?
Looks like we‘d need better instrumentation in the influx provider to see whats going on. 

Viele Grüße,
Andreas

dmesser, 2019-08-05:
I was watching the network stack via `tcpdump` during this last test (no container) and it's not showing any outgoing packages to Influx - I don't think `curl` is every successfully able to make a call and somehow fails silently.

andig, 2019-08-05:
Then https://github.com/volkszaehler/vzlogger/blob/master/src/api/InfluxDB.cpp would be the place to start and add more logging.

/cc @muffl0n @Stefan-Code @okrause

andig, 2019-08-05:
> I was watching the network stack via tcpdump during this last test

@dmesser you might also just implement a "dumb" PHP endpoint that accepts and logs any GET and POST request- I feel influxdb is not much more than that. See if that works, if not its probably not specific to influxdb but rather network-related.

dmesser, 2019-08-05:
Yeah, I've found the `tcpdump` approach much faster though. Right now vzlogger is not sending any TCP packets.

okrause, 2019-08-07:
I am an deliberate C++ noob, so I am of little help. The logs show that both channels often get data at the same time and two vz::api::InfluxDB::send() are running in parallel.

I would suggest the do another test with only chn0 configured (comment out the other channel in vzlogger.conf). If it is a race condition in vz::api::InfluxDB::send(), it should work with only one channel configured. Then I would assume it is connected to buf handling. If it still fails, I would argue that the problem it is not in vz::api::InfluxDB::send(), but in the layers below (curl, network).

dmesser, 2019-08-10:
@okrause thanks for the suggestion. I tried with a single channel only and the result is the same. I concur this seems like a problem at some lower level (IMHO `curl`). Interestingly enough this was working on an x86_64 machine with the same OS (Debian 9 Stretch). It had the following versions of `libcurl` installed:

```
libcurl3/stable,stable,now 7.52.1-5+deb9u8 amd64 [installed,automatic]                                                   
libcurl4-openssl-dev/stable,stable,now 7.52.1-5+deb9u8 amd64 [installed]
```

Whereas this RPI environment has the following:

```
libcurl3/oldstable,now 7.52.1-5+deb9u9 armhf [installed,automatic]
libcurl3-gnutls/oldstable,now 7.52.1-5+deb9u9 armhf [installed,automatic]
libcurl4-openssl-dev/oldstable,now 7.52.1-5+deb9u9 armhf [installed]
```

mbehr1, 2019-08-11:
I think the problem is the following:

you’re not running vzlogger as a daemon. So it will try to do just one reading and then exit.
But the sendind is done async. 
This sending is aborted by the shutdown after the one readings have been read.

 [Jul 29 08:29:19][] Trying to delete curlSessionProvider…   <- here the two requests are still pending and get cancelled.

> [Jul 29 08:29:19][CURL] CURL: Curl_http_done: called premature == 1
> [Jul 29 08:29:19][CURL] CURL: stopped the pause stream!
> [Jul 29 08:29:19][CURL] CURL: Closing connection 0
> [Jul 29 08:29:19][CURL] CURL: Curl_http_done: called premature == 1
> [Jul 29 08:29:19][CURL] CURL: stopped the pause stream!
> [Jul 29 08:29:19][CURL] CURL: Closing connection 0
> [Jul 29 08:29:19][] deleted curlSessionProvider
> This is my vzlogger.conf with redacted passwords:
> 
> {
> "retry" : 3,
> "daemon": false,   <- not as daemon

Can you try to enable daemon mode and see whether that removes/hides the problem?

> @okrause <https://github.com/okrause> thanks for the suggestion. I tried with a single channel only and the result is the same. I concur this seems like a problem at some lower level (IMHO curl). [...]

Gruß

Matthias

mbehr1, 2019-08-11:
A better fix would be to wait a few seconds at shutdown if the connection is still in use:
CurlSessionProvider::~CurlSessionProvider()
{
    // curl_easy_cleanup for each CURL*
    timespec ts;
    clock_gettime(CLOCK_REALTIME , &ts);
    ts.tv_sec+=1;
    pthread_mutex_timedlock(&_map_mutex, &ts); // try max 1s to acquire the lock. There might have been another thread pthread_cancelled while owning the lock.
    for (map_it it = _easy_handle_map.begin(); it!=_easy_handle_map.end(); ++it)
    {
        CurlUsage cu = (*it).second;

would need to wait here is cu.inUse!

        curl_easy_cleanup(cu.eh);
    }
    curl_global_cleanup();

    pthread_mutex_unlock(&_map_mutex);
    pthread_mutex_destroy(&_map_mutex);
}

I can add a patch for waiting a few seconds at shutdown until the connection is not in use any longer.
How long does the access to your database usually take? (How long shall we wait at shutdown?)

> 
> I think the problem is the following:
> 
> you’re not running vzlogger as a daemon. So it will try to do just one reading and then exit.
> But the sendind is done async. 
> This sending is aborted by the shutdown after the one readings have been read.
> 
>  [Jul 29 08:29:19][] Trying to delete curlSessionProvider…   <- here the two requests are still pending and get cancelled.
> > [Jul 29 08:29:19][CURL] CURL: Curl_http_done: called premature == 1
> > [Jul 29 08:29:19][CURL] CURL: stopped the pause stream!
> > [Jul 29 08:29:19][CURL] CURL: Closing connection 0
> > [Jul 29 08:29:19][CURL] CURL: Curl_http_done: called premature == 1
> > [Jul 29 08:29:19][CURL] CURL: stopped the pause stream!
> > [Jul 29 08:29:19][CURL] CURL: Closing connection 0
> > [Jul 29 08:29:19][] deleted curlSessionProvider
> > This is my vzlogger.conf with redacted passwords:
> > 
> > {
> > "retry" : 3,
> > "daemon": false,   <- not as daemon
> 
> 
> Can you try to enable daemon mode and see whether that removes/hides the problem?
> 
>> 
>> @okrause <https://github.com/okrause> thanks for the suggestion. I tried with a single channel only and the result is the same. I concur this seems like a problem at some lower level (IMHO curl). Interestingly enough this was working on an x86_64 machine with the same OS (Debian 9 Stretch). It had the following versions of libcurl installed:
>> 
>> libcurl3/stable,stable,now 7.52.1-5+deb9u8 amd64 [installed,automatic]                                                   
>> libcurl4-openssl-dev/stable,stable,now 7.52.1-5+deb9u8 amd64 [installed]
>> Whereas this RPI environment has the following:
>> 
>> libcurl3/oldstable,now 7.52.1-5+deb9u9 armhf [installed,automatic]
>> libcurl3-gnutls/oldstable,now 7.52.1-5+deb9u9 armhf [installed,automatic]
>> libcurl4-openssl-dev/oldstable,now 7.52.1-5+deb9u9 armhf [installed]
>> 
> 
> Gruß
> 
> Matthias

Gruß

Matthias

mbehr1, 2019-08-11:
See (potential) fix in PR#387. Can you give it a try? (@dmesser)

dmesser, 2019-08-11:
@mbehr1 thanks for that PR - I can confirm that it's now sending data. Also in daemon mode it's not working at least on the PI itself (outside a container). Will report back when it's also working when running in a container.

dmesser, 2019-08-11:
So it's working in the container again now as well, from the original source code base. I am not entirely sure what was wrong in the end - it works flawlessly now. Thank you for bearing with me.

I guess I got confused with your meaning of the term `daemon`. In traditional UNIX world it means that a process is running in the background, detached from any TTY. It is not tied to the nature of the processing: one-off vs. continuous.
Therefore, not to daemonize a process however shouldn't mean that the utility stops after a single reading but rather runs in the foreground printing all output to stdout/stderr. This way you are not forced to read logs from a file (especially cumbersome when running on a platform like k8s or generally in a container).

I opened #389 and will close this issue. Thanks for all your help.

PR #387 (2019-08-11): If a curl session is still in use retry up to five times.
To fix #386 we wait up to 5 times with 1sec in between (so max 4s)
at shutdown before we cancel the pending curl sessions.
Issue: #386
commit 2db1af9b (2019-08-11): If a curl session is still in use retry up to five times. (#387)

To fix #386 we wait up to 5 times with 1sec in between (so max 4s)
at shutdown before we cancel the pending curl sessions.
Issue: #386
closed 2019-08-11 by dmesser