volkszaehler/vzlogger #389: vzlogger should have a --one-time CLI switch

dmesser, 2019-08-11:
Related to #378 the behavior of `daemon: false` is highly unintuitive. A users normal expectation towards software that reads metering data and forwards this somewhere else is that it does this continuously under normal conditions. A user is usually interested in continuous metering, not a single value.

Therefore I suggest that the default behavior of vzlogger without setting `daemon: true` explicitly is that it runs in the foreground, printing all output to the attached terminal. With `daemon: true` set, the process should detach from the terminal and forward all output to the log file.
The one-time mode can be obtained via CLI switch `--one-time` that should work independently of daemon mode.

This aids debugging and conformance with regular UNIX daemons.

r00t-, 2019-08-11:
if you want to confuse yourself with the full history if this, read #40 #93 #94 .
specifically https://github.com/volkszaehler/vzlogger/issues/94#issuecomment-75153851

dmesser, 2019-08-11:
@r00t- yeah I've skimmed over this before I posted. I don't thing a `foreground` option makes sense - `daemon:` should be enough to control this. I think it's well understood behavior in the FOSS and Linux world.

r00t-, 2019-08-11:
@dmesser: i'm only trying to point out that the issue arises because vzlogger gives a different meaning to "daemon" - which may very well be a bug.

r00t-, 2019-08-11:
imho, it would make more sense to have a single ticket to fix the whole thing to conform to usual conventions, instead of single tickets demanding new features.

this would probably mean:
 * remove "deamon" setting (default to true, warning/error if set to false?)
 * add single-shot mode switch #389
 * add foreground mode switch #378

(i wonder if anybody actually ever uses daemon:false mode.)

J-A-U, 2019-08-11:
Darf ich mich in deutsch dazugesellen? Ich fürchte mich sonst falsch auszudrücken.

Soweit ich mich erinnere ist der Umstand das vzlogger bei daemon:false nur einen Durchgang macht eigentlich ein Bug. Nebeneffekt ist dabei auch das keine Daten an die Middleware geschickt werden weil der Prozess zu früh abgebrochen wird.
Wurde aber zum Feature erklärt weil es für Diagnosezwecke nicht ganz unpraktisch ist. Und weil sich bislang noch keiner gefunden hat das eigentliche Problem zu beseitigen.

Gut möglich das ich was durcheinander bringe oder nicht ganz verstehe, also bitte nicht all zu viel Gewicht auf meine Schlussfolgerungen legen.

dmesser, 2019-08-11:
@r00t- I'd use `daemon:false` on any container platform where I would like the container orchestration platform to grab the logs from stdout and also tie the lifecycle of the vzlogger process to that of the container / pod.

andig, 2019-08-11:
Wofür brauchen wir egtl. foreground? Singleshot würde ja reichen um zu beweisen dass die Config funktioniert.  Bliebe nur Singleshot („once“) als Option übrig.

andig, 2019-08-11:
Oder wir lassen das Daemon Verhalten einfach komplett weg- mache ich bei mbmd auch ohne dass es bisher jemand vermisst hätte...

cweiske, 2019-08-27:
Ich würde so eine "one-time"-Option gern nutzen, weil ich den Stromverbrauch nur 1x am Tag auslesen und loggen möchte.
`daemon:false` führt wie oben beschrieben dazu, daß keine Daten an die Middleware geschickt werden und ist deshalb für meine Zwecke nicht zu gebrauchen.

J-A-U, 2019-08-28:
Hast du mal mit entsprechend großer aggtime oder interval versucht?

cweiske, 2019-08-28:
> Hast du mal mit entsprechend großer aggtime oder interval versucht?

Anstatt das per cron zu machen?
Mein "Problem" ist, daß der vzlogger permanent 2% CPU benötigt, wenn er läuft. Das würde er weiterhin tun, wenn ich die aggtime auf einen Tag setze.
Cron wäre ideal, um hier CPU-Leistung und damit Strom zu sparen :)

andig, 2019-08-28:
Find ich jetzt nicht schlimm, aber ich frag mich wofür die 2% anfallen. Um was für einen Zähler geht es eigentlich? Sml? Den sollte man ja auch per interval auslesen können und dann egtl. agr keine Rechenleistung brauchen?

cweiske, 2019-08-29:
Ja, ist ein SML-Zähler.

BiasF, 2020-10-20:
Würde einen One-Shot mode auch begrüßen
Ich kann meinen Zähler nur etwa pro Stunde abrufen, da ich das ganze über einen ESP8266 mit Batterie über Ser2Net betreiben muss, da ich keine Steckdose im Zählerschrank habe. 
Intervall wird wohl nicht funktionieren, da man das ziemlich exakt mit dem DeepSleep synchronisieren müsste.

Außerdem peilt vzlogger es nicht wenn der ESP offline ist und somit die Telnet Verbindung wegbricht.
Es wird im daemon mode dann keine neue aufgebaut. 

Meine Idee war daher das der ESP selber den read Vorgang per MQTT triggert.

juergen4mueller, 2021-02-21:
The readme file still tells me that there are two modes available: „can run as a daemon or via cron“. Obviously it’s not working at the moment. 
Are there plans to make the cron mode running again or do I have to make a workaround? For me a single shot mode would also be perfect.

r00t-, 2021-02-21:
@juergen4mueller:
#449 does point out this documentation bug aswell.

as you can see if you read the discussion there, that issue is kind of deadlocked, with no developer picking it up for years.
i guess you have to live with what's there, or patch the code locally to behave as you require.
you are ofcourse also invited to send a merge-request with a possible fix, that might help to get this topic moving again.

r00t-, 2021-02-22:
(the trivial part of ditching "daemon" and adding foreground mode is now in #450 .)

single-shot mode is actually quite a bit more complex.

as pointed out before somewhere, if we exit the reading thread(s),
we still somehow need to ensure that the logging thread uploads the data (and then exits).

also,
basically, only periodic meters (where vzlogger itself can choose the poll intervall) can be supported.
(s0/sml is 'periodic' only when sending a pullseq.)
for others we may get away with simply waiting for the first reading, but the time until that is received is undefined.
some meters would need very special handling.
for example, the `s0` meter is not periodid, and it will, with the first reading, only output one impulse, which would be completely useless.
we would have to wait for the second reading, in which it at least calculates power.

r00t-, 2021-02-22:
for anybody feeling extremely adventurous,  #450 has an implementation of single-shot mode now,
i tested this ONLY with protocol=random and api=null.
if you try anything else and it crashes or doesn't work, don't bother posting.