volkszaehler/vzlogger #394: Compile error Buffer.cpp OpenSuse Tumbleweed

opensprinklershop, 2019-09-20:
When compiling vzlogger on OpenSuse Tumbleweed (09-2019):

[ 31%] Building CXX object src/CMakeFiles/vz.dir/Buffer.cpp.o
cd /opt/vzlogger/build/src && /usr/bin/c++  -DHAVE_CONFIG_HPP -I/opt/vzlogger/build -I/opt/vzlogger/include -I/opt/vzlogger  -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++11 -fpermissive -Wno-error=redundant-decls -Wno-ignored-qualifiers   -g3 -o CMakeFiles/vz.dir/Buffer.cpp.o -c /opt/vzlogger/src/Buffer.cpp
/opt/vzlogger/src/Buffer.cpp: In member function âvoid Buffer::aggregate(int, bool)â:
/opt/vzlogger/src/Buffer.cpp:119:25: error: implicitly-declared âReading& Reading::operator=(const Reading&)â is deprecated [-Werror=deprecated-copy]
  119 |      else *_last_avg = *it;
      |                         ^~
In file included from /opt/vzlogger/include/Buffer.hpp:35,
                 from /opt/vzlogger/src/Buffer.cpp:37:
/opt/vzlogger/include/Reading.hpp:145:2: note: because âReadingâ has user-provided âReading::Reading(const Reading&)â
  145 |  Reading(const Reading &orig);
      |  ^~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [src/CMakeFiles/vz.dir/build.make:105: src/CMakeFiles/vz.dir/Buffer.cpp.o] Error 1
make[2]: Leaving directory '/opt/vzlogger/build'
make[1]: *** [CMakeFiles/Makefile2:945: src/CMakeFiles/vz.dir/all] Error 2
make[1]: Leaving directory '/opt/vzlogger/build'
make: *** [Makefile:166: all] Error 2

mbehr1, 2019-09-20:
I’ll take a look. (but only in a few days, do you need a quick solution ?)

Mit freundlichen Grüßen / Best Regards
Matthias 

Sent from a mobile device.

mbehr1, 2019-09-22:
see PR #395. @opensprinklershop  can you try whether this fixes the warning? Thanks!

andig, 2019-11-03:
ping @opensprinklershop any news?

closed 2019-11-05 by andig