volkszaehler/vzlogger #396: Problem with OBIS identifier in sml

winfriedp, 2019-11-03:
Within sml protocol works only a full qualified OBIS identifier, e.g. "1-0:1.8.0". Identifier "1.8.0" or alias "counter" doesn't work. In d0 protocol was a "1.8.0" ok. I can not test if the alias worked there.

andig, 2019-11-03:
/cc @mbehr1 könntest Du Dir das anschauen?

mbehr1, 2019-11-05:
yep. any log available?

mbehr1, 2019-11-05:
I could add a workaround to add parsing like with D0 (I'd try the current way and if the Obis is not valid (which is currently the case for short forms or alias) I could try the string parsing as in D0 and check whether that returns a valid obis.) But I have no means to test it... So any log would be good or if @winfriedp can support in testing.

winfriedp, 2019-11-05:
Attached a file with logs of several identifiers configurations:
[Logs with several identifiers.txt](https://github.com/volkszaehler/vzlogger/files/3811403/Logs.with.several.identifiers.txt)
I can hopefully support the test.

mbehr1, 2019-11-16:
hmm. I think this is due to a change done in 2014: 3d28083466b852c6a684ddd7bff44d57b0c95708

mbehr1, 2019-11-16:
so this was explicitly decided as part of PR #71

mbehr1, 2019-11-16:
so if you use "1.8.0" it will be
   "ObisIdentifier:255-255:1.8.0*255"
whereas your meter returns 
  "ObisIdentifier:1-0:1.8.0*255"
If you use "counter" it's 
 "ObisIdentifier:1-0:1.8.255*255"
...

mbehr1, 2019-11-16:
I could add a logic that identifiers can be specified e.g. as "counter*" or "1.8.0*" (or * in front) and then wildcard comparison is enabled for that one. Opinions? @r00t- @andig ?