volkszaehler/vzlogger #397: Within sml protocol parameter "duplicates" has no effect

winfriedp, 2019-11-03:
Within the sml protocol: even though specified the parameter duplicates > 0 (here "duplicates": 1800) identical values are sent to the api (here HTTPd) every second.
Attached my configuration:
[vzlogger.conf.issueDuplicates.txt](https://github.com/volkszaehler/vzlogger/files/3801415/vzlogger.conf.issueDuplicates.txt)

andig, 2019-11-05:
From the ML: Frank hat geschrieben, dass "duplicates" nur für die Volkszähler-API implementiert ist. Wenn dem so ist, habe ich die Anforderung "duplicates" für HTTPd zu implementieren, so dass ich die Funktion für sml nutzen kann.

grseidel, 2022-12-15:
Die Implementierung dieses Features wäre natürlich auch für die api influxdb wünschenswert.