volkszaehler/vzlogger #398: vzlogger ignores "direction" for active power

michis0806, 2019-11-15:
Hello, 

I use vzlogger to read out my two electricity meters (EMH eHZ). One is a two-way meter for feed-in and consumption. The value at 15.7.0 of the two-way counter is always positive, although the sign in the display is A-.

Is this a known issue?

Michael

mbehr1, 2019-11-15:
do you have a trace/log?

J-A-U, 2019-11-15:
> Is this a known issue?

Kind of. We can only process what the counter delivers. Unfortunately its very common only positive values are given for power and current.

michis0806, 2019-11-15:
Thanks for your responses. I have a log but it shows only positive values:
This should be "negative" as the photovoltaik delivers ~5000W and I use ~800
`Reading: id=1-0:15.7.0*255/ObisIdentifier:1-0:15.7.0*255 value=4198.90 ts=1573822799591`
This is correct as I use ~2500W
`Reading: id=1-0:15.7.0*255/ObisIdentifier:1-0:15.7.0*255 value=2469.70 ts=1573838752750`

I read somewhere™️ that there is a hex-marker in the raw-readings which identifies the direction but I lost the link :( 

I dare remember something that the line ends with 82 instead of Ax? 

Michael

michis0806, 2019-11-15:
Ah, found the bit:
https://www.photovoltaikforum.com/thread/102181-sml-openhab-binding-für-ehz-stromzähler/?postID=1135313#post1135313

It says, that 

> I once contacted Itron and got the answer that the meters could not be configured and that they were developed according to an FNN specification.

>In fact, after some research in the network about this FNN requirement specification, it turns out that the amount of active power is always output; the energy direction is coded in status, i.e. byte 150 or 226, 'A2', coded binary, etc. the energy direction. With '82' it would therefore be consumption. I check this as soon as I have time...

mbehr1, 2019-11-15:
Isn’t 15.7.0 defined as
15.7.0	Absolute active instantaneous power (|A|) [kW]

(any other readings available?)
Mit freundlichen Grüßen / Best Regards
Matthias Behr

Sent from a mobile device.

michis0806, 2019-11-15:
Yes, I have

`Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=23038999.10 ts=1573505429417
Reading: id=1-0:2.8.0*255/ObisIdentifier:1-0:2.8.0*255 value=65730629.60 ts=1573505429417
Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=23038999.10 ts=1573505429417
Reading: id=1-0:2.8.1*255/ObisIdentifier:1-0:2.8.1*255 value=65730629.60 ts=1573505429417
Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=0.00 ts=1573505429417
Reading: id=1-0:15.7.0*255/ObisIdentifier:1-0:15.7.0*255 value=507.80 ts=1573505429417`

but the others are totals, not current values...

frankrichter, 2019-11-15:
Do you really need the power reading? volkszaehler middleware calculates
power values from totals. This is what we usually recommend.

>

michis0806, 2019-11-17:
It would be great to have the correct values. Im setting up my own home-surveillance system...

andig, 2020-01-20:
> @frankrichter <https://github.com/frankrichter> I tried volkszaehler
> middleware now some time but I have some problems with the
> data/visualisation. There are often some negative values... is this a
> time/offset issue?
>
> [image: Bildschirmfoto 2020-01-20 um 13 11 29]
> <https://user-images.githubusercontent.com/11705647/72725623-84546000-3b86-11ea-8a8a-e42fff458f71.png>

Sieht aus als würdest Du Zählerstände loggen und Deine Werte wären nicht
streng monoton steigend?

vg
Andreas

michis0806, 2020-01-20:
@andig 
> Do you really need the power reading? volkszaehler middleware calculates power values from totals. This is what we usually recommend.
Ich bekomme aus dem vzlogger nur absolute Werte.. Daher bleibt mir nichts anderes übrig als berechnen...

andig, 2020-01-20:
Ich versteh nicht welches Problem hier gelöst werden soll. Middleware loggt was vzlogger schickt und vzlogger schickt was der Zähler ausgibt. M.E. hat das nix mit den Negativausreissern zu tun?

michis0806, 2020-01-20:
Mein Problem ist, dass ich keine saubere Auswertung bekomme, da ich 
1. beim direkten abgreifen der Leistungswerte aus vzlogger keine positiv/negativ-Markierung bekomme
2. beim empfohlenen Umweg über die Berechnung der Werte negative Ausreisser habe
;)

andig, 2020-01-20:
Aus Impuls kommen keine Vorzeichen sie.  M.E. gehört die ganze Diskussion auf die ML da es hier nicht um konkrete Fehler geht. Vorschlag: wir schliessen hier.

closed 2020-03-26 by andig