volkszaehler/vzlogger #401: Compiling issue

Gitsaibot, 2019-11-19:
Tried to update vzlogger but i get always this error: 

> [ 83%] Linking CXX executable vzlogger
> cd /home/pi/vzlogger/build/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/vzlogger.dir/link.txt --verbose=1
> /usr/bin/c++   -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++11 -fpermissive -Wno-error=redundant-decls -Wno-ignored-qualifiers  -rdynamic CMakeFiles/vzlogger.dir/vzlogger.cpp.o CMakeFiles/vzlogger.dir/ltqnorm.cpp.o CMakeFiles/vzlogger.dir/Meter.cpp.o CMakeFiles/vzlogger.dir/__/gitSha1.cpp.o CMakeFiles/vzlogger.dir/CurlSessionProvider.cpp.o CMakeFiles/vzlogger.dir/PushData.cpp.o  -o vzlogger -Wl,-rpath,/usr/local/lib protocols/libproto.a libvz.a api/libvz-api.a /usr/local/lib/libjson-c.a -lrt -Wl,-Bstatic -lsml -Wl,-Bdynamic -lrt /usr/local/lib/libmbus.so -lm -Wl,-Bstatic -lmicrohttpd -Wl,-Bdynamic -lrt -lgcrypt -lpthread -lm -luuid -ldl -lcurl -lunistring -Wl,-Bstatic -lgnutls -lz -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lgmp -lhogweed -lgmp -lnettle -ltasn1 -lidn -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lz -Wl,-Bdynamic -lsasl2 -lgcrypt -lssl -lcrypto -Wl,-Bstatic -lsml -Wl,-Bdynamic /usr/local/lib/libmbus.so -lm -Wl,-Bstatic -lmicrohttpd -Wl,-Bdynamic -lgcrypt -lpthread -lm -luuid -ldl -lcurl -lunistring -Wl,-Bstatic -lgnutls -lz -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lgmp -lhogweed -lnettle -ltasn1 -lidn -lz -Wl,-Bdynamic -lp11-kit -Wl,-Bstatic -lgmp -lhogweed -lnettle -ltasn1 -lidn -Wl,-Bdynamic -lsasl2 -lgcrypt -lssl -lcrypto 
> /**usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/libgnutls.a(str-idna.o): undefined reference to symbol 'idn2_to_unicode_8z8z@@IDN2_0.0.0'**
> **/usr/bin/ld: //usr/lib/arm-linux-gnueabihf/libidn2.so.0: error adding symbols: DSO missing from command line****
> collect2: error: ld returned 1 exit status
> make[2]: *** [src/CMakeFiles/vzlogger.dir/build.make:204: src/vzlogger] Fehler 1
> make[2]: Verzeichnis „/home/pi/vzlogger/build“ wird verlassen
> make[1]: *** [CMakeFiles/Makefile2:1031: src/CMakeFiles/vzlogger.dir/all] Fehler 2
> make[1]: Verzeichnis „/home/pi/vzlogger/build“ wird verlassen
> make: *** [Makefile:166: all] Fehler 2

I run this on a raspberry Pi with Raspberry Buster installed and used this guide: https://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version
Are there more dependencies than listed in the guide?

andig, 2019-11-19:
Sieht aus wie falsche Reihenfolge beim linken: https://stackoverflow.com/questions/19901934/libpthread-so-0-error-adding-symbols-dso-missing-from-command-line <https://stackoverflow.com/questions/19901934/libpthread-so-0-error-adding-symbols-dso-missing-from-command-line>

Viele Grüße, 
Andreas

Gitsaibot, 2019-11-20:
Ja scheint so. Hab es jetzt auch mal mit dem "./install.sh vzlogger" script versucht aber da kommt auch der gleiche Fehler :( Bevor ich micht tiefer mit der Sache beschäftige läuft das script unter buster ?

J-A-U, 2019-11-20:
> läuft das script unter buster ?

Ja.

alexkuehn, 2019-11-21:
Hi,

i had the same problem. I solved it with cleaning all the prevoius build artefacts:
./install.sh vzlogger libjson libsml clean

Best regards
    Alex

Gitsaibot, 2019-11-21:
@alexkuehn Vielen Dank für den Hinweis! Nachdem ich den build Ordner komplett gelöscht habe lief das ./install.sh script ohne Probleme durch ;)

closed 2019-11-21 by Gitsaibot