volkszaehler/vzlogger #402: Cross-compile error: 'GLOB_BRACE' was not declared

winfriedp, 2019-11-22:
Cross compile for YPORT-OpenWrt environment failed. Here the last part of the log:
`[  7%] Building CXX object src/protocols/CMakeFiles/proto.dir/MeterW1therm.cpp.o
cd /home/romokerkid/openwrt/build_dir/target-mips_24kc_musl/vzlogger-master/src/protocols && /home/romokerkid/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/bin/mips-openwrt-linux-musl-g++  -DHAVE_CONFIG_HPP -DNDEBUG -I/home/romokerkid/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/romokerkid/openwrt/build_dir/target-mips_24kc_musl/vzlogger-master -I/home/romokerkid/openwrt/build_dir/target-mips_24kc_musl/vzlogger-master/include  -W -Wall -Wextra -Werror -Wnon-virtual-dtor -Wno-system-headers -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls -Wno-unused-parameter -std=c++11 -fpermissive -Wno-error=redundant-decls -Wno-ignored-qualifiers -O3 -Wno-unused-parameter -Wno-redundant-decls   -g3 -o CMakeFiles/proto.dir/MeterW1therm.cpp.o -c /home/romokerkid/openwrt/build_dir/target-mips_24kc_musl/vzlogger-master/src/protocols/MeterW1therm.cpp
/home/romokerkid/openwrt/build_dir/target-mips_24kc_musl/vzlogger-master/src/protocols/MeterW1therm.cpp: In member function 'virtual bool MeterW1therm::W1sysHWif::scanW1devices()':
/home/romokerkid/openwrt/build_dir/target-mips_24kc_musl/vzlogger-master/src/protocols/MeterW1therm.cpp:35:50: error: 'GLOB_BRACE' was not declared in this scope
   glob("/sys/bus/w1/devices/{10,22,28,3b,42}-*", GLOB_BRACE | GLOB_NOSORT, NULL, &glob_res)) {
                                                  ^~~~~~~~~~
/home/romokerkid/openwrt/build_dir/target-mips_24kc_musl/vzlogger-master/src/protocols/MeterW1therm.cpp:35:50: note: suggested alternative: 'GLOB_MARK'
   glob("/sys/bus/w1/devices/{10,22,28,3b,42}-*", GLOB_BRACE | GLOB_NOSORT, NULL, &glob_res)) {
                                                  ^~~~~~~~~~
                                                  GLOB_MARK
src/protocols/CMakeFiles/proto.dir/build.make:143: recipe for target 'src/protocols/CMakeFiles/proto.dir/MeterW1therm.cpp.o' failed
make[5]: *** [src/protocols/CMakeFiles/proto.dir/MeterW1therm.cpp.o] Error 1
make[5]: Leaving directory '/home/romokerkid/openwrt/build_dir/target-mips_24kc_musl/vzlogger-master'
CMakeFiles/Makefile2:1172: recipe for target 'src/protocols/CMakeFiles/proto.dir/all' failed
make[4]: *** [src/protocols/CMakeFiles/proto.dir/all] Error 2
make[4]: Leaving directory '/home/romokerkid/openwrt/build_dir/target-mips_24kc_musl/vzlogger-master'
Makefile:165: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/home/romokerkid/openwrt/build_dir/target-mips_24kc_musl/vzlogger-master'
Makefile:49: recipe for target '/home/romokerkid/openwrt/build_dir/target-mips_24kc_musl/vzlogger-master/.built' failed
make[2]: *** [/home/romokerkid/openwrt/build_dir/target-mips_24kc_musl/vzlogger-master/.built] Error 2
make[2]: Leaving directory '/home/romokerkid/openwrt/package/utils/vzlogger'
time: package/utils/vzlogger/compile#19.24#2.40#26.48
package/Makefile:111: recipe for target 'package/utils/vzlogger/compile' failed
make[1]: *** [package/utils/vzlogger/compile] Error 2
make[1]: Leaving directory '/home/romokerkid/openwrt'
/home/romokerkid/openwrt/include/toplevel.mk:216: recipe for target 'package/vzlogger/compile' failed
make: *** [package/vzlogger/compile] Error 2
`
My host environment is
- Ubuntu based Linux Mint v19.2 system 
- OpenWrt 18.06
- Target System: "Atheros AR7xxx/AR9xxx"
- Target Profile: "8devices Carambola2"
Compilation with `make package/vzlogger/compile V=s`

This is the vzlogger Makefile:
`include $(TOPDIR)/rules.mk

PKG_NAME:=vzlogger
PKG_VERSION:=master
PKG_RELEASE:=1
PKG_REV:=master
PKG_FIXUP:=autoreconf
PKG_BUILD_DEPENDS:=libmosquitto libsml libmicrohttpd libjson libcurl libopenssl libstdcpp libgcrypt librt libsasl2
#PKG_BUILD_DEPENDS:=mosquitto sml microhttpd json curl openssl stdcpp gcrypt rt sasl2
#PKG_BUILD_PARALLEL:=1

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
#PKG_SOURCE_URL:=git://github.com/mbehr1/vzlogger.git
PKG_SOURCE_URL:=git://github.com/volkszaehler/vzlogger.git
PKG_SOURCE_VERSION:=$(PKG_REV)
PKG_SOURCE_SUBDIR:=$(PKG_NAME)-$(PKG_VERSION)
PKG_SOURCE_PROTO:=git

CMAKE_INSTALL:=1

include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/cmake.mk

define Package/vzlogger
  SECTION:=utils
  CATEGORY:=Utilities
  DEPENDS:=+libmosquitto +libsml +libmicrohttpd +libjson +libcurl +libopenssl +libstdcpp +libgcrypt +librt +libsasl2
  URL:=https://github.com/volkszaehler/vzlogger
  TITLE:=Logging utility for several meters & sensors
endef

define Package/vzlogger/description
  Logging utility for several meters & sensors
endef

define Build/InstallDev
	$(CP) $(PKG_INSTALL_DIR)/* $(1)/
endef

define Package/vzlogger/install
	$(INSTALL_DIR) $(1)/usr/bin
	$(CP) $(PKG_INSTALL_DIR)/usr/bin/vzlogger $(1)/usr/bin
	#$(INSTALL_DIR) $(1)/etc
	#$(CP) $(PKG_INSTALL_DIR)/etc/vzlogger.conf $(1)/etc
endef

$(eval $(call BuildPackage,vzlogger,+libmosquitto +libsml,+libmicrohttpd,+libjson,+libcurl,+libopenssl,+libstdcpp,+libgcrypt,+librt,+libsasl2))`

Is there something wrong with my configuration or with the code?

andig, 2020-03-26:
Yport doesn‘t seem to be widely used any more, sorry.

m-reuter, 2021-03-25:
Hi, I am also getting: 

```
/tmp/vzlogger/src/protocols/MeterW1therm.cpp:35:50: error: 'GLOB_BRACE' was not declared in this scope
   35 |   glob("/sys/bus/w1/devices/{10,22,28,3b,42}-*", GLOB_BRACE | GLOB_NOSORT, NULL, &glob_res)) {
      |                                                  ^~~~~~~~~~
```

on alpine (which is used a lot in various frameworks that use alpine docker images as standard for add-ons).

r00t-, 2021-03-25:
caused by
https://github.com/volkszaehler/vzlogger/blob/master/src/protocols/MeterW1therm.cpp#L35

`GLOB_BRACE` is a GNU-extension,
and probably only available with glibc...
https://www.gnu.org/software/libc/manual/html_node/More-Flags-for-Globbing.html

that code (the whole meter then) should either only be built if glibc is available,
or the code rewritten without `GLOB_BRACE`.

the easy fix would be to remove the brace expression from the glob,
and do the filtering in the loop afterwards.
the slightly more elegant way might be to replace `glob()` with `opendir()`/`readdir()`.

@m-reuter: i guess alpine uses a libc other then glibc, for smaller size?

m-reuter, 2021-03-25:
Alpine uses musl libc instead, supposed to be a lightweight, fast, simple and standards-conform C library)

MichaelUray, 2022-04-09:
Did you actually successfully cross-compile vzlogger for OpenWRT?
Which steps are required for it?

r00t-, 2022-04-09:
i think he included his configuration files above, albeit badly formatted.

i cross-compile vzlogger in buildroot here (openwrt seems to use or support buildroot):
https://github.com/volkszaehler/vzlogger/pull/486

PR #470 (2021-03-31): [MeterW1therm] avoid GLOB_BRACE, don't require glibc
fix #402

~~UN~~TESTED :white_check_mark: