volkszaehler/vzlogger #404: Issus during aggreate data

magicWF, 2019-12-12:
Hello,

I have two electric meters with each 17 value:
```
[mtr0] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=410031.40 ts=1576080230051
[mtr0] Reading: id=1-0:2.8.0*255/ObisIdentifier:1-0:2.8.0*255 value=550754.00 ts=1576080230051
[mtr0] Reading: id=1-0:16.7.0*255/ObisIdentifier:1-0:16.7.0*255 value=770.00 ts=1576080230051
[mtr0] Reading: id=1-0:32.7.0*255/ObisIdentifier:1-0:32.7.0*255 value=228.20 ts=1576080230051
[mtr0] Reading: id=1-0:52.7.0*255/ObisIdentifier:1-0:52.7.0*255 value=228.20 ts=1576080230051
[mtr0] Reading: id=1-0:72.7.0*255/ObisIdentifier:1-0:72.7.0*255 value=226.80 ts=1576080230051
[mtr0] Reading: id=1-0:31.7.0*255/ObisIdentifier:1-0:31.7.0*255 value=1.10 ts=1576080230051
[mtr0] Reading: id=1-0:51.7.0*255/ObisIdentifier:1-0:51.7.0*255 value=2.49 ts=1576080230051
[mtr0] Reading: id=1-0:71.7.0*255/ObisIdentifier:1-0:71.7.0*255 value=0.92 ts=1576080230051
[mtr0] Reading: id=1-0:81.7.1*255/ObisIdentifier:1-0:81.7.1*255 value=120.00 ts=1576080230051
[mtr0] Reading: id=1-0:81.7.2*255/ObisIdentifier:1-0:81.7.2*255 value=240.00 ts=1576080230051
[mtr0] Reading: id=1-0:81.7.4*255/ObisIdentifier:1-0:81.7.4*255 value=294.00 ts=1576080230051
[mtr0] Reading: id=1-0:81.7.15*255/ObisIdentifier:1-0:81.7.15*255 value=326.00 ts=1576080230051
[mtr0] Reading: id=1-0:81.7.26*255/ObisIdentifier:1-0:81.7.26*255 value=335.00 ts=1576080230051
[mtr0] Reading: id=1-0:14.7.0*255/ObisIdentifier:1-0:14.7.0*255 value=50.00 ts=1576080230051
[mtr0] Reading: id=1-0:96.5.0*255/ObisIdentifier:1-0:96.5.0*255 value=18352680.00 ts=1576080230051
[mtr1] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=0.00 ts=1576080228754
[mtr1] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=663179.30 ts=1576080228754
[mtr1] Reading: id=1-0:2.8.0*255/ObisIdentifier:1-0:2.8.0*255 value=1786.00 ts=1576080228754
[mtr1] Reading: id=1-0:16.7.0*255/ObisIdentifier:1-0:16.7.0*255 value=0.00 ts=1576080228754
[mtr1] Reading: id=1-0:32.7.0*255/ObisIdentifier:1-0:32.7.0*255 value=228.20 ts=1576080228754
[mtr1] Reading: id=1-0:52.7.0*255/ObisIdentifier:1-0:52.7.0*255 value=228.30 ts=1576080228754
[mtr1] Reading: id=1-0:72.7.0*255/ObisIdentifier:1-0:72.7.0*255 value=226.80 ts=1576080228754
[mtr1] Reading: id=1-0:31.7.0*255/ObisIdentifier:1-0:31.7.0*255 value=0.38 ts=1576080228754
[mtr1] Reading: id=1-0:51.7.0*255/ObisIdentifier:1-0:51.7.0*255 value=0.47 ts=1576080228754
[mtr1] Reading: id=1-0:71.7.0*255/ObisIdentifier:1-0:71.7.0*255 value=0.74 ts=1576080228754
[mtr1] Reading: id=1-0:81.7.1*255/ObisIdentifier:1-0:81.7.1*255 value=239.00 ts=1576080228754
[mtr1] Reading: id=1-0:81.7.2*255/ObisIdentifier:1-0:81.7.2*255 value=121.00 ts=1576080228754
[mtr1] Reading: id=1-0:81.7.4*255/ObisIdentifier:1-0:81.7.4*255 value=90.00 ts=1576080228754
[mtr1] Reading: id=1-0:81.7.15*255/ObisIdentifier:1-0:81.7.15*255 value=90.00 ts=1576080228754
[mtr1] Reading: id=1-0:81.7.26*255/ObisIdentifier:1-0:81.7.26*255 value=90.00 ts=1576080228754
[mtr1] Reading: id=1-0:14.7.0*255/ObisIdentifier:1-0:14.7.0*255 value=50.00 ts=1576080228754
[mtr1] Reading: id=1-0:96.5.0*255/ObisIdentifier:1-0:96.5.0*255 value=18677800.00 ts=1576080228754 
```
From this value I fill 19 channles:
```
[chn0] Adding reading to queue (value=770.00 ts=1576080230051)
[chn1] Adding reading to queue (value=410031.40 ts=1576080230051)
[chn2] Adding reading to queue (value=550754.00 ts=1576080230051)
[chn3] Adding reading to queue (value=50.00 ts=1576080230051)
[chn4] Adding reading to queue (value=228.20 ts=1576080230051)
[chn5] Adding reading to queue (value=228.20 ts=1576080230051)
[chn6] Adding reading to queue (value=226.80 ts=1576080230051)
[chn7] Adding reading to queue (value=1.10 ts=1576080230051)
[chn8] Adding reading to queue (value=2.49 ts=1576080230051)
[chn9] Adding reading to queue (value=0.92 ts=1576080230051)
[chn10]Adding reading to queue (value=0.00 ts=1576080228754)
[chn11]Adding reading to queue (value=663179.30 ts=1576080228754)
[chn12]Adding reading to queue (value=1786.00 ts=1576080228754)
[chn13]Adding reading to queue (value=50.00 ts=1576080228754)
[chn14]Adding reading to queue (value=228.20 ts=1576080228754)
[chn15]Adding reading to queue (value=228.30 ts=1576080228754)
[chn16]Adding reading to queue (value=226.80 ts=1576080228754)
[chn17]Adding reading to queue (value=0.38 ts=1576080228754)
[chn18]Adding reading to queue (value=0.47 ts=1576080228754)
[chn19]Adding reading to queue (value=0.74 ts=1576080228754)
```
I have configure, that all 2 min. a value should transmit to the middleware. For that I use the parameter 
```
aggtime = 120
aggfixedinterval = true
aggmode = max
```
One time during aggregation this value is transmitted:
```
[Dec 11 17:02:08][chn1] Adding reading to queue (value=410009.50 ts=1576080128183)
[... 147 lines cut ...]
[Dec 11 17:03:50][MAX]  410031.000000 @ 1576080228401
[Dec 11 17:03:50][chn0] compare: 1576080223421 1576080225104
[Dec 11 17:03:50][MAX]  410031.400000 @ 1576080230051
[Dec 11 17:03:50][chn0] compare: 1576080225104 1576080226751
[Dec 11 17:03:50][MAX]  RESULT 410031.400000 @ 1576080230051
[Dec 11 17:03:50][chn0] compare: 1576080226751 1576080228401
[Dec 11 17:03:50][chn0] compare: 1576080228401 1576080230051
[Dec 11 17:03:50][MAX]  550754.000000 @ 1576080129850
[Dec 11 17:03:50][chn1] ==> number of tuples: 1[Dec 11 17:03:50][MAX]  550754.000000 @ 1576080131516
[Dec 11 17:03:50][MAX]  550754.000000 @ 1576080133172
[Dec 11 17:03:50][chn1] compare: 1576080120000 1576080120000
[Dec 11 17:03:50][chn0] copied 64/79 values for middleware transmission
[Dec 11 17:03:50][chn1] JSON request body is null. Nothing to send now.
```

After this, the vzlogger mean to transmit values for channel 0 but no value for channel 1, although there was new values for channel 1.
This happens several times, but not always. It results in big jumps in the power curve in the frontend.

What is going wrong?

The complete logfile can be found here:
https://www.weidemann.online/vzlogger.log.bug.txt

Best regards
Maik

andig, 2019-12-12:
Please contact the mailing list, German preferred.

magicWF, 2019-12-12:
Hallo Andig,
welche Mailing Liste?

> volkszaehler@lists.volkszaehler.org:
> SMTP error from remote server for RCPT TO command, host: mail.volkszaehler.org (217.28.98.205) reason: 550 5.1.1 <volkszaehler@lists.volkszaehler.org>: Recipient address rejecte
> d: User unknown in virtual alias table

Gruß
Maik

magicWF, 2019-12-12:
Okay gefunden:
https://demo.volkszaehler.org/mailman/listinfo/volkszaehler-users
Ist aber in der Readme.md im vzlogger Repo falsch.
Gruß
Maik

andig, 2019-12-13:
Danke für den Hinweis, das korrigiere ich.

closed 2020-03-26 by andig